fwupd/fwupd #7159: CSME version 18 not recognized

JimmyJichi, 2024-04-23:
**Describe the bug**
When running fwupdmgr security I get the following output:
✘ MEI version: Unknown

```shell
$ cat /sys/class/mei/mei0/fw_ver
0:18.0.5.2028
```

**Steps to Reproduce**
1. run fwupdmgr security
2. see MEI version unknown

**Expected behavior**
MEI version doesn't fail

**fwupd version information**

```
compile   com.hughsie.libxmlb           0.3.17
compile   org.freedesktop.Passim        0.1.5
compile   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.fwupd         1.9.16
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.Passim        0.1.5
runtime   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.8.7-300.fc40.x86_64
runtime   org.freedesktop.fwupd         1.9.16
```
Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): dnf

<details>

<summary>**fwupd device information**</summary>

```
LENOVO 83AC
│
├─Unknown Device:
│     Device ID:          a68c39791960b3933e30fa803e3ad3bb813e8850
│     GUID:               e4d9b4f9-ae47-5fe3-9ef0-a27b093b8553 ← GPIO\ID_INTC1083:00
│   
├─ATNA40YK20-0:
│     Device ID:          aec1a869eb0df71b7cea6b3ac71d39b830faf164
│     GUID:               ac5271d2-805b-5318-aa35-056f5564a2c2 ← DRM\VEN_SDC&DEV_419F
│     Device Flags:       • Internal device
│   
├─Core™ Ultra 7 155H:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000017
│     Vendor:             Intel
│     GUIDs:              e38fa480-0c3d-5240-be48-b590e507eed9 ← CPUID\PRO_0&FAM_06&MOD_AA
│                         189c1794-cc75-56bd-b1f7-8f4af3aa3f82 ← CPUID\PRO_0&FAM_06&MOD_AA&STP_4
│     Device Flags:       • Internal device
│   
├─Integrated Camera:
│     Device ID:          1bcca71b3480ec5662e5da8a742c4c805275ffbf
│     Current version:    67.5
│     Vendor:             SunplusIT Inc (USB:0x04F2)
│     Serial Number:      01.00.00
│     GUID:               cb6a721f-b424-565d-8b0b-db6bc33facc9 ← USB\VID_04F2&PID_B7ED
│     Device Flags:       • Updatable
│   
├─MZAL81T0HDLB-00BL2:
│     Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Summary:            NVM Express solid state drive
│     Current version:    5L1QKXD7
│     Vendor:             Samsung (NVME:0x144D)
[... 269 lines cut ...]
│     Current version:    207
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               832af090-2ef9-7c47-8f6d-b405c8c7f156
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
└─UEFI Platform Key:
      Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
      Summary:            Trust - Lenovo Certificate
      GUID:               a5ace873-b5ff-5d82-9051-52a46d71b6a4 ← UEFI\CRT_D1FA4BAE5073E699B7D04455F4F1126530BD69A4

```

</details>

**Additional questions**

- Operating system and version: Fedora 40
- Have you tried rebooting? Yes
- Is this a regression? No

hughsie, 2024-04-23:
Frustratingly, Intel doesn't seem to mention CMSE 17 or 18 anywhere. Can you find any references to CSME 18.x anywhere?

JimmyJichi, 2024-04-23:
> Can you find any references to CSME 18.x anywhere?
I was able to find a reference [here](https://software.intel.com/sites/manageability/HLAPI_Documentation/default.htm?turl=Documents%2FdetaileddescriptionRPE.htm). Seems to be exclusive to the Meteor Lake platform.

closed 2024-04-24 by hughsie