fwupd/fwupd #7164: Ninja build shows fwupd-efi not found.

sam412081go, 2024-04-24:
**Describe the bug**
When I use ninja build, it shows message as below:

```
Dependency fwupd-efi from subproject subprojects/fwupd-efi found: NO found 1.5 but need: '>= 1.6'

../plugins/uefi-capsule/meson.build:67:2: ERROR: Dependency 'fwupd-efi' is required but not found.

A full log can be found at /home/sdrf/fwupd_fw2_20231004/fwupd/build/meson-logs/meson-log.txt
FAILED: build.ninja 
/usr/local/bin/meson --internal regenerate /home/sdrf/fwupd_fw2_20231004/fwupd /home/sdrf/fwupd_fw2_20231004/fwupd/build --backend ninja
ninja: error: rebuilding 'build.ninja': subcommand failed
```

**Steps to Reproduce**
1. git pull to update my repository to newest status. commit(52d0a90ca7a4cb856c06a6539541320c692d4560).
2. Using "sudo ninja" in build folder.
3. Shows error message.

I also tried set efi_binary=false and rebuild meson, but still error. 

`sudo meson --reconfigure ../ -Dsystemd=false -Defi_binary=false -Ddocs=none -Dintrospection=false -Dudevdir=/tmp --prefix=/home/sdrf/fwupd_fw2_20231004/fwupd-exe`

My meson build result:

```
fwupd 2.0.0

  uefi capsule options
    efi_os_dir          : 
    efi binary          : False
    capsule splash      : True

  build targets
    fwupdtool           : True
    fwupd (daemon)      : True

  project features
    cbor                : NO
    consolekit          : True
    dbus_socket_address : 
    docs                : False
    elogind             : YES 245
    gnutls              : YES 3.6.13
    introspection       : False
    libcurl             : YES 7.68.0
    lzma                : YES 5.2.4
    polkit              : YES 0.105
    python3             : /usr/bin/python3
    supported_build     : False
    static_analysis     : False
    tests               : True
    umockdev_tests      : False

  daemon features
    bluez               : True
    gusb                : YES
    gudev               : YES 233
    hsi                 : True
    lvfs_metadata_format: xz
    libarchive          : True
    offline             : False
    passim              : NO
    sqlite              : YES 3.31.1
    thinklmi_compat     : False
    GPG support         : True
    PKCS7 support       : True

  plugins
    enabled             : acpi-dmar, acpi-facp, acpi-ivrs, acpi-phat, algoltek-usb, amd-pmc, analogix, android-boot, ata, audio-s5gen2, aver-hid, bcm57xx, bios, ccgx, ccgx-dmc, cfu, ch341a, ch347, colorhug, corsair, cpu, cros-ec, dell, dell-dock, dfu, dfu-csr, ebitdo, elanfp, elantp, emmc, ep963x, fastboot, flashrom, focalfp, fpc, fresco-pd, genesys, genesys-gl32xx, goodix-moc, goodix-tp, hailuck, intel-gsc, intel-me, intel-usb4, iommu, jabra, jabra-gnp, kinetic-dp, lenovo-thinklmi, linux-display, linux-lockdown, linux-sleep, linux-swap, linux-tainted, logind, logitech-bulkcontroller, logitech-hidpp, logitech-rallysystem, logitech-scribe, logitech-tap, mediatek-scaler, modem-manager, msr, mtd, nitrokey, nordic-hid, nvme, optionrom, parade-lspcon, pci-bcr, pci-mei, pci-psp, pixart-rf, powerd, qsi-dock, realtek-mst, redfish, rts54hid, rts54hub, scsi, steelseries, synaptics-cape, synaptics-cxaudio, synaptics-mst, synaptics-prometheus, synaptics-rmi, system76-launch, test, thelio-io, thunderbolt, ti-tps6598x, tpm, uefi-capsule, uefi-dbx, uefi-esrt, uefi-pk, uefi-recovery, uf2, upower, usi-dock, vbe, vli, wacom-raw, wacom-usb, wistron-dock
    disabled            : amd-gpu, gpio

  Subprojects
    gusb                : YES 2 warnings
    libjcat             : YES
    libxmlb             : YES 5 warnings
    passim              : NO Dependency "libsoup-3.0" not found, tried pkgconfig and cmake

  User defined options
    prefix              : /home/sdrf/fwupd_fw2_20231004/fwupd-exe
    docs                : none
    efi_binary          : false
    introspection       : false
    systemd             : false
    udevdir             : /tmp

Found ninja-1.10.0 at /usr/bin/ninja
```

This issue seems to have started from this commit (e6ba207561a5519d317dfd21db9052c7613eacc4) "trivial: bump minimum fwupd-efi version to 1.6"

**Expected behavior**
Can build normally.

**fwupd version information**
commit(52d0a90ca7a4cb856c06a6539541320c692d4560).

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Just follow fwupd wiki [build on Fedora 31](https://github.com/fwupd/fwupd/wiki/Fedora-31)

**Additional questions**

- Operating system and version:

```
sdrf@OA-109048:~/fwupd_fw2_20231004/fwupd/build$ uname -a
Linux OA-109048 5.15.0-75-generic #82~20.04.1-Ubuntu SMP Wed Jun 7 19:37:37 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

```

- Have you tried rebooting?
Yes
- Is this a regression?
No

hughsie, 2024-04-24:
Does `meson configure -Dplugin_uefi_capsule=disabled` work around the problem for now?

sam412081go, 2024-04-25:
> Does `meson configure -Dplugin_uefi_capsule=disabled` work around the problem for now?

@hughsie  Yes, it works around now. thanks!

superm1, 2024-04-25:
`-Defi_binary=false` is a better solution IMO

sam412081go, 2024-04-25:
> `-Defi_binary=false` is a better solution IMO

@superm1 yes, I tried it again earlier, it's work. Thanks!

closed 2024-04-25 by superm1