fwupd/fwupd #7166: Firmware update fails with "Error splicing file: Operation not permitted"

gkralik, 2024-04-24:
**Describe the bug**

Updating firmware fails with fwupd 1.9.18 with the error message:

`Failed to copy /usr/lib/fwupd/efi/fwupdx64.efi to /boot/EFI/arch/fwupdx64.efi: Error splicing file: Operation not permitted`

The error message appears immediately after downloading the update.  The file /boot/EFI/arch/fwupdx64.efi does get created, but it is 0 bytes. There is plenty of available space on the /boot partition (94MB).

The update works just as expected after downgrading to 1.9.16.

Maybe it's related to the SystemCallFilter changes in 1.9.17?

**Steps to Reproduce**

1. Update from 1.9.16 to 1.9.18
2. sudo fwupdmgr get-updates
3. sudo fwupdmgr update

**Expected behavior**

The firmware update should be downloaded and installed after a reboot.

**fwupd version information**

```shell
$ fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.19
compile   org.freedesktop.Passim        0.1.7
compile   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.fwupd         1.9.18
runtime   org.freedesktop.fwupd-efi     1.6
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libxmlb           0.3.19
runtime   org.freedesktop.Passim        0.1.7
runtime   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.8.7-arch1-1
runtime   org.freedesktop.fwupd         1.9.18
```

Installed via pacman on Arch Linux:

```
extra/fwupd-efi 1.6-1 (27.0 KiB 65.9 KiB) (Installed)
    EFI Application used by uefi-capsule plugin in fwupd
extra/fwupd 1.9.18-1 (5.3 MiB 11.6 MiB) (Installed)
    Simple daemon to allow session software to update firmware
```

<details>

<summary>**fwupd device information**</summary>

```
LENOVO 21AHCTO1WW
│
├─ELAN0676:00 04F3:3195:
│     Device ID:          e0e5d26dd3fa9940f593886e80a7e1eb8f564836
│     Summary:            Touchpad
│     Current version:    0x0003
│     Bootloader Version: 0x0002
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              4e8cda57-37ba-5dc6-b114-7610e38714e2 ← HIDRAW\VEN_04F3&DEV_3195
│                         cc456594-6f7e-5d1b-82a6-b4558a813152 ← HIDRAW\VEN_04F3&DEV_3195&MOD_0006
│                         d15e804c-570f-511b-803a-b38a4eddae29 ← ELANTP\ICTYPE_14&MOD_0006
│                         296e64f7-9e08-5717-bff2-651e41b0477e ← ELANTP\ICTYPE_14&MOD_0006&DRIVER_HID
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─Unknown Device:
[... 355 lines cut ...]
│     Current version:    3
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               d5fc4589-7329-4e80-a05b-1985fd8f6632
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
└─UEFI Platform Key:
      Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
      Summary:            Lenovo Ltd. PK CA 2012
      Vendor:             Lenovo Ltd.
      GUID:               71599d14-9b31-5270-b3bd-74c494585820 ← UEFI\CRT_9AEF2123F4DE7C19AFABD909BB2C8CAC4411E07E
```

</details>

**Additional questions**

- Operating system and version: Arch Linux, latest release, all updates installed
- Have you tried rebooting? yes
- Is this a regression? yes

gkralik, 2024-04-24:
Possibly related to #7144 ?

gkralik, 2024-04-24:
Reported by another affected user in Arch Linux forums: [Error updating firmware: "Error splicing file"](https://bbs.archlinux.org/viewtopic.php?id=295160)

gkralik, 2024-04-24:
Regression seems to be c4af843 . Previously, `@system-service` (which includes `splice`) was allowed, now it is not in the filter list any more.

hughsie, 2024-04-24:
@gkralik if you add `splice` to the `syscall_filter` list does that work? If so, can you do a PR please.

gkralik, 2024-04-24:
@hughsie Can't test it, the update ran with v1.9.16. I'll ask the original reporter in the Arch Forums if they can try that.

gkralik, 2024-04-24:
After a little bit of digging, I'm almost certain the syscall_filter needs splice.

[`fu_uefi_esp_target_copy()`](https://github.com/fwupd/fwupd/blob/main/plugins/uefi-capsule/fu-uefi-common.c#L318) uses [`g_file_copy()`](https://github.com/GNOME/glib/blob/main/gio/gfile.c#L3694), which does use splice if available on the system (via `file_copy_fallback()`, `splice_stream_with_progress()` - see [source](https://github.com/GNOME/glib/blob/main/gio/gfile.c#L3571).

gkralik, 2024-04-25:
The original reporter in the Arch Linux forums tested the improved filter from #7171 and reports it fixed the issue they experienced.

joakim-tjernlund, 2024-05-02:
found a similar error:
```
wupdmgr reinstall 484ea4d334faa809334a6cfb8d86f37ca9898dc9
╔══════════════════════════════════════════════════════════════════════════════╗
║ Reinstall Intel Management Engine to 16.1.30.2307?                           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Initial Production ME release.                                             ║
║                                                                              ║
║ 30GS001SMT must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: 
Decompressing…           [                                       ] Less than one minute remaining…
Failed to copy /usr/libexec/fwupd/efi/fwupdx64.efi to /boot/EFI/gentoo/fwupdx64.efi: Error splicing file: Operation not permitted
```
Then I added splice to SystemCallFilter in /usr/lib/systemd/system/fwupd.service but that did not help

gkralik, 2024-05-02:
Check #7171, `splice` might not be enough on your system.

superm1, 2024-05-02:
Also https://github.com/fwupd/fwupd/pull/7174, we're going to back this out from the stable branch, it's too much breakage.

joakim-tjernlund, 2024-05-02:
> Check #7171, `splice` might not be enough on your system.

I added all those and now it works, thanks

closed 2024-05-02 by superm1

joakim-tjernlund, 2024-05-02:
@hughsie is there a 1.9.19 coming soon?

joakim-tjernlund, 2024-05-02:
> I pondered tomorrow.

Great, then I don't need to bother Gentoo with backporting :)