fwupd/fwupd #7173: Suggestion: using dlopen() and dlsym() for libpassim.so

mohemiv, 2024-04-24:
Hello there,

Currently, the option to use passim isn't configurable after the build has been completed. This means that users who built fwupd without specific settings cannot use it later and vice versa.

I suggest using dlopen() and dlsym() for libpassim.so to provide end-users with greater flexibility in configuring the system when using fwupd.

superm1, 2024-05-01:
Considering we already do split off other things that pull in optional dependencies like flashrom plugin and modem manager plugin I could see a stronger argument for this request.

If we did this conceivably a distro packager can have a `fwupd-passim` package just like they do their own `fwupd-plugin-flashrom` and `fwupd-plugin-modem-manager`.

I think if we were to do something like this it should be with `GModule` instead of `libdl` though.

hughsie, 2024-11-11:
Nah, I think this is a slippery slope. If distros want to split up the package of passim that's fine, but I don't want to support another layer of complexity here. I think the docs changes are good enough.

closed 2024-11-11 by hughsie