fwupd/fwupd #7180: `fwupdmgr security` on Lenovo Thinkpad X1 Carbon (Gen 12)

iyanmv, 2024-04-26:
**Describe the bug**
Running `fwupdmgr security` on a Lenovo Thinkpad X1 Carbon (Gen 12) returns the following:

```
$ fwupdmgr security
Host Security ID: HSI:0! (v1.9.18)

HSI-1
✔ BIOS firmware updates:         Enabled
✔ MEI key manifest:              Valid
✔ unknown manufacturing mode:    Locked
✔ unknown override:              Locked
✔ Platform debugging:            Disabled
✔ SPI write:                     Disabled
✔ SPI lock:                      Enabled
✔ SPI BIOS region:               Locked
✔ Supported CPU:                 Valid
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI bootservice variables:    Locked
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled
✘ MEI version:                   Unknown

HSI-2
✔ BIOS rollback protection:      Enabled
✔ Intel BootGuard:               Enabled
✔ Intel BootGuard OTP fuse:      Valid
✔ IOMMU:                         Enabled
✔ Platform debugging:            Locked
✔ TPM PCR0 reconstruction:       Valid
✘ Intel BootGuard ACM protected: Invalid
✘ Intel BootGuard verified boot: Invalid

HSI-3
✔ CET Platform:                  Supported
✔ Pre-boot DMA protection:       Enabled
✔ Suspend-to-idle:               Enabled
✔ Suspend-to-ram:                Disabled
✘ Intel BootGuard error policy:  Invalid

HSI-4
✔ SMAP:                          Enabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -!
✔ fwupd plugins:                 Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Disabled
✔ Linux kernel:                  Untainted
✘ CET OS Support:                Not supported

This system has a low HSI security level.
 » https://fwupd.github.io/hsi.html#low-security-level

This system has HSI runtime issues.
 » https://fwupd.github.io/hsi.html#hsi-runtime-suffix

Host Security Events
  2024-04-24 22:59:56:  ✔ Kernel lockdown enabled
  2024-04-24 22:53:13:  ✔ BIOS rollback protection changed: Disabled → Enabled
  2024-04-24 20:55:32:  ✔ Secure Boot enabled

Host Security ID attributes uploaded successfully, thanks!
```

MEI version is missing and Intel BootGuard ACM protected and Intel BootGuard verified boot are marked as invalid although should be supported according to [the CPU specs](https://ark.intel.com/content/www/us/en/ark/products/236847/intel-core-ultra-7-processor-155h-24m-cache-up-to-4-80-ghz.html).

**fwupd version information**
Please provide the version of the daemon and client.
[... 397 lines cut ...]
│     Minimum Version:    1
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     Problems:           • Device requires AC power to be connected
│     GUID:               0dbbd775-d7ac-4b92-8e6f-eec04800ebdf
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│     Device Requests:    • Message
│   
└─UEFI Platform Key:
      Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
      Summary:            Platform Key
      GUID:               0d495878-553a-5803-ad0b-307df77318d6 ← UEFI\CRT_FEB00E5A22F47D40463AF378E6A5E774C0F0ED9E
```

</details>

**Additional questions**

- Operating system and version: Arch Linux
- Have you tried rebooting?
- Is this a regression?

hughsie, 2024-04-29:
Can you try with the 1_9_X branch in fwupd upstream please? I've pushed https://github.com/fwupd/fwupd/commit/bae12849f518eb2a9489bfb88e17e725ebe690e9 already.

hughsie, 2024-04-29:
But the bootguard problem remains I guess.

hughsie, 2024-04-29:
@iyanmv can you attach the output of `sudo fwupdtool get-plugins --plugins pci-mei -vv` please.

iyanmv, 2024-04-29:
Sure, here it is:

<details>
<summary>sudo fwupdtool get-plugins --plugins pci-mei -vv</summary>

```
$ sudo fwupdtool get-plugins --plugins pci-mei -vv                                                                                                                                            iyan@bespin
18:55:03.085 FuDebug              verbose to debug (on console 1)
18:55:03.085 FuEngine             starting fwupd 1.9.18…
18:55:03.085 FuMain               locked /run/lock/fwupdtool
18:55:03.088 FuMain               failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
Loading…                 [ -                                     ]18:55:03.088 FuConfig             loading config /etc/fwupd/fwupd.conf
18:55:03.088 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
18:55:03.088 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
18:55:03.088 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
18:55:03.088 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
18:55:03.088 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
18:55:03.088 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
18:55:03.088 FuConfig             ::configuration loaded
Loading…                 [                                       ]18:55:03.089 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
18:55:03.089 Jcat                 loaded 1 certificates
18:55:03.089 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
18:55:03.089 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
18:55:03.089 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
18:55:03.089 Jcat                 loaded 1 certificates
18:55:03.089 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
18:55:03.089 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
18:55:03.089 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
18:55:03.089 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
18:55:03.089 FuEngine             client certificate now exists: nothing to do
Loading…                 [                                       ]18:55:03.089 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
18:55:03.089 FuHistory            got schema version of 12
Loading…                 [*                                      ]18:55:03.098 FuPlugin             load(/usr/lib/fwupd-1.9.18/libfu_plugin_modem_manager.so)
18:55:03.100 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
18:55:03.100 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
18:55:03.100 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
18:55:03.100 XbSilo               file: 0dd90c7a-7be0-9815-75d8-fe37ccfd5c13, current:0dd90c7a-7be0-9815-75d8-fe37ccfd5c13, cached: (null)
18:55:03.100 XbSilo               loading silo with file contents
Loading…                 [*                                      ]18:55:03.101 FuConfig             loading config /etc/fwupd/fwupd.conf
18:55:03.101 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
18:55:03.101 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
18:55:03.101 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
18:55:03.101 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
18:55:03.101 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
18:55:03.101 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
18:55:03.101 FuConfig             ::configuration loaded
18:55:03.101 FuStruct             SmbiosEp64:
  anchor_str: _SM3_
  entry_point_csum: 0xa7
  entry_point_len: 0x18
  smbios_major_ver: 0x3
  smbios_minor_ver: 0x6
  smbios_docrev: 0x0
  entry_point_rev: 0x1
  reserved0: 0x0
  structure_table_len: 0x1450
  structure_table_addr: 0x5c06e000
18:55:03.101 FuStruct             SmbiosStructure:
  type: 0x87
  length: 0x13
  handle: 0x0
18:55:03.101 FuStruct             SmbiosStructure:
  type: 0x86
  length: 0xd
  handle: 0x1
18:55:03.101 FuStruct             SmbiosStructure:
  type: 0x2b
  length: 0x1f
  handle: 0x2
18:55:03.101 FuStruct             SmbiosStructure:
[... 774 lines cut ...]
upower:
  Flags:                • Disabled

usi_dock:
  Flags:                • Disabled

vbe:
  Flags:                • Disabled

vli:
  Flags:                • Disabled

wacom_raw:
  Flags:                • Disabled

wacom_usb:
  Flags:                • Disabled

wistron_dock:
  Flags:                • Disabled

18:55:03.224 FuPluginTest         destroy
```

</details>

hughsie, 2024-04-29:
So all the HFSTSx registers are zero. @mrhpearson do you know if Lenovo might have disabled reading the MEI config registers (PCI_CFG_HFS_x) on newer hardware? The defines I have are:

	#define PCI_CFG_HFS_1 0x40
	#define PCI_CFG_HFS_2 0x48
	#define PCI_CFG_HFS_3 0x60
	#define PCI_CFG_HFS_4 0x64
	#define PCI_CFG_HFS_5 0x68
	#define PCI_CFG_HFS_6 0x6c

Also, @iyanmv do you get the same result when disabling secure boot in the firmware setup?

iyanmv, 2024-04-29:
So here it is after disabling secure boot:

```bash
fwupdmgr security                                                                                                                                                                           iyan@bespin
Host Security ID: HSI:1! (v1.9.18)

HSI-1
✔ BIOS firmware updates:         Enabled
✔ MEI key manifest:              Valid
✔ csme manufacturing mode:       Locked
✔ csme override:                 Locked
✔ csme v0:18.0.5.2098:           Valid
✔ Platform debugging:            Disabled
✔ SPI write:                     Disabled
✔ SPI lock:                      Enabled
✔ SPI BIOS region:               Locked
✔ Supported CPU:                 Valid
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI bootservice variables:    Locked
✔ UEFI platform key:             Valid

HSI-2
✔ BIOS rollback protection:      Enabled
✔ Intel BootGuard:               Enabled
✔ Intel BootGuard OTP fuse:      Valid
✔ IOMMU:                         Enabled
✔ Platform debugging:            Locked
✔ TPM PCR0 reconstruction:       Valid
✘ Intel BootGuard ACM protected: Invalid
✘ Intel BootGuard verified boot: Invalid

HSI-3
✔ CET Platform:                  Supported
✔ Pre-boot DMA protection:       Enabled
✔ Suspend-to-idle:               Enabled
✔ Suspend-to-ram:                Disabled
✘ Intel BootGuard error policy:  Invalid

HSI-4
✔ SMAP:                          Enabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -!
✔ fwupd plugins:                 Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Encrypted
✔ Linux kernel:                  Untainted
✘ CET OS Support:                Not supported
✘ UEFI secure boot:              Disabled

This system has HSI runtime issues.
 » https://fwupd.github.io/hsi.html#hsi-runtime-suffix

Host Security Events
  2024-04-29 12:28:05:  ✘ Secure Boot disabled
  2024-04-25 22:07:12:  ✔ Linux swap changed: Disabled → Encrypted
  2024-04-24 22:59:56:  ✔ Kernel lockdown enabled
  2024-04-24 22:53:13:  ✔ BIOS rollback protection changed: Disabled → Enabled
  2024-04-24 20:55:32:  ✔ Secure Boot enabled
```

<details>
<summary>sudo fwupdtool get-plugins --plugins pci-mei -vv</summary>

```
sudo fwupdtool get-plugins --plugins pci-mei -vv                                                                                                                                            iyan@bespin
19:29:52.743 FuDebug              verbose to debug (on console 1)
19:29:52.743 FuEngine             starting fwupd 1.9.18…
19:29:52.743 FuMain               locked /run/lock/fwupdtool
[... 835 lines cut ...]

usi_dock:
  Flags:                • Disabled

vbe:
  Flags:                • Disabled

vli:
  Flags:                • Disabled

wacom_raw:
  Flags:                • Disabled

wacom_usb:
  Flags:                • Disabled

wistron_dock:
  Flags:                • Disabled

19:29:52.894 FuPluginTest         destroy
```

</details>

I think it's the same.

mrhpearson, 2024-04-30:
Created internal ticket LO-3022 to get feedback from the FW team. A note that it's a holiday in Japan and China this week so I won't get answers until at least next week.
I am seeing the same on my system - but it's a prototype so these may not have been programmed. Can I confirm that the system is a ship-level supported system please?
Mark

mrhpearson, 2024-05-08:
FW team thinks this will be fixed with https://github.com/fwupd/fwupd/pull/7163/commits/fb18ce3b0e333310731397a437b529391646bcb3
But I tested on my (prototype) system and it didn't help. Following up for more insight.

hughsie, 2024-05-08:
> FW team thinks this will be fixed with https://github.com/fwupd/fwupd/commit/fb18ce3b0e333310731397a437b529391646bcb3

Nah, that's the cosmetic fix -- the real problem is the register reads are failing.

mrhpearson, 2024-05-10:
Hi @iyanmv - could you do 'sudo lspci -xxx -s 00:16.0' on your system please? I'd like to capture the registers from a ship level support system (all of the ones in our team are pre-ship)

Thanks
Mark

iyanmv, 2024-05-11:
> Hi @iyanmv - could you do 'sudo lspci -xxx -s 00:16.0' on your system please? I'd like to capture the registers from a ship level support system (all of the ones in our team are pre-ship)
> 
> Thanks Mark
```
00:16.0 Communication controller: Intel Corporation Device 7e70 (rev 20)
00: 86 80 70 7e 06 04 10 00 20 00 80 07 00 00 80 00
10: 04 90 70 5a 40 00 00 00 00 00 00 00 00 00 00 00
20: 00 00 00 00 00 00 00 00 00 00 00 00 aa 17 1e 23
30: 00 00 00 00 50 00 00 00 00 00 00 00 ff 01 00 00
40: 45 02 00 a0 00 00 00 80 00 05 00 00 00 00 00 00
50: 01 8c 03 40 08 00 00 00 00 00 00 00 00 00 00 00
60: 20 00 00 00 00 00 00 00 03 1f f4 02 00 00 00 40
70: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
80: 00 00 00 00 00 00 00 00 00 00 00 00 05 a4 81 00
90: 38 03 e0 fe 00 00 00 00 00 00 00 00 00 00 00 00
a0: 04 00 00 00 09 00 14 f0 10 00 40 01 00 00 00 00
b0: 01 80 00 00 38 0d 0e 00 00 00 00 00 00 00 00 40
c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
f0: 00 00 00 00 00 00 00 00 00 00 20 00 00 00 00 00
```

iyanmv, 2024-05-12:
@mrhpearson (unrelated question but perhaps you can help) Do you know if the Intel Ultra 7 155H supports TME? It's not clear in [the specs](https://ark.intel.com/content/www/us/en/ark/products/236847/intel-core-ultra-7-processor-155h-24m-cache-up-to-4-80-ghz.html) (they do mention that TME-MK is not supported, but nothing about TME). It would be weird if Intel decided to drop this with previous generations supporting RAM encryption. At least the X1 Carbon Gen 11 had an option in the BIOS (I think disabled by default) to enable the TME, but this option is missing in the X1 Carbon Gen 12.

mrhpearson, 2024-05-13:
Should be there - under Security->Memory protection
At least it's showing up in the BIOS simulator (and is on my proto system): https://download.lenovo.com/bsco/index.html#/graphicalsimulator/ThinkPad%20X1%20Carbon%2012th%20Gen%20(21KC,21KD)

iyanmv, 2024-05-13:
> Should be there - under Security->Memory protection At least it's showing up in the BIOS simulator (and is on my proto system): https://download.lenovo.com/bsco/index.html#/graphicalsi
mulator/ThinkPad%20X1%20Carbon%2012th%20Gen%20(21KC,21KD)

I can't see that option on my system, only the "Execution Prevention" item.

[image]
[image]

mrhpearson, 2024-05-13:
Interesting...
I have two systems. One early proto (non-vPro) and one later proto (vPro with MIPI camera) - the option is there on my vPro system but not the non-vPro. I don't know if this is by design, or a factor of when I got my system, or something else.

Just to confirm - your system is a regular purchased Lenovo unit?

I will need to check with the FW team.
Mark

iyanmv, 2024-05-13:
> Just to confirm - your system is a regular purchased Lenovo unit?

Yes, that is correct. Bought in Switzerland (not directly from Lenovo, but from Computacenter TS GmbH) but it is registered in Lenovo website. It is the Type 21KD with the Intel 7 155H and the 1080P FHD IR+RGB camera, not the MIPI one. Not sure about the vPro or how to check that.

mrhpearson, 2024-05-16:
Some updates: According to Intel this is a industry wide issue, and is related to the contents of HFSTS6 changing - meaning fwupdmgr cannot determine the bootguard configuration correctly.
We're discussing with Intel how we address this.

@hughsie  - once I have the details I'll likely reach out offline on how to get this fixed in fwupd.

iyanmv, 2024-07-15:
@mrhpearson any news regarding Total Memory Encryption?

iyanmv, 2024-07-15:
No problem, I only ping you about the issue because I got this answer from a ticket I opened at Lenovo some months ago (also forgot about it):

> «(...) After checking your github discussion you seem to be in contact with the correct people for this issue. Sadly, the Lenovo Premier Support is "only" an extended hardware warranty to replace defective parts onsite. Anything regarding BIOS issues or faults can get reported by us, but we do not receive an immediate feedback.»

I also tried to reach Intel some time ago because I couldn't find anywhere the exact differences between their new marketing terms "Intel vPro Essentials" and "Intel vPro Enterprise" (perhaps TME in the new gen is only supported by the vPro Enterprise?), but they kindly told me to ask my laptop manufacturer 😅

mrhpearson, 2024-07-16:
FW team confirmed that the encrypted memory feature is only available with vPro systems I'm afraid.

There is an Intel white paper on it (white-paper-intel-tme.pdf) - though it seems to not be available any more (I logged into my Intel support account and checked and it wasn't there either)

iyanmv, 2024-07-16:
Thanks for confirming! Sad...

iyanmv, 2024-07-16:
Wait... you meant vPro Enterprise? This CPU is vPro but Essentials.

mrhpearson, 2024-07-16:
I thought it was just vPro, didn't matter which flavour.
The white paper seems to be back up now (https://www.intel.com/content/dam/www/central-libraries/us/en/documents/white-paper-intel-tme.pdf) and I can't find anything that specifies it being enterprise only.

Do you know what you have?
If you go into the BIOS setup (F1 during early boot) is there an AMT option under the Config menu?

Mark

iyanmv, 2024-07-16:
> Do you know what you have? If you go into the BIOS setup (F1 during early boot) is there an AMT option under the Config menu?

No, I don't have that submenu in config. Mine finishes with the Thunderbolt 4.

[image]

But I guess that is not a surprise because in [the CPU specs](https://ark.intel.com/content/www/us/en/ark/products/236847/intel-core-ultra-7-processor-155h-24m-cache-up-to-4-80-ghz.html) it says that Intel® Active Management Technology (AMT) is not supported.

If it helps, this my Machine Type Model: `21KDS00600`.

mrhpearson, 2024-07-16:
Yeah - non-vPro I'm afraid.

iyanmv, 2024-07-17:
So what you are saying is that vPro Essentials ≠ vPro? (at least regarding TME) Is AMT a requirement for TME?

[image]

iyanmv, 2024-07-18:
@mrhpearson [Intel says](community.intel.com/t5/Intel-vPro-Platform/Total-Memory-Encryption-and-Ultra-7-Processor-155H/m-p/1615881#M11773) that TME is part of Intel vPro Essentials

mrhpearson, 2024-07-18:
Interesting - let me go check with the FW team again.

mrhpearson, 2024-07-19:
FW team confirmed that we don't have a 'vPro essentials" SKU for the X1 Carbon G12.

What determines if a system is vPro or not depends on multiple components (CPU, WLAN, FW, etc) - so whilst your system may have a vPro essentials CPU, the system itself is not a vPro system - and hence TME is not supported.
I learnt something today - I thought it was all related to the CPU too.

iyanmv, 2024-07-19:
Oh, I see... I'm so confused... What can prevent a system from encrypting the memory other than the CPU? But fine, at least now I understand there is no contradiction here: the CPU is vPro but the laptop does not support TME. But then, why does TME appear in the X1 Carbon G12 BIOS simulator if it's not supported at all?

mrhpearson, 2024-07-19:
As I understand it  - It's supported on the vPro configurations (where all the HW is vPro compliant). Your system isn't considered a vPro configuration I'm afraid.
From my point of view it's a bit frustrating - I generally don't recommend vPro for Linux users, as there's no point and it's just a hassle. But having features like TME tied in with it now is.....urgh.

iyanmv, 2024-10-15:
Going back to the original topic, this is the output I get from `fwupdmgr security` after updating to 2.0.1:

```
Host Security ID: HSI:1 (v2.0.1)

HSI-1
✔ BIOS firmware updates:         Enabled
✔ MEI manufacturing mode:        Locked
✔ csme18 override:               Locked
✔ csme18 v0:18.0.5.2098:         Valid
✔ Platform debugging:            Disabled
✔ SPI write:                     Disabled
✔ SPI lock:                      Enabled
✔ SPI BIOS region:               Locked
✔ Supported CPU:                 Valid
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI bootservice variables:    Locked
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled

HSI-2
✔ BIOS rollback protection:      Enabled
✔ Intel BootGuard ACM protected: Valid
✔ Intel BootGuard:               Enabled
✔ IOMMU:                         Enabled
✔ Platform debugging:            Locked
✔ TPM PCR0 reconstruction:       Valid
✘ Intel BootGuard OTP fuse:      Unlocked

HSI-3
✔ CET Platform:                  Supported
✔ Pre-boot DMA protection:       Enabled
✔ Suspend-to-idle:               Enabled
✔ Suspend-to-ram:                Disabled

HSI-4
✔ SMAP:                          Enabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -!
✔ CET OS Support:                Supported
✔ fwupd plugins:                 Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Encrypted
✔ Linux kernel:                  Untainted
```

For completeness, here is `fwupdmgr --version` as well:

```
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.1
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.6
runtime   org.freedesktop.Passim        0.1.8
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.11.3-arch1-1
runtime   org.freedesktop.fwupd         2.0.1
```

hughsie, 2024-11-11:
@superm1 for CSME18 HFSTS6 I think the Manufacturing Lock is supposed to be set. According to Intel Doc. No.: 729124, Rev.: 1.1 I read:

    Manufacturing Lock : Indicates manufacturing variables
    0 - Manufacturing variables lock is not set.
    1 - Manufacturing variables locked.

...although the table formatting is very messed up for me -- so there might be a fix in later document versions (which I've not got). The struct def I'm using is:

    struct FuMeiCsme18Hfsts6 {
        _reserved0: u21,
        manufacturing_lock: u1,
        _reserved1: u8,
        fpf_soc_configuration_lock: u1,
        _sx_resume_type: u1,
    }

superm1, 2024-11-11:
Did you mean to ping me? Maybe you meant to ping mark?

hughsie, 2024-11-11:
Gah, i did -- sorry for the noise. @mrhpearson one for you I'm afraid.

mrhpearson, 2024-11-11:
Your structure looks correct to me. Checking the doc I have:
```
bit 21 - manufacturing lock
bit 30 - FPF SOC config lock
bit 31- Sx resume type.
```
@hughsie - looking at the code, does it matter that in the old HFST6 it's called fpf_soc_lock and in the new one it's fpf_soc_configuration_lock?

I'll build and try here - but as my system is a proto I'm not sure how reliable it is....

Mark

mrhpearson, 2024-11-11:
Ignore the comment about naming - that's not the issue. I'll create a ticket for input from the FW team.

hughsie, 2024-11-11:
> does it matter that in the old HFST6 it's called fpf_soc_lock and in the new one it's fpf_soc_configuration_lock

Nah, it's namespaced; e.g. `fu_mei_csme18_hfsts6_get_manufacturing_lock` vs. `fu_mei_csme11_hfsts6_get_fpf_soc_lock` -- but you did make me check all the types, and there's one that was wrong. It's not this bug tho.

siwei-luo, 2024-11-29:
Having issues with the OTP fuse on my X1 2-in-1 Gen 9 (with vPro SoC) too. Could this be related to this one? Also not sure if disabling Intel AMT "permanently" could have caused this?

The output of `fwupdmgr security`:
```
Host Security ID: HSI:1 (v2.0.1)

HSI-1
✔ BIOS firmware updates:         Enabled
✔ MEI manufacturing mode:        Locked
✔ csme18 override:               Locked
✔ csme18 v0:18.0.5.2098:         Valid
✔ Platform debugging:            Disabled
✔ SPI write:                     Disabled
✔ SPI lock:                      Enabled
✔ SPI BIOS region:               Locked
✔ Supported CPU:                 Valid
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI bootservice variables:    Locked
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled

HSI-2
✔ BIOS rollback protection:      Enabled
✔ Intel BootGuard ACM protected: Valid
✔ Intel BootGuard:               Enabled
✔ IOMMU:                         Enabled
✔ Platform debugging:            Locked
✔ TPM PCR0 reconstruction:       Valid
✘ Intel BootGuard OTP fuse:      Invalid

HSI-3
✔ CET Platform:                  Supported
✔ Pre-boot DMA protection:       Enabled
✔ Suspend-to-idle:               Enabled
✔ Suspend-to-ram:                Disabled

HSI-4
✔ Encrypted RAM:                 Enabled
✔ SMAP:                          Enabled

Runtime Suffix -!
✔ CET OS Support:                Supported
✔ fwupd plugins:                 Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Disabled
✔ Linux kernel:                  Untainted

Host Security Events
  2024-11-26 15:59:55:  ✘ Intel BootGuard OTP fuse changed: Unlocked → Invalid
  2024-11-25 17:13:43:  ✘ Intel BootGuard OTP fuse changed: Invalid → Unlocked
  2024-11-24 21:14:32:  ✘ Intel BootGuard OTP fuse changed: Unlocked → Invalid
  2024-11-24 20:48:06:  ✔ Kernel lockdown enabled
  2024-11-23 10:00:53:  ✘ Intel BootGuard OTP fuse changed: Invalid → Unlocked
  2024-11-18 20:36:20:  ✘ Intel BootGuard OTP fuse changed: Unlocked → Invalid
  2024-11-18 20:33:41:  ✘ Intel BootGuard OTP fuse changed: Invalid → Unlocked
  2024-11-18 19:21:41:  ✘ Intel BootGuard OTP fuse changed: Unlocked → Invalid
```

Also for completeness `fwupdmgr --version`:
```
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.1
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.7
runtime   org.freedesktop.Passim        0.1.8
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.2
runtime   org.kernel                    6.12.1-arch1-1
runtime   org.freedesktop.fwupd         2.0.1
```

siwei-luo, 2024-12-01:
Short follow-up: with version `2.0.2` it became `Unlocked` like in the OP's post in my case.

```
HSI-2
✔ BIOS rollback protection:      Enabled
✔ Intel BootGuard ACM protected: Valid
✔ Intel BootGuard:               Enabled
✔ IOMMU:                         Enabled
✔ Platform debugging:            Locked
✔ TPM PCR0 reconstruction:       Valid
✘ Intel BootGuard OTP fuse:      Unlocked
```

```
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.2
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.2
runtime   org.freedesktop.fwupd-efi     1.7
runtime   org.freedesktop.Passim        0.1.8
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.2
runtime   org.kernel                    6.12.1-arch1-1
runtime   org.freedesktop.fwupd         2.0.2
```

iyanmv, 2024-12-01:
For me it's flipping between invalid and unlocked all the times:

```
Host Security Events
  2024-11-29 15:24:02:  ✘ Intel BootGuard OTP fuse changed: Invalid → Unlocked
  2024-11-29 15:20:29:  ✘ Intel BootGuard OTP fuse changed: Unlocked → Invalid
  2024-11-28 00:10:29:  ✘ Intel BootGuard OTP fuse changed: Invalid → Unlocked
  2024-11-27 15:05:09:  ✘ Intel BootGuard OTP fuse changed: Unlocked → Invalid
  2024-11-27 08:10:58:  ✘ Intel BootGuard OTP fuse changed: Invalid → Unlocked
  2024-11-25 12:37:28:  ✘ Intel BootGuard OTP fuse changed: Unlocked → Invalid
```

hughsie, 2024-12-02:
So this is CSME 18 -- not-locked is when `fu_mei_csme18_hfsts6_get_manufacturing_lock()` is **not** set. Not valid is when `fu_mei_csme18_hfsts6_get_fpf_soc_configuration_lock()` is not set.` Debugging now.

hughsie, 2024-12-02:
@iyanmv would it be possible to build the `hughsie/pci-mei-to-string` branch and attach the complete output of `./src/fwupdtool get-plugins -vv --plugins pci-mei` please?

iyanmv, 2024-12-03:
> @iyanmv would it be possible to build the `hughsie/pci-mei-to-string` branch and attach the complete output of `./src/fwupdtool get-plugins -vv --plugins pci-mei` please?

Sure, but I think you forgot to push that branch here.

hughsie, 2024-12-03:
Ahh sorry -- it was pushed, but we merged it to main (which then deleted the branch) -- can you try with main pls.

iyanmv, 2024-12-03:
Here it is: [output_fwupdtool.txt](https://github.com/user-attachments/files/17992913/output_fwupdtool.txt)

hughsie, 2024-12-03:
I didn't expect that:

	Hfsts1:               0xa0000245
	Hfsts2:               0x9f10500
	Hfsts3:               0x20
	Hfsts5:               0x2f41f03
	Family:               csme18
	VersionMajor:         18
	VersionHotfix:        5
	VersionBuildno:       2098

Hfsts4 being missing is half expected, but Hfsts6 should be there. I wonder, could you add some debugging for fu_udev_device_pread() in fu_pci_mei_plugin_backend_device_added() -- i.e. is the value of `self->hfsts_buf[i]` being read as zero?

mcatanzaro, 2024-12-05:
> Hfsts4 being missing is half expected, but Hfsts6 should be there. I wonder, could you add some debugging for fu_udev_device_pread() in fu_pci_mei_plugin_backend_device_added() -- i.e. is the value of `self->hfsts_buf[i]` being read as zero?

I have an X1 Carbon 2-in-1 Gen 9 with what we think is the same problem. `self->hfsts_buf[6]` is a non-null pointer. `self->hfsts_buf[6][0]`, `self->hfsts_buf[6][1]`, and `self->hfsts_buf[6][2]` are all 0. `self->hfsts_buf[6][0]` is 64.

mcatanzaro, 2024-12-06:
I found a bug in the debug logging and opened https://github.com/fwupd/fwupd/pull/8212. It's just not printing Hfsts6 by mistake. With the bug fixed, I now see this:

```
Hfsts1:               0xa0000245
Hfsts2:               0xf10500
Hfsts3:               0x20
Hfsts5:               0x2f41f03
Hfsts6:               0x40000000
Family:               csme18
VersionMajor:         18
VersionHotfix:        5
VersionBuildno:       2098
```

Note that my Hfsts2 is different from Iyán's (0xf10500 vs. 0x9f10500).

mcatanzaro, 2024-12-06:
My debug log (taken before the fix, so missing Hfsts6 line): https://github.com/user-attachments/files/18030216/debug.log

mcatanzaro, 2024-12-06:
Maybe fwupdtool or the pci-mei plugin should print an error if run as non-root? By default, it appears to work, but doesn't print the data that you're interested in. Anyway, here's my debug log, in case it helps: [debug.txt](https://github.com/user-attachments/files/18040695/debug.txt)

mcatanzaro, 2024-12-11:
I found this buried in my build directory (in fu-mei-struct.c):

```
guint32
fu_mei_csme18_hfsts6_get_manufacturing_lock(const FuMeiCsme18Hfsts6 *st)
{
    guint32 val;
    g_return_val_if_fail(st != NULL, 0x0);
    g_return_val_if_fail(st->len >= sizeof(guint32), 0x0);
    val = fu_memread_uint32(st->data + 0, G_LITTLE_ENDIAN);
    return (val >> 21) & 1;
}
```

So Hfsts6 is initially read as 64 (0x40). But that's a 32-byte big endian value being interpreted as little endian. After swapping the byte order, it's really 0x40000000, or 0b01000000000000000000000000000000. Bit 30 is set and all other bits are unset. The code shifts right 21 bits to test whether bit 21 is set. It's not and the check fails.

mcatanzaro, 2024-12-11:
And notably, bit 30 is checked by `fu_mei_csme18_hfsts6_get_fpf_soc_configuration_lock()` which is used to decide valid/invalid. So the value is determined to be valid but not locked.

Presumably either (a) the lock check is somehow wrong, or (b) Lenovo really forgot to disable manufacturing mode.

mcatanzaro, 2024-12-11:
One more thing: the code in `fu_pci_mei_plugin_detect_family()` just assumes it's csme18 if the version is higher than 17. Seems a little fragile; if the version is now actually higher than 18, then all bets are off. In this case, it will even print the wrong VersionMajor in the debug log.

mcatanzaro, 2024-12-18:
My return window for this laptop is closing soon, so I've called Lenovo technical support and what they want me to do is send the laptop in for replacement, then they'll send me a new laptop of the same model, which we agree will likely have the same problem since it will use the same firmware. Then they will create an escalation after confirming the problem does indeed occur on multiple devices. Sigh. It's a shame since there is nothing wrong with the hardware, but I'll plan to initiate this tomorrow due to the very limited time remaining in my return window.

@mrhpearson if there's been any action on that LO-3390, that would be good to know.

siwei-luo, 2024-12-18:
@mcatanzaro Let me know if I should test something on my device still. My C skills are very limited to non-existent though, so is my knowledge whether that fuse is actually reversible at all.

Still have my hopes that this is either a display bug and/or is going to be fixed in a upcoming firmware update somehow.

My device is still in active support but since it is my daily driver and everything else is working properly, my motivations are currently low to get through the same process.

mcatanzaro, 2024-12-18:
I think we don't need anything from you right now.

> so is my knowledge whether that fuse is actually reversible at all.

OTP stands for "one time programmable"; the idea is it's indeed irreversible once locked. But the security test failure indicates it hasn't been locked yet.

> Also not sure if disabling Intel AMT "permanently" could have caused this? I also understand that there is no way back :(.

I think you're talking about Absolute Persistence. This is unrelated and disabling it is not a problem.

mrhpearson, 2024-12-18:
Right now I'm still waiting on the FW team for insight to LO-3390. Admittedly I'd held off nagging them the last week or so based on the discussion in this thread - just in case it was SW  related (that's my excuse anyway....).

Can you do  'sudo lspci -xxx -s 00:16.0' so I can add a register dump to the ticket and remove fwupd from the equation please (which will make reviewing with the FW team easier)

siwei-luo, 2024-12-18:
Here's my output `sudo lspci -xxx -s 00:16.0`

```
00:16.0 Communication controller: Intel Corporation Meteor Lake-P CSME HECI #1 (rev 20)
00: 86 80 70 7e 06 04 10 00 20 00 80 07 00 00 80 00
10: 04 d0 71 5a 40 00 00 00 00 00 00 00 00 00 00 00
20: 00 00 00 00 00 00 00 00 00 00 00 00 aa 17 1f 23
30: 00 00 00 00 50 00 00 00 00 00 00 00 ff 01 00 00
40: 45 02 00 94 00 00 00 80 00 85 11 00 00 00 00 00
50: 01 8c 03 40 08 00 00 00 00 00 00 00 00 00 00 00
60: 30 00 00 00 00 00 00 00 03 1f f6 02 00 00 00 40
70: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
80: 00 00 00 00 00 00 00 00 00 00 00 00 05 a4 81 00
90: 38 07 e0 fe 00 00 00 00 00 00 00 00 00 00 00 00
a0: 04 00 00 00 09 00 14 f0 10 00 40 01 00 00 00 00
b0: 01 80 00 00 38 0d 0e 00 00 00 00 00 00 00 00 40
c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
f0: 00 00 00 00 00 00 00 00 00 00 20 00 00 00 00 00
```

Thanks for your efforts!

mcatanzaro, 2024-12-18:
Thank you Mark, much appreciated!

My lspci output (slightly different from Siwei's):

```
00:16.0 Communication controller: Intel Corporation Meteor Lake-P CSME HECI #1 (rev 20)
00: 86 80 70 7e 06 04 10 00 20 00 80 07 00 00 80 00
10: 04 d0 71 5a 40 00 00 00 00 00 00 00 00 00 00 00
20: 00 00 00 00 00 00 00 00 00 00 00 00 aa 17 1f 23
30: 00 00 00 00 50 00 00 00 00 00 00 00 ff 01 00 00
40: 45 02 00 a0 00 00 00 80 00 05 f1 00 00 00 00 00
50: 01 8c 03 40 08 00 00 00 00 00 00 00 00 00 00 00
60: 20 00 00 00 00 00 00 00 03 1f f4 02 00 00 00 40
70: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
80: 00 00 00 00 00 00 00 00 00 00 00 00 05 a4 81 00
90: 78 07 e0 fe 00 00 00 00 00 00 00 00 00 00 00 00
a0: 04 00 00 00 09 00 14 f0 10 00 40 01 00 00 00 00
b0: 01 80 00 00 38 0d 0e 00 00 00 00 00 00 00 00 40
c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
f0: 00 00 00 00 00 00 00 00 00 00 20 00 00 00 00 00
```

[7 more replies]

closed 2025-01-14 by hughsie