fwupd/fwupd #7198: Dell WD19 upgrade issue

Karmanuk, 2024-05-02:
**Describe the bug**
Fwupd installed via brew on macOS Sonoma, after update it gives an error
Failed to lock: another instance has locked /var/run/fwupdtool

**Steps to Reproduce**
sudo fwupdtool refresh
sudo fwupdtool get-updates
sudo fwupdtool update 

**Expected behavior**
the update worked

**fwupd version information**
1.9.18

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): brew

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the external fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --filter=~internal
```

</details>

**Dock SKU**
Please mention which module is installed in your WD19.

- [ +] WD19 (Single-C)
- [ ] WD19TB (Thunderbolt)
- [ ] WD19DC (Dual-C)

**Peripherals connected to the dock**
Please describe all devices connected to the dock.  Be as specific as possible,
including USB devices, hubs, monitors, and downstream type-C devices.

**Verbose daemon logs**
First enable daemon verbose logs collection.

```shell
fwupdmgr modify-config "VerboseDomains" "*"
```

Then try to reproduce the issue.  Even if it doesn't reproduce, please attach the
daemon verbose logs collected from the system journal.

```shell
journalctl -b -u fwupd.service
```

**Additional questions**

- Operating system and version:
- Have you tried unplugging the dock or any peripherals from your machine?
- Have you tried to power cycle the dock from the AC adapter?
- Is this a regression?

hughsie, 2024-05-02:
> Failed to lock: another instance has locked /var/run/fwupdtool

I don't see that on brew-installed fwupd with my Sonoma 14.4 system here -- can you get the `sudo fwupdtool update -vv` output please.

Karmanuk, 2024-05-02:
Password:
12:40:18.680 FuDebug              verbose to debug (on console 1)
12:40:18.692 FuEngine             starting fwupd 1.9.18…
12:40:18.692 FuMain               locked /var/run/fwupdtool
Loading…                 [ -                                     ]12:40:18.693 FuEngine             failed to build machine-id: The machine-id is not present
12:40:18.694 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
12:40:18.694 FuConfig             loading config /usr/local/Cellar/fwupd/1.9.18/etc/fwupd/fwupd.conf
12:40:18.694 FuConfig             not loading config /usr/local/Cellar/fwupd/1.9.18/var/etc/fwupd/fwupd.conf
12:40:18.694 FuConfig             skipping mode check for /usr/local/Cellar/fwupd/1.9.18/var/etc/fwupd/fwupd.conf as not writable
12:40:18.694 FuConfig             trying to load config values from /usr/local/Cellar/fwupd/1.9.18/etc/fwupd/fwupd.conf
12:40:18.695 FuCommon             mapped file /usr/local/Cellar/fwupd/1.9.18/etc/fwupd/fwupd.conf of size 0x33
12:40:18.695 FuConfig             trying to load config values from /usr/local/Cellar/fwupd/1.9.18/var/etc/fwupd/fwupd.conf
12:40:18.695 FuConfig             Failed to open file “/usr/local/Cellar/fwupd/1.9.18/var/etc/fwupd/fwupd.conf”: No such file or directory
12:40:18.696 FuConfig             ::configuration loaded
12:40:18.697 FuRemoteList         loading /usr/local/Cellar/fwupd/1.9.18/share/fwupd/metainfo
12:40:18.700 XbSilo               attempting to load /usr/local/Cellar/fwupd/1.9.18/var/cache/fwupd/metainfo.xmlb
12:40:18.701 XbSilo               file: f265a22e-a39a-4dc9-9502-bc269a5e8ee2, current:f265a22e-a39a-4dc9-9502-bc269a5e8ee2, cached: (null)
12:40:18.701 XbSilo               loading silo with file contents
12:40:18.713 FuRemoteList         loading remote from /usr/local/Cellar/fwupd/1.9.18/etc/fwupd/remotes.d/lvfs-testing.conf
12:40:18.714 FuRemoteList         ::remote_list changed
12:40:18.714 FuRemoteList         loading remote from /usr/local/Cellar/fwupd/1.9.18/etc/fwupd/remotes.d/lvfs.conf
12:40:18.715 FuRemoteList         ::remote_list changed
12:40:18.715 FuRemoteList         loading remote from /usr/local/Cellar/fwupd/1.9.18/etc/fwupd/remotes.d/vendor-directory.conf
12:40:18.716 FuRemoteList         ::remote_list changed
12:40:18.716 FuEngine             priority local and vendor-directory is not download remote, so bumping
12:40:18.716 FuRemoteList         ordering lvfs-testing=lvfs+1
12:40:18.716 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs[0]
Loading…                 [                                       ]12:40:18.716 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
12:40:18.716 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
12:40:18.716 Jcat                 reading /usr/local/Cellar/fwupd/1.9.18/etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
12:40:18.720 Jcat                 loaded 1 certificates
12:40:18.720 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
12:40:18.720 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
12:40:18.720 Jcat                 reading /usr/local/Cellar/fwupd/1.9.18/etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
12:40:18.720 Jcat                 loaded 1 certificates
12:40:18.720 Jcat                 reading /usr/local/Cellar/fwupd/1.9.18/var/lib/fwupd/pki/secret.key with 2459 bytes
12:40:18.721 Jcat                 reading /usr/local/Cellar/fwupd/1.9.18/var/lib/fwupd/pki/client.pem with 1383 bytes
12:40:18.721 FuEngine             client certificate now exists: nothing to do
Loading…                 [                                       ]12:40:18.721 FuHistory            trying to open database '/usr/local/Cellar/fwupd/1.9.18/var/lib/fwupd/pending.db'
12:40:18.727 FuHistory            got schema version of 12
Loading…                 [*                                      ]12:40:18.732 FuQuirks             loading quirks from /usr/local/Cellar/fwupd/1.9.18/share/fwupd/quirks.d
12:40:18.732 FuQuirks             loading quirks from /usr/local/Cellar/fwupd/1.9.18/var/lib/fwupd/quirks.d
12:40:18.733 XbSilo               attempting to load /usr/local/Cellar/fwupd/1.9.18/var/cache/fwupd/quirks.xmlb
12:40:18.733 XbSilo               file: d152df89-4149-d53c-06c7-54bf1e9075fb, current:d152df89-4149-d53c-06c7-54bf1e9075fb, cached: (null)
12:40:18.733 XbSilo               loading silo with file contents
Loading…                 [*                                      ]12:40:18.735 FuConfig             loading config /usr/local/Cellar/fwupd/1.9.18/etc/fwupd/fwupd.conf
12:40:18.735 FuConfig             not loading config /usr/local/Cellar/fwupd/1.9.18/var/etc/fwupd/fwupd.conf
12:40:18.735 FuConfig             skipping mode check for /usr/local/Cellar/fwupd/1.9.18/var/etc/fwupd/fwupd.conf as not writable
12:40:18.735 FuConfig             trying to load config values from /usr/local/Cellar/fwupd/1.9.18/etc/fwupd/fwupd.conf
12:40:18.735 FuCommon             mapped file /usr/local/Cellar/fwupd/1.9.18/etc/fwupd/fwupd.conf of size 0x33
12:40:18.735 FuConfig             trying to load config values from /usr/local/Cellar/fwupd/1.9.18/var/etc/fwupd/fwupd.conf
12:40:18.735 FuConfig             Failed to open file “/usr/local/Cellar/fwupd/1.9.18/var/etc/fwupd/fwupd.conf”: No such file or directory
[... cut ...]
║ Upgrade WD19S from 01.01.00.05 to 01.01.00.07?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ -Added the Universal Power Button feature for the Dell Dock WD19.            ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: 
Failed to lock: another instance has locked /var/run/fwupdtool
12:40:51.135 FuPluginTest         destroy
administrator@Administrators-MacBook-Pro ~ %

hughsie, 2024-05-02:
Doh, I should have backported e835714257f5ddd0f67f18c04b4ebffb67679690 into 1_9_X apologies.

Karmanuk, 2024-05-03:
Thanks, it worked 1.9.19.

closed 2024-05-03 by Karmanuk