fwupd/fwupd #7200: [regression] Wacom TitanLite drawing tablet (Wacom One 12 model DTC121)  fwupd upgrade or downgrade fails with "device version not updated on success, 01.01 != 1.0"

bleungatchromium, 2024-05-02:
**Describe the bug**
A clear and concise description of what the bug is.
Upgrading or downgrading wacom titanlite results in an error, despite the fact that the current version is updated correctly.

**Steps to Reproduce**
Steps to reproduce the behavior.

`fwupd get-devices` and check the Wacom One pen display 11.6" version:

```
└─Wacom One pen display 11.6":
  │   Device ID:          9713db10df106d95a97340ba7b2a82700a0c3838
  │   Current version:    1.1
  │   Bootloader Version: 1.45
  │   Vendor:             Wacom Co.,Ltd. (USB:0x056A)
  │   Install Duration:   10 seconds
  │   Serial Number:      2GHS1E1000082
  │   GUID:               f69e78d8-ff54-5162-8b90-90f6d37fc5bc ← USB\VID_056A&PID_03CE
  │   Device Flags:       • Updatable
  │                       • Supported on remote server
  │                       • Unsigned Payload
  │
  └─Wacom One pen display 11.6" [Scaler Module]:
        Device ID:        1175e6b0cc5d4c616f179ac382d4fa7679c9fbd6
        Current version:  0.1
        Vendor:           Wacom Co.,Ltd. (USB:0x056A)
        Install Duration: 2 minutes
        GUID:             dba11a22-090b-5142-8310-38bdd033c73b ← USB\VID_056A&PID_03CE-scaler
        Device Flags:     • Updatable
                          • Unsigned Payload
```

Ask `fwupdmgr` to install the older 1.0 version
`fwupdmgr install 9713db10df106d95a97340ba7b2a82700a0c3838 1.0 --allow-older`

This returns:

```
device version not updated on success, 01.00 != 1.1
```

**Expected behavior**
A clear and concise description of what you expected to happen.

Successfully installed firmware

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
```

```
compile   org.freedesktop.fwupd         1.9.14
compile   com.hughsie.libxmlb           0.3.15
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.14
runtime   org.kernel                    6.7.7-200.fc39.x86_64
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
came with fedora.

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**Additional questions**

- Operating system and version: Fedora Workstation 39.
- Have you tried rebooting?
- Is this a regression? Yes, this is a regression. Other systems running 1.9.10 do not exhibit the problem.

bleungatchromium, 2024-05-02:
Issue doesn't occur on 1_9_X branch right now. I'm doing a bisect between 1.9.15 and 1_9_X to see where we fixed it.

bleungatchromium, 2024-05-02:
This commit fixes the issue, as @hughsie suggested in the commit message.

```
286c89eb1cf01b34d958492aab1aeb3a305eaee7 is the first new commit
commit 286c89eb1cf01b34d958492aab1aeb3a305eaee7
Author: Richard Hughes <richard@hughsie.com>
Date:   Tue Mar 12 20:37:15 2024 +0000

    trivial: Move the new version check to after the composite cleanup
    
    This might fix Wacom tablet success reporting.

 src/fu-engine.c | 64 +++++++++++++++++++++++++++++++++++++++------------------
```

bleungatchromium, 2024-05-02:
Hi @superm1 

Yeah, thanks for the heads up.

I'm trying to figure out how many ChromeOS branches were affected. We have builds on 1.9.10, 1.9.13, 1.9.15 and 1.9.16.

We know that 15 is affected, and that 10, and 16 are passing, but I need to check 13. I want to figure out if it's worth cherry-picking a few changes to close a gap on one build of ChromeOS.

bleungatchromium, 2024-05-02:
1.9.13 also fails. This regressed somewhere between 1.9.10 and 1.9.13.

hughsie, 2024-05-03:
I've created https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Wacom-device-version-not-updated-on-success and added a known issue rule on the LVFS -- which matched 7 failed reports. I'll close this one now -- and FYI I tagged 1.9.19 this morning.

closed 2024-05-03 by hughsie

bleungatchromium, 2024-05-03:
For posterity, I went ahead and bisected down between 1.9.10 and 1.9.13 to figure out when this happened.

Commit 0d5b3f31e87d7929b3724d3c6eaa564b05ce7909 introduced in 1.9.11 caused the regression:

```
0d5b3f31e87d7929b3724d3c6eaa564b05ce7909 is the first bad commit
commit 0d5b3f31e87d7929b3724d3c6eaa564b05ce7909
Author: Denis Pynkin <denis.pynkin@collabora.com>
Date:   Thu Dec 7 20:46:23 2023 +0100

    Fail the update if version doesn't match expected
    
    If the flashed version doesn't match version read from the CAB file,
    the upgrade must fail.
    
    Fixes https://github.com/fwupd/fwupd/issues/6513

 src/fu-engine.c | 12 +++++++-----
 1 file changed, 7 insertions(+), 5 deletions(-)
```