fwupd/fwupd #7201: I have no idea what Fused Platform is or how to enable it

TheCorruptCat, 2024-05-02:
**Describe the question**
I want to pass hsi-1 checks but it says Fused Platform has failed I don't known what is means or how to enable it can someone please help me. Here is the link to the report

https://pastebin.com/raV3HBJ1

**fwupd version information**
fwupd-1.9.16-1.fc40.x86_64 and Fedora 40

```shell
compile   org.freedesktop.fwupd         1.9.16
compile   com.hughsie.libxmlb           0.3.17
compile   org.freedesktop.Passim        0.1.5
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.6
compile   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.Passim        0.1.7
runtime   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd         1.9.16
runtime   org.kernel                    6.8.7-300.fc40.x86_64
```

superm1, 2024-05-02:
Is your BIOS up to date? Can you please share your dmesg log?

TheCorruptCat, 2024-05-03:
I'm on the latest bios firmware

TheCorruptCat, 2024-05-03:
https://pastebin.com/DG4ws6c1

superm1, 2024-05-03:
Got it; unfortunately the Platform firmware security processor doesn't export security attributes for this system so there isn't anything that fwupd can do to change this.

closed 2024-05-03 by superm1

IPlayZed, 2024-06-23:
> Got it; unfortunately the Platform firmware security processor doesn't export security attributes for this system so there isn't anything that fwupd can do to change this.

I want to check myself for as well, why is the state _Unsupported_, how did you conclude from the uploaded pastbin that the security processor does not export the needed security attributes?

superm1, 2024-06-23:
Try this series:

https://lore.kernel.org/linux-crypto/20240528210712.1268-1-mario.limonciello@amd.com/