fwupd/fwupd #7207: `fwupd.org` TLS implementation bug or misconfiguration?

iyanmv, 2024-05-06:
I recently noticed that `fwupd` failed to download some updates when [`oqsprovider`](https://github.com/open-quantum-safe/oqs-provider) was enabled on my system. To give a bit of context for those unfamiliar with `oqsprovider`, this is a provider for OpenSSL 3.x that exposes post-quantum KEM and DS algorithms from [`liboqs`](https://github.com/open-quantum-safe/liboqs). Both are part of [the Open Quantum Safe project](https://openquantumsafe.org/), which since recently is part of the Linux Foundation's [Post-Quantum Cryptography Alliance](https://pqca.org/). Using this provider, it is already possible, e.g., to generate certificates with the new ML-DSA ([FIPS 204](https://csrc.nist.gov/pubs/fips/204/ipd)). Of course, it is also possible to start [testing TLS connections](https://test.openquantumsafe.org/) with these new algorithms.

Going back to the issue. After noticing the problem with `fwupdmgr`, I was able to replicate the issue using `openssl s_client` directly trying to connect to `fwupd.org:443`. In particular, with the following command:

```bash
# This works
echo Q | openssl s_client -provider default -connect fwupd.org:443
# This fails
echo Q | openssl s_client -provider default -provider oqsprovider -connect fwupd.org:443
```

This works as expected without `oqsprovider` enabled but it fails with it. After some further experiments (see [this discussion](https://github.com/open-quantum-safe/oqs-provider/issues/399) for more details), it seems that the real issue is the server aborting the TLS handshake when the clients supports too many signature algorithms (more than ~40-43). Other servers behave correctly, [others fails in a different way](https://github.com/open-quantum-safe/oqs-provider/issues/399#issuecomment-2085300308).

I think it would be useful to know more details about how fwupd server is configured and why it fails in this case. This could help the oqsprovider project to decide whether to enable less sig algorithms by default, or perhaps expose less sig algorithms just to TLS.

iyanmv, 2024-05-06:
> > it would be useful to know more details about how fwupd server is configured
> 
> It's a standard AWS load balancer in front of gunicorn. Does that help at all?

Is the TLS terminated at the AWS load balancer? Or does gunicorn also handle TLS?

hughsie, 2024-05-06:
> Is the TLS terminated at the AWS load balancer

This one; and IIRC it's all lets-encrypt under the hood.

iyanmv, 2024-05-07:
Fyi it was an issue with the AWS TLS library and it was fixed here: https://github.com/aws/s2n-tls/pull/4544

closed 2024-05-07 by iyanmv