fwupd/fwupd #7217: About auto upload report behavior

vincent-tang1, 2024-05-08:
**Describe the question**
After update finish, type below command, system will ask if user want to upload reports automatically for future updates
```
vincent@vincent:~$ sudo fwupdmgr get-devices
...
Uploading firmware reports helps hardware vendors to quickly identify failing and successful updates on real devices.
Review and upload report now? (Requires internet connection) [Y|n]: Y
...
Proceed with upload? [Y|n]: Y
Successfully uploaded 1 report
Do you want to upload reports automatically for future updates? [y|N]: Y
```
If choice yes for automatical upload in future, there are two test cases as following :
1)  when update success, system will automatically update the 'success report'
```
vincent@vincent:~$ sudo fwupdmgr install
Selected device: FWA6172 Platform Management
0.      Cancel
1.      1.03.0000
2.      1.01.0000
3.      0.99.0000
4.      0.97.0000
Choose release [0-4]: 1
╔══════════════════════════════════════════════════════════════════════════════╗
║ Reinstall FWA6172 Platform Management to 1.03.0000?                          ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This unstable release fixes the following issues:                            ║
║                                                                              ║
║ • Fix bug for test 1                                                         ║
║ • Improve performance for test 1                                             ║
║                                                                              ║
║ Some new functionality has also been added:                                  ║
║                                                                              ║
║ • Add support for test 1                                                     ║
║                                                                              ║
║ FWA6172 Platform Management and all connected devices may not be usable      ║
║ while updating.                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Waiting…                 [***************************************] Less than one minute remaining…
Successfully installed firmware
Successfully uploaded 1 report
```

2) when update fail, system will not automatically update the 'fail report', need to manually upload it with "report-history" command
```
vincent@vincent:~$ sudo fwupdmgr install
Selected device: FWA6172 Platform Management
0.      Cancel
1.      1.03.0000
2.      1.01.0000
3.      0.99.0000
4.      0.97.0000
Choose release [0-4]: 1
╔══════════════════════════════════════════════════════════════════════════════╗
║ Reinstall FWA6172 Platform Management to 1.03.0000?                          ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This unstable release fixes the following issues:                            ║
║                                                                              ║
║ • Fix bug for test 1                                                         ║
║ • Improve performance for test 1                                             ║
║                                                                              ║
║ Some new functionality has also been added:                                  ║
║                                                                              ║
║ • Add support for test 1                                                     ║
║                                                                              ║
║ FWA6172 Platform Management and all connected devices may not be usable      ║
║ while updating.                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Writing…                 [***********                            ] Less than one minute remaining…
failed to parse https://10.234.147.1:443/redfish/v1/TaskService/Tasks/0: The URI /redfish/v1/TaskService/Tasks/0 was not found.
```

**From above, my question is :**
Seems like 'fail report' will not automatically upload, not sure if it is your expected behavior ?

**fwupd version information**
Please provide the version of the daemon and client if applicable to your current installation of `fwupd`.

```shell
vincent@vincent:~$ fwupdmgr  --version
compile   org.freedesktop.fwupd         1.9.16
compile   com.hughsie.libxmlb           0.3.17
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.8.0-31-generic
runtime   org.freedesktop.fwupd         1.9.16

```

hughsie, 2024-05-08:
> when update fail, system will not automatically update the 'fail report'

@superm1 I think fwupd is doing what we want here, but sanity check please.

superm1, 2024-05-08:
I actually think the proposed behavior makes more sense than what we do now.

We're never going to get failed reports that fail immediately if users weren't prompted to upload.

But this also brings up another (bigger) point.
Especially if we're going to be going bigger on opt in stuff for all device lists I don't think we should be uploading from the client after get-devices etc.

I think we should probably have a systemd service that batches the stuff from the history database and uploads reports for everything.

Maybe even make it part of fwupd-refesh.service?

As a major perk we can start to get data from GUI clients if they were opted in already to it.

vincent-tang1, 2024-05-09:
> As a major perk we can start to get data from GUI clients if they were opted in already to it.

from my understanding, your expected result is, user can enable auto-upload option through GUI or command line, then even 'fail report' also can auto upload if update fail, right ?

superm1, 2024-05-09:
> > As a major perk we can start to get data from GUI clients if they were opted in already to it.
> 
> from my understanding, your expected result is, user can enable auto-upload option through GUI or command line, then even 'fail report' also can auto upload if update fail, right ? 

Yes exactly.

closed 2024-05-10 by hughsie

superm1, 2025-01-23:
> Looking at the PR, it seems the change was to upload it automatically more frequently (when refreshing). But not for failed reports?

shouldn't both handle the same?

JohnAZoidberg, 2025-01-23:
> > Looking at the PR, it seems the change was to upload it automatically more frequently (when refreshing). But not for failed reports?
> 
> shouldn't both handle the same?

The original issue description here indicates that that is not or was not the case.

superm1, 2025-01-23:
Oh... I see what you mean.  Basically if there is a failure, the error is populated all the way back to the `fwupdmgr` client but there is no opportunity to upload that failure report.

So the solution here is that `fwupd-refresh` will upload it later on it's next refresh cycle within the next day.