fwupd/fwupd #7227: Build gets stuck if Vala is not installed

lucaswerkmeister, 2024-05-09:
**Describe the bug**
After I saw a request to build fwupd from source [on Mastodon](https://mastodon.social/@hughsie/112412001905663059) and tried it out, I noticed that the build can get into an undesirable state if a dependency is initially missing, requiring additional action even after the build dependency is installed.

**Steps to Reproduce**
(These are the steps I followed, but if you follow the same steps, I expect you might get a different result depending on your system.)
- Environment: Arch Linux, with an assortment of development-related packages installed but without Vala
- Clone https://github.com/fwupd/fwupd/
- Run `contrib/setup` as mentioned on Mastodon
  - Do not install build dependencies (the default)
  - Do not run tests locally before pushing (the default)
- Run `source venv/bin/activate` as advised by setup’s output
- Run `build-fwupd` as advised by activate’s output
  > Program vapigen found: NO
  > 
  > libfwupd/meson.build:159:8: ERROR: Program 'vapigen' not found or not executable
- Look up which package is supposed to supply `vapigen` (the [Arch manpage](https://man.archlinux.org/man/vapigen-0.56.1.en) shows the package in the sidebar), then install Vala with `sudo pacman -S vala`
- Run `build-fwupd` again

**Expected behavior**
The build resumes, and ideally completes successfully.

**Actual behavior**
The command instantly (0.006 s real time) fails (exit code 1) with:
```
ninja: Entering directory `/tmp/fwupd/venv/bin/../build'
ninja: error: loading 'build.ninja': No such file or directory
```

To make the build work again, I have to remove `venv/build/`; after that, `build-fwupd` mostly works. (Towards the end, the installation fails with `find: ‘/usr/libexec/fwupd/efi’: No such file or directory`, but I assume that’s unrelated.)

**fwupd version information**
Git commit 664e94bfca.

```shell
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.19
compile   com.hughsie.libxmlb           0.3.19
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.6
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libxmlb           0.3.19
runtime   org.freedesktop.Passim        0.1.8
runtime   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.8.9-arch1-1
runtime   org.freedesktop.fwupd         1.9.19
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): source under Arch, as above.

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
WARNING: This package has not been validated, it may not work properly.
Unsupported daemon version 1.9.19, client version is 1.9.10-805-g664e94bfca
```

Let me know if you want the output of some similar command anyway (e.g. using system `fwupdmgr` instead of the built-from-source one), but I suspect it’s not actually relevant for this issues.

</details>

**Additional questions**

- Operating system and version: Arch Linux
- Have you tried rebooting? no
- Is this a regression? not to my knowledge

lucaswerkmeister, 2024-05-09:
I’m guessing that [this build-venv.sh check(https://github.com/fwupd/fwupd/blob/664e94bfca60d840ea402a64c54b82dee7fc7160/contrib/build-venv.sh#L12-L15) is a bit too simplistic:

```sh
if [ ! -d ${BUILD} ]; then
        meson setup ${BUILD} --prefix=${DIST} -Dudevdir=${DIST} ${EXTRA_ARGS} $@
fi
ninja -C ${BUILD} install
```

Because it’s possible that the build dir exists but isn’t fully set up yet. But I don’t think I know enough Meson to say what it should be instead, sorry.

superm1, 2024-05-09:
Why did you say no to build deps?

I would expect that gets you the missing deps...

lucaswerkmeister, 2024-05-09:
> But I don’t think I know enough Meson to say what it should be instead, sorry.

Actually, scratch that. It’s probably pretty simple, isn’t it? Ninja tells us the file it’s looking for, after all.

```diff
diff --git a/contrib/build-venv.sh b/contrib/build-venv.sh
index 3f07f49d57..1676c47228 100755
--- a/contrib/build-venv.sh
+++ b/contrib/build-venv.sh
@@ -9,7 +9,7 @@ EXTRA_ARGS="-Dsystemd=disabled -Dlaunchd=disabled"
 if [ -d /opt/homebrew/opt/libarchive/lib/pkgconfig ]; then
         EXTRA_ARGS="${EXTRA_ARGS} -Dpkg_config_path=/opt/homebrew/opt/libarchive/lib/pkgconfig"
 fi
-if [ ! -d ${BUILD} ]; then
+if [ ! -d ${BUILD} ] || ! [ -e ${BUILD}/build.ninja ]; then
         meson setup ${BUILD} --prefix=${DIST} -Dudevdir=${DIST} ${EXTRA_ARGS} $@
 fi
 ninja -C ${BUILD} install
```

(Could even drop the first part of the condition and only check `-e ${BUILD}/build.ninja`, I suppose.)

Repeating the reproduction steps with that patch applied (after first uninstalling Vala, of course) seems to produce a better result.

---

> Why did you say no to build deps?

Because I don’t know what fwupd does then, and I prefer to know which packages I’ll have to uninstall afterwards :)

lucaswerkmeister, 2024-05-09:
> > Why did you say no to build deps?
>
> Because I don’t know what fwupd does then, and I prefer to know which packages I’ll have to uninstall afterwards :)

(I’ve had bad experiences with software auto-installing its own dependencies before, see [tweet 1](https://twitter.com/LucasWerkmeistr/status/718800589182930944) and [2](https://twitter.com/LucasWerkmeistr/status/718870350210146305).)

superm1, 2024-05-09:
It uses pacman to do it. You'll see it all logged

superm1, 2024-05-09:
Could you send that patch up as a PR?

lucaswerkmeister, 2024-05-09:
Sure!

lucaswerkmeister, 2024-05-09:
*runs `git commit -a`*

*lots of git hooks that I did not explicitly install god knows what (hopefully only in the venv), “this may take a few minutes”*

yeah this is exactly what I don’t like :) PR incoming once the hook either finishes installing or I lose patience and switch to `git commit -n` :P

lucaswerkmeister, 2024-05-09:
I lost patience and decided to leave the pre-commit linting to CI;) PR here: #7228

PR #7228 (2024-05-09): trivial: venv: check build.ninja presence
If Meson failed partway through, it’s possible that the build directory exists but build.ninja inside doesn’t, which would make Ninja fail immediately. Fixes #7227.

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit fe8fa956 (2024-05-09): trivial: venv: check build.ninja presence

If Meson failed partway through, it’s possible that the build directory
exists but build.ninja inside doesn’t, which would make Ninja fail
immediately. Fixes #7227.
closed 2024-05-09 by superm1