fwupd/fwupd #7257: Support installing larger firmware when running on low-memory devices

rcyeh, 2024-05-14:
**Is your feature request related to a problem? Please describe.**

In fu-engine-config.c, the default maximum archive size is the lesser of 2 GB and 25% of RAM.

https://github.com/fwupd/fwupd/blob/a86eb352f98f4a7707e74b3f5ef0b430b99dfe6e/src/fu-engine-config.c#L372

I would like to be able to install a larger firmware update package on a low-memory (4 GB RAM) device, without running out of memory. I do run out of memory when I configure `ArchiveSizeMax=3072000000`.

**Describe the solution you'd like**

Would it be possible to read and deliver the contents of the archive in chunks? Maybe with a memory-mapped file or some other technique so that the entire firmware update does not need to be in RAM at one time?

**Describe alternatives you've considered**

For larger-memory devices, I could send a pull request enabling larger-than-2-GB firmware to be installed, but that doesn't solve the problem for low-memory devices.

hughsie, 2024-05-14:
> Would it be possible to read and deliver the contents of the archive in chunks

Yes, this is exactly what the main branch implements. The only catch if you need to use uncompressed archives, with no mszip compression.

hughsie, 2024-05-14:
I've talked a bit with @superm1 about it, and hopefully in the next couple of weeks.