fwupd/fwupd #7258: requirement error in cab for org.freedesktop.fwupd prevents downgrade to resolve broken lvfs upgrade

ianbmacdonald, 2024-05-14:
**Describe the bug**
Lenovo continues to ship Intel ME updates via fwupdmgr that break external monitors (my case is dual 4K/60Hz via TB3 on a Dell WD19TB)

**Steps to Reproduce**
1) On an X1 Yoga G6, update to current 240.45.2411 from working 240.42.2235 (shipped).    Upgrade summary exerpt from fwupdmgr is below. The upgrade is successful, but introduces an Intel ME bug that prevents external monitors from working on boot.  

```
└─Intel Management Engine:
  │   Device ID:          23156eb4e14af860c0d4e04a401cc82b89298733
  │   Previous version:   240.42.2235
  │   Update State:       Success
  │   Last modified:      2024-05-14 13:24
  │   GUID:               c1b2be54-d7ed-4e24-a577-7c5f32bb7587
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad X1 Yoga Gen 6/ThinkPad X1 Carbon Gen 9 Corporate ME Update:
        New version:      240.45.2411
        Remote ID:        lvfs
        Release ID:       49973
        Summary:          Lenovo ThinkPad X1 Yoga Gen 6/X1 Carbon Gen 9 Corporate ME Firmware
        License:          Proprietary
        Size:             10.7 MB
        Created:          2023-08-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N32RG27W
        Vendor:           Lenovo
        Release Flags:    • Trusted metadata
        Description:      
        Intel Platform Update 2023.3 Product Version Maintenance Release
        
        Corporate Version 15.0.45.2411 (LVFS: 240.45.2411)
        
        Problem Fixes
        
        • Mitigated the following security vulnerabilities under issues.Please see fixed issues for details.
        Issues:           CVE-2022-38102
                          CVE-2022-29871
                          CVE-2022-36392
                          INTEL-TA-00783
        Checksum:         c564d6a33d7ce0322c6710037979b144d938cc113fa4d0a8c34a462b3efd5264
```
2) The downgrade does not work, likely due to unstated dependencies enforced in the UEFI installer, as the only downgrade option is to to 240.23.1706 (via fwupdmgr).   The previous version 240.42.2235 is not available. 

```
:~# fwupdmgr downgrade 23156eb4e14af860c0d4e04a401cc82b89298733
╔══════════════════════════════════════════════════════════════════════════════╗
║ Downgrade Intel Management Engine from 240.45.2411 to 240.23.1706?           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ • 0 Intel Platform Update 2021.1 Hotfix Release                              ║
║                                                                              ║
║ Version 15.0.23.1706 (LVFS: 240.23.1706)                                     ║
║                                                                              ║
║ Problem Fixes                                                                ║
║                                                                              ║
║ • Mitigated the following security vulnerabilities under issues.Please       ║
║ see fixed issues for details.                                                ║
║                                                                              ║
║ 20XY0027US must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y

... after UEFI script execution and reboot .. 
└─Intel Management Engine:
  │   Device ID:          23156eb4e14af860c0d4e04a401cc82b89298733
[... 496 lines cut ...]
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
└─UEFI Platform Key:
      Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
      Summary:            Lenovo Ltd. PK CA 2012
      Vendor:             Lenovo Ltd.
      GUID:               71599d14-9b31-5270-b3bd-74c494585820 ← UEFI\CRT_9AEF2123F4DE7C19AFABD909BB2C8CAC4411E07E
```

</details>

**Additional questions**

- Operating system and version: Ubuntu 24.04
- Have you tried rebooting? Yes, multiple execution, attempted execution on the AMT (shares version, but not firmware upgradable via fwupdmr),  execution without TB3 dock attached (bare laptop)
- Is this a regression? Notably, this happened with Intel ME previously, as part of a BIOS update, and we were able to revert the BIOS update.  Also on the Lenovo website, there is a newer version (15.0.47.2521) posted for Windows notes thunderbolt updates (https://download.lenovo.com/pccbbs/mobiles/n32rg30w.exe) , and the version we want to go back to is posted for linux (15.0.42.2235) posted for Linux, so perhaps they know about this issue that fwupdmgr dropped us into. 

Options we are considering, to try to avoid deploying Windows to fix this, are 
a) try to get the .cab out of the newer .exe update for windows, and apply using fwupdmgr (if that is possible)
b) try to downgrade the whole BIOS, possibly with a version that comes with an older ME firmware as well (some research required, might introduce new dependency issues)

ianbmacdonald, 2024-05-14:
Inside the .exe I can see these contents;  I am thinking I am interested in TglMe_15_0_47_2521_Lp_Corp.cap  however, a cab file probably has some meta, that is missing from the direct capsule, if I were to guess. 

CLI seems to confirm the missing magic, which I assume is where the broken fwupd dependency is defined. 

```
$ sudo fwupdmgr local-install TglMe_15_0_47_2521_Lp_Corp.cap 23156eb4e14af860c0d4e04a401cc82b89298733
Decompressing…           [   -                                   ]
did not find magic
```

```
 innoextract n32rg30w.exe 
Extracting "version 15.0.47.2521 (n32rg30w)" - setup data version 5.5.7 (unicode)
 - "code$GetExtractPath$/FWCapUpdate_v46.exe" - overwritten
 - "code$GetExtractPath$/config.ini"
 - "code$GetExtractPath$/csme-version-detection-tool-console.exe"
 - "code$GetExtractPath$/FwCapUpdate_v46.exe"
 - "code$GetExtractPath$/FWDetect.exe"
 - "code$GetExtractPath$/fwdetect_Config.ini"
 - "code$GetExtractPath$/MEFirmwareSKUCheckTool.exe"
 - "code$GetExtractPath$/MEInfoWin64.exe"
 - "code$GetExtractPath$/SLA_TOOLS.PDF"
 - "code$GetExtractPath$/SusBde.bat"
 - "code$GetExtractPath$/Capsule/Consumer/TglMe_15_0_47_2521_Lp_Cons.cap"
 - "code$GetExtractPath$/Capsule/Consumer/tglme_15_0_47_2521_lp_cons.cat"
 - "code$GetExtractPath$/Capsule/Consumer/TglMe_15_0_47_2521_Lp_Cons.inf"
 - "code$GetExtractPath$/Capsule/Corporate/TglMe_15_0_47_2521_Lp_Corp.cap"
 - "code$GetExtractPath$/Capsule/Corporate/tglme_15_0_47_2521_lp_corp.cat"
 - "code$GetExtractPath$/Capsule/Corporate/TglMe_15_0_47_2521_Lp_Corp.inf"
 - "code$GetExtractPath$/misc/StartFWUpdate_OLD.CMD"
 - "code$GetExtractPath$/misc/Non_Capsule/Config.ini"
 - "code$GetExtractPath$/misc/Non_Capsule/FWDetect.exe"
 - "code$GetExtractPath$/misc/Non_Capsule/fwdetect_Config.ini"
 - "code$GetExtractPath$/misc/Non_Capsule/FWUpdate_v41.exe"
 - "code$GetExtractPath$/misc/Non_Capsule/FWUpdLcl64.exe"
 - "code$GetExtractPath$/misc/Non_Capsule/MEInfoWin64.exe"
 - "code$GetExtractPath$/misc/Non_Capsule/ME_15.0_Cons.bin"
 - "code$GetExtractPath$/misc/Non_Capsule/ME_15.0_Corp.bin"
 - "code$GetExtractPath$/misc/Non_Capsule/Non-Capsule_Update_ReadME.txt"
 - "code$GetExtractPath$/misc/Non_Capsule/SusBde.bat"
 - "code$GetExtractPath$/misc/TGL_FWdetect/csme-version-detection-tool-console.exe"
 - "code$GetExtractPath$/misc/TGL_FWdetect/FWDetect.exe"
 - "code$GetExtractPath$/misc/TGL_FWdetect/fwdetect_Config.ini"
 - "code$GetExtractPath$/misc/TGL_FWdetect/HowToUse.txt"
 - "code$GetExtractPath$/misc/TGL_FWdetect/MEInfoWin64.exe"
Done.

```

hughsie, 2024-05-14:
You can use `sudo fwupdtool install-blob foo.cap` -- but note that without the metadata it's entirely possible to flash the wrong firmware to the wrong device -- so be careful!

ianbmacdonald, 2024-05-14:
Okay, installing the blob directly worked great.  I know from the fwupdmgr release note breadcrumbs during the breaking upgrade that my system uses the Corporate ME engine, as opposed to the Consumer, so my confidence on the right capsule was high. 

```
fwupdtool install-blob TglMe_15_0_47_2521_Lp_Corp.cap 
Writing…                 [************                           ]
0.	Cancel
1.	0867c07f040e810fa5ea153981e63e324c29eb1f (ELAN0672:00 04F3:3187)
2.	073624c16dd99abe01ba1da223a70321e4f29beb (Package level of Dell dock)
3.	0acceea54e71c4d8002593822afe0f4705616613 (RTS5413 in Dell dock)
4.	96969b82d977fdcbf607df084ef0dcf10119409b (RTS5487 in Dell dock)
5.	228ac6846b64d532dd1ea1b7e651e8f5f55fbc34 (VMM5331 in Dell dock)
6.	570867f4ddda9915445da59bd85cd0e0e507270e (WD19TB)
7.	34210774738cce62e77e32d3a08a71619df371ba (Embedded Controller)
8.	3fa281ddf80d8a06b8ee5d8beb48d38ee95a9627 (Integrated Camera)
9.	23156eb4e14af860c0d4e04a401cc82b89298733 (Intel Management Engine)
10.	0d5d05911800242bb1f35287012cdcbd9b381148 (Prometheus)
11.	73e606488fec47b3e3f9288094f66fded0051446 (Prometheus IOTA Config)
12.	04e17fcf7d3de91da49a163ffe4907855c3648be (SSD 970 EVO Plus 500GB)
13.	61725c7e17fb355013585de4fd31fbe8ae919444 (System Firmware)
14.	4e767d355b374fa5269b32568d1758736b567a6e (Thunderbolt controller in Dell dock)
15.	d3201cead9d5fc62712ce7258f0658db9d3f2540 (UEFI Device Firmware)
16.	f905a7bd8805851b14be474d006372f37cc9342d (UEFI Device Firmware)
17.	f3df5955811d72e6ae18957a6260043a3f03ebdb (UEFI Device Firmware)
18.	0ac6a08a1e51ee11e3657f3a672eb808c0c2e0e8 (UEFI Device Firmware)
19.	411c6f23d9326317948c7efca60068d31eb53b21 (UEFI Device Firmware)
20.	eabaaa6ba3756a3590d823e2a49a6a2986eb4312 (UEFI Device Firmware)
21.	7e6427fb78fdb31553c3589b14cf9a40bd2e1f2e (UEFI Device Firmware)
22.	587c6425a53c7997d7d62a4fb5986fad193f8433 (UEFI Device Firmware)
23.	7dccda75b1f76a97e675b08e7e0f8ab9924ad9ce (UEFI Device Firmware)
24.	81568164782039dbd4a74856cacb9800f408d44e (UEFI Device Firmware)
25.	6ad5f455fbfa33903c16a954c2bbed10dc37244a (UEFI Device Firmware)
26.	97a1ebd5e0b64abc18f9a94d3051bfbda7bff6d7 (UEFI Device Firmware)
27.	362301da643102b9f38477387e2193e57abaa590 (UEFI dbx)
Choose device [0-27]: 9
Waiting…                 [***************************************]
An update requires a reboot to complete. Restart now? [y|N]: y
```
The reboot followed by the long UEFI update process, so it definately got the AMT component as well. 

In hindsight, a safer option would have been to use cabextract and force the blob in, working around why I originally filed this bug;  Rather than to push forward to a version not yet in the online database. 

But it worked, and it solved the broken external monitor enumeration on boot, which is great.  Below is my new enumerated firmware, noting my *lid is closed* since my monitors are working on boot now. 
```
fwupdmgr get-devices --show-all-devices
LENOVO 20XY0027US
│
├─ELAN0672:00 04F3:3187:
│     Device ID:          0867c07f040e810fa5ea153981e63e324c29eb1f
│     Summary:            Touchpad
│     Current version:    0x0003
│     Bootloader Version: 0x0002
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              ab16bce2-e219-52f5-89a6-fd6b1c4244c0 ← HIDRAW\VEN_04F3&DEV_3187
│                         e4c6161e-2b1c-59cb-ad07-c8fb7feb155e ← HIDRAW\VEN_04F3&DEV_3187&MOD_0002
│                         5b73c06e-3dfd-5e6c-b06a-24e3489206c2 ← ELANTP\ICTYPE_14&MOD_0002
│                         31a158de-a671-50bf-b787-3bba647a8c49 ← ELANTP\ICTYPE_14&MOD_0002&DRIVER_HID
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─WD19TB:
│ │   Device ID:          570867f4ddda9915445da59bd85cd0e0e507270e
│ │   Summary:            High performance dock
│ │   Current version:    01.01.00.07
│ │   Vendor:             Dell Inc. (USB:0x413C)
│ │   Install Duration:   1 minute
│ │   Serial Number:      H4HRVV2/3192058705161460
│ │   GUID:               cd357cf1-40b2-5d87-b8df-bb2dd82774aa ← USB\VID_413C&PID_B06E&hub&embedded
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Device stages updates
[... 443 lines cut ...]
│                         • Updatable
│     Device Requests:    • Message
│   
└─UEFI Platform Key:
      Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
      Summary:            Lenovo Ltd. PK CA 2012
      Vendor:             Lenovo Ltd.
      GUID:               71599d14-9b31-5270-b3bd-74c494585820 ← UEFI\CRT_9AEF2123F4DE7C19AFABD909BB2C8CAC4411E07E
```

Following up on the original intent, to make fwupdmgr follow published instructions for older versions of the tool, like those included in the readme that comes along with the Lenovo cab file for manual installation,  I did go and extract the cab, and look at the meta. 

```
 cabextract n32rk04w.cab 
Extracting cabinet: n32rk04w.cab
  extracting firmware.bin
  extracting firmware.metainfo.xml

All done, no errors.
```
There is no specification for the fwupd version in the firmware.metainfo.xml, so the dependency, it would seem on fwupd <= 1.1.0 stems from newer fwupd versions when other things are defined in the meta, likely the version, given the output "firmware with version" which is clearly not an issue with older fwupd included in Ubuntu LTS 20.04. 

It seems logical, another --allow-xxxx should be added for this new constraint to allow for compatability with existing documentation and to allow manual updates with older .cab files, that were packaged before the change to fwupd logic, to be applied without soo much effort. 

Lenovo for their part, releasing fixes for Windows without making simple tweaks to their Linux equivalent cab could be seen as just lazy, but the issue is more likely a set of missing QA procedures to catch this problem.  I'll do my part to file a ticket with them, letting them know they can safely update and release a new cab supporting Ubuntu 24.04.

hughsie, 2024-05-14:
> to allow manual updates with older .cab

I'm confused -- what's wrong with installing older files with a new fwupd?

ianbmacdonald, 2024-05-14:
It causes the `requirement 'firmware with version' also needs org.freedesktop.fwupd >= 1.1.0` error.  Presumably the same shipped meta (below) works on Ubuntu 20.04 as described in Lenovo's readme https://download.lenovo.com/pccbbs/mobiles/n32rk04w.txt 

But not in 24.04, where meta with a version defined, spits out this failure with no obvious way to override and get the previous behavior.    Assuming I have not made an error, it seems like a problem for any existing cab files before the change. 
```
cat firmware.metainfo.xml 
<?xml version="1.0" encoding="UTF-8"?>
<!-- Copyright 2016 Richard Hughes  -->
<component type="firmware">
  <id>com.lenovo.ThinkPadN32RM.firmware</id>
  <name>ThinkPad X1 Yoga Gen 6/ThinkPad X1 Carbon Gen 9</name>
  <summary>Lenovo ThinkPad X1 Yoga Gen 6/X1 Carbon Gen 9 Corporate ME Firmware</summary>
  <description>
    <p>
      Lenovo ThinkPad X1 Yoga Gen 6/X1 Carbon Gen 9 Corporate ME Firmware N32RG26W, 15.0.42.2235
    </p>
  </description>
  <provides>
    <firmware type="flashed">c1b2be54-d7ed-4e24-a577-7c5f32bb7587</firmware>
  </provides>
  <url type="homepage">http://www.lenovo.com/</url>
  <metadata_license>CC0-1.0</metadata_license>
  <project_license>proprietary</project_license>
  <developer_name>Lenovo Ltd.</developer_name>
  <categories>
     <category>X-CorporateManagementEngine</category>
     <category>X-ManagementEngine</category>
  </categories>
  <custom>
     <value key="LVFS::UpdateProtocol">org.uefi.capsule</value>
     <value key="LVFS::VersionFormat">triplet</value>
  </custom>
  <releases>
    <release version="240.42.2235" date="2022-09-21" urgency="high" tag="N32RG26W">
      <description>
        <p>Intel Platform Update 2022.3 Product Version Maintenance Release</p>
        <p>Version 15.0.42.2235 (LVFS: 240.42.2235)</p>
        <p>Problem Fixes</p>
           <ul>
              <li>   Mitigated the following security vulnerabilities under issues. Please see fixed issues for details.</li>
           </ul>
      </description>
      <issues>
        <issue type="intel">INTEL-TA-00610</issue>
        <issue type="cve">CVE-2022-21181</issue>
        <issue type="cve">CVE-2022-27497</issue>
        <issue type="cve">CVE-2022-29893</issue>
        <issue type="cve">CVE-2021-33159</issue>
        <issue type="cve">CVE-2022-29515</issue>
      </issues>
    </release>
  </releases>
  <requires>
       <firmware compare="ge" version="0.1.49">14f3350e-cf63-4e68-a0d9-0af1d5389a17</firmware>
       <firmware compare="ge" version="0.1.30">61b65ccc-0116-4b62-80ed-ec5f089ae523</firmware>
  </requires>

```

hughsie, 2024-05-15:
I think this lets us detect this on the LVFS: https://gitlab.com/fwupd/lvfs-website/-/commit/81d5e5258ddd393fbdb34b0d90e49fef92522b1c -- deploying now.

hughsie, 2024-05-15:
If you re-download https://fwupd.org/downloads/8f3899045a2d6024c6aee493632e53335b601bcea3fd149d61db9f345227eff1-Lenovo-ThinkPad-X1YogaGen6X1CarbonGen9-CorporateMEFirmware-15.0.42.2235.cab does it install locally now?

mrhpearson, 2024-05-16:
Hi - Lenovo premiere support pointed me at this (I assume this got escalated to them?)

Seems to me there are two problems - can I confirm my understanding:

1. There is an issue with the 240.45.2411 ME FW that causes problems with your monitor (or is it with the Dell dock?). Is this ticket raised anywhere or not? (apologies if I've missed it)
2. fwdupd not allowing downgrades. Looks like this is understood and potentially fixed. @hughsie - holler if you need anything from me.

If I've missed anything important let me know.
Mark

closed 2024-11-11 by hughsie