fwupd/fwupd #7260: Package fwupd is not configured yet

estatistics, 2024-05-14:
`Summary:
  Upgrading: 0, Installing: 0, Removing: 0, Not Upgrading: 195
  3 not fully installed or removed.
  Download size: 0 B / 330 kB
  Space needed: 0 B / 1830 GB available

Continue? [Y/n] y
Setting up fwupd (1.9.19-1) ...
+ [ configure = configure ]
+ systemd-sysusers fwupd.conf
Creating group 'fwupd-refresh' with GID 994.
Creating user 'fwupd-refresh' (Firmware update daemon) with UID 994 and GID 994.
dpkg: error processing package fwupd (--configure):
 installed fwupd package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of fwupd-tests:
 fwupd-tests depends on fwupd; however:
  Package fwupd is not configured yet.

dpkg: error processing package fwupd-tests (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 fwupd
 fwupd-tests
Error: Sub-process /usr/bin/dpkg returned an error code (1)
eros@eros:/mnt/Projects/home/eros/Downloads/_linux/vokoscreenNG/src$ sudo apt upgrade

 sudo apt install fwupd
fwupd is already the newest version (1.9.19-1).
Summary:
  Upgrading: 0, Installing: 0, Removing: 0, Not Upgrading: 195
  3 not fully installed or removed.
  Download size: 0 B / 330 kB
  Space needed: 0 B / 1830 GB available

Continue? [Y/n] y
Setting up fwupd (1.9.19-1) ...
+ [ configure = configure ]
+ systemd-sysusers fwupd.conf
Creating group 'fwupd-refresh' with GID 994.
Creating user 'fwupd-refresh' (Firmware update daemon) with UID 994 and GID 994.
dpkg: error processing package fwupd (--configure):
 installed fwupd package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of fwupd-tests:
 fwupd-tests depends on fwupd; however:
  Package fwupd is not configured yet.

dpkg: error processing package fwupd-tests (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 fwupd
 fwupd-tests
Error: Sub-process /usr/bin/dpkg returned an error code (1)

sudo dpkg --configure -a
Setting up fwupd (1.9.19-1) ...
+ [ configure = configure ]
+ systemd-sysusers fwupd.conf
Creating group 'fwupd-refresh' with GID 994.
Creating user 'fwupd-refresh' (Firmware update daemon) with UID 994 and GID 994.
dpkg: error processing package fwupd (--configure):
 installed fwupd package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of fwupd-tests:
 fwupd-tests depends on fwupd; however:
  Package fwupd is not configured yet.

dpkg: error processing package fwupd-tests (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 fwupd
 fwupd-tests
[... 102 lines cut ...]
?     GUIDs:              9305de1c-1e12-5665-81c4-37f8e51219b8 ? TPM\VEN_AMD&DEV_0001
?                         78a291ae-b499-5b0f-8f1d-74e1fefd0b1c ? TPM\VEN_AMD&MOD_AMD
?                         65a3fced-b423-563f-8098-bf5c329fc063 ? TPM\VEN_AMD&DEV_0001&VER_2.0
?                         5e704f0d-83cb-5364-8384-f46d725a23b8 ? TPM\VEN_AMD&MOD_AMD&VER_2.0
?     Device Flags:       ? Internal device
?                         ? System requires external power source
?                         ? Needs a reboot after installation
?                         ? Device can recover flash failures
?                         ? Full disk encryption secrets may be invalidated when updating
?                         ? Signed Payload
?   
??WDC WD50NDZW-11BCSS0:
      Device ID:          d05265ff31433b8099efe5515ddfa5e26639faf9
      Summary:            ATA drive
      Current version:    01.01A01
      Vendor:             Western Digital (ATA:0x101C, OUI:0014ee)
      Serial Number:      WD-WX82D32P5J98
      GUIDs:              eac7baa8-8da8-56b1-a7d9-851cd6d99263 ? IDE\WDC_WD50NDZW-11BCSS0____________________01.01A01
                          512196cc-e302-5b07-8726-334d91ad4326 ? IDE\0WDC_WD50NDZW-11BCSS0____________________
                          11fbdbcb-5a74-5b7c-924d-8a2c2f2a9140 ? WDC WD50NDZW-11BCSS0
      Device Flags:       ? Updatable
                          ? System requires external power source
                          ? Needs a reboot after installation
    
`

estatistics, 2024-05-14:
sudo fwupdmgr update 
Devices with no available firmware updates: 
 ? Intenso SSD SATAIII
 ? HDWD110
 ? WDC WD50NDZW-11BCSS0
Devices with the latest available firmware version:
 ? Integrated Webcam?

hughsie, 2024-05-14:
I think you need to file an issue with Ubuntu or Debian -- this is for upstream issues I'm afraid.

closed 2024-05-14 by hughsie

estatistics, 2024-05-14:
i removed completely the package and upgrade completed. thanks