fwupd/fwupd #7277: Possible regression on issue #7045

minnoce, 2024-05-22:
I'm experiencing the same problem described in the issue #7045.
No problem occured before the 8 May, when Fedora updates upgraded the fwupd package from version 1.9.16 to 1.9.19.
After the upgrade, at every start of fwupd.service, the journalctl expose these errors:
```
fwupd[95567]: 07:08:53.584 FuPluginUefiCapsule  failed to mount ESP: GDBus.Error:org.freedesktop.DBus.Error.UnknownMethod: No such interface "org.freedesktop.UDisks2.Filesystem" on object at path /org/freedesktop/UDisks2/block_devices/md127
fwupd[95567]: 07:08:53.834 FuPluginIntelMe      failed to get public key using /fpf/OemCred: generic failure [0xb]

```
Furthermore, after the upgrade, trying to update System firmware with `fwupdmgr update` fails with this error:
`Failed to create '/home/EFI/fedora/fw': Read-only file system`

fwupd version:
```
compile   org.freedesktop.fwupd         1.9.19
compile   com.hughsie.libxmlb           0.3.19
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.6
compile   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libxmlb           0.3.19
runtime   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd         1.9.19
runtime   org.kernel                    6.8.9-200.fc39.x86_64
```

Package installed with `dnf`.

<details>

<summary>**fwupd device information**</summary>

```
Dell Inc. Precision Tower 3620
│
├─Ellesmere [Radeon Pro WX 5100]:
│ │   Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│ │   Summary:            C95417 Polaris10 GL PRO A1 GDDR5 256Mx32 8GB 300e/300m
│ │   Current version:    015.050.000.001.011213
│ │   Vendor:             Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002, PCI:0x8086)
│ │   GUID:               4ae93667-48c3-5ba2-aeb1-1e79292b5423 ← AMD\113-C9541711-101
│ │   Device Flags:       • Cryptographic hash verification is available
│ │
│ ├─Unknown Device:
│ │     Device ID:        5f7dee69f5c6dc34f6af5a8ecff65b51ce478670
│ │     Serial Number:    UK02045045503
│ │     GUID:             885bd48b-2fad-533b-b639-e1d25b9e2705 ← DRM\VEN_PHL&DEV_0943
│ │
│ ├─Unknown Device:
│ │     Device ID:        a6c25fea64794a797f39d32a3277659be669963d
│ │     Serial Number:    5Y196901NB
│ │     GUID:             062c1c51-dde4-58c5-93dc-485644c5e1d2 ← DRM\VEN_NEC&DEV_68D6
│ │
│ └─Unknown Device:
│       Device ID:        270cc74de90fd845bd7956e61af39c8188856abf
│       Serial Number:    5Y196874NB
│       GUID:             062c1c51-dde4-58c5-93dc-485644c5e1d2 ← DRM\VEN_NEC&DEV_68D6
│
├─KXG50ZNV256G NVMe TOSHIBA 256GB:
│     Device ID:          03281da317dccd2b18de2bd1cc70a782df40ed7e
│     Summary:            NVM Express solid state drive
│     Current version:    AADA4105
│     Vendor:             Toshiba Corporation (NVME:0x1179)
│     Serial Number:      78BB635EKAUP
│     GUIDs:              4d0aed03-a30c-52c6-99e7-a8977797c3d9 ← NVME\VEN_1179&DEV_0116
│                         23e442d8-2af4-5c60-81f5-ac41aaee72c7 ← NVME\VEN_1179&DEV_0116&SUBSYS_11790001
│                         e67bb8f0-4e33-53f7-9f43-249a203b67fb ← KXG50ZNV256G NVMe TOSHIBA 256GB
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
[... 107 lines cut ...]
│     GUIDs:              9fe86f59-f248-5c53-98a5-f0c330423f37 ← IDE\WDC_WD2002FFSX-68PF8N0__________________81.00A81
│                         32d3e093-847b-59c9-9425-16aaef4af918 ← IDE\0WDC_WD2002FFSX-68PF8N0__________________
│                         1370d29b-d6a3-52c5-bd60-9910b8e97fc8 ← WDC WD2002FFSX-68PF8N0
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
└─Xeon CPU E3-1245 v5 @ 3.50GHz:
      Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
      Current version:    0x000000f0
      Vendor:             Intel
      GUIDs:              215d01de-fd1a-57cf-9153-72e29c0b7ffe ← CPUID\PRO_0&FAM_06&MOD_5E
                          c944e5cd-430c-5731-a9d5-c1b63c61cf27 ← CPUID\PRO_0&FAM_06&MOD_5E&STP_3
      Device Flags:       • Internal device
```

</details>

**Additional questions**

- Operating system and version: Fedora 39 fully updated with a RAID setup.
- Have you tried rebooting? Yes
- Is this a regression? Probably

hughsie, 2024-05-22:
> /home/EFI/fedora/fw

This looks very wrong -- how did you install Fedora 39? Can you share the output of `udisksctl dump` please

hughsie, 2024-05-22:
And the log from `sudo fwupdtool --plugins uefi_capsule -vv get-devices &> log.txt` please.

superm1, 2024-05-22:
>  @superm1 i guess we could add the fs type in the "more than one ESP possible" logic.

IMO not if `fu_volume_get_partition_kind()` is returning the wrong kind to start.

> I think some of your partitions are of the wrong type, e.g.

Yup I agree.  `md125` and `md126` should be fixed.  I don't believe we have any action for fwupd.

minnoce, 2024-05-22:
But... `fwupd` worked well on my exact same partitions up to v. 1.9.16...

superm1, 2024-05-22:
Yeah it's basically because of the fix to https://github.com/fwupd/fwupd/issues/7045 (https://github.com/fwupd/fwupd/commit/13a44f03e) that this happened.  We started to allow people to have ESP's within MD arrays, but if you have a busted MD array type then you're caught in the cross fire.

IDK what we can really do.  We either support MD arrays or we don't.

minnoce, 2024-05-22:
My RAID setup was created using Fedora's standard partition creation and configuration procedure...

superm1, 2024-05-22:
Test PR: https://github.com/fwupd/fwupd/pull/7282.  Once you can confirm that helps we'll merge that.

minnoce, 2024-05-24:
I tried building a new RPM from your PR this way:
- started from `fwupd-1.9.19-1.fc39.src.rpm`
- prepared a patch file with your changes on file `fu-uefi-capsule-plugin.c`
- try to rebuild the package with rpmbuild

I got an error at the last test stage on `fwupd-self-test`
I attach the rpmbuild log.
[testlog.txt](https://github.com/fwupd/fwupd/files/15436209/testlog.txt)

superm1, 2024-05-24:
You can just run it from fwupd development environment instead of a full rpm.

But for rpm issue just skip test stage perhaps?

@hughsie ideas?

minnoce, 2024-05-24:
I managed to build the modified RPM package skipping test stage and installed on my system.
Although the error messages remain in the logs, I was finally able to execute the command `fwupdmgr update` for applying firmware updates.
Thank you very match!

closed 2024-05-24 by superm1

hughsie, 2024-05-24:
Could be a locale issue:

	   Checksum:             SHA1(checksum)
	   Tags:                 tag
	   License:              license
	-  Size:                 1.2 kB
	+  Size:                 1,2 kB
	   Created:              1970-01-01
	   Uri:                  location
	   Homepage:             homepage

PR #7282 (2024-05-24): trivial: catch miscategorized ESP partitions
Fedora 38 appears to have set up an ESP within an md array that is btrfs.  This isn't possible for a BIOS to use.

Reject this combination.

Fixes: #7277

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit b42ad1d4 (2024-05-24): trivial: catch miscategorized ESP partitions

Fedora 38 appears to have set up an ESP within an md array that is
btrfs.  This isn't possible for a BIOS to use.

Reject this combination.

Fixes: #7277
commit ee7a94e2 (2024-05-24): trivial: catch miscategorized ESP partitions

Fedora 38 appears to have set up an ESP within an md array that is
btrfs.  This isn't possible for a BIOS to use.

Reject this combination.

Fixes: #7277
(cherry picked from commit b42ad1d450ae022e8b98709e92947c5c3a535b38)