fwupd/fwupd #7278: Can't active fwupd daemon

sam412081go, 2024-05-22:
**Describe the bug**
I'm trying to use fwupdmgr, but it shows :
`'fwupdmgr: symbol lookup error: /lib/x86_64-linux-gnu/libfwupdplugin.so.5: undefined symbol: g_usb_interface_get_subclass, version LIBGUSB_0.1.0'. `

Then I used systemctl status to check fwupd.service, and it shows as failed :
```
● fwupd.service - Firmware update daemon
     Loaded: loaded (/lib/systemd/system/fwupd.service; static; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2024-05-22 17:48:58 CST; 11min ago
       Docs: https://fwupd.org/
    Process: 113845 ExecStart=/usr/libexec/fwupd/fwupd (code=exited, status=127)
   Main PID: 113845 (code=exited, status=127)

 五  22 17:48:58 OA-109048 systemd[1]: Starting Firmware update daemon...
 五  22 17:48:58 OA-109048 fwupd[113845]: /usr/libexec/fwupd/fwupd: symbol lookup error: /lib/x86_64-linux-gnu/libfwupdplugin.so.5: undefined symbol: g_usb_interface_get_subclass, version LIBGUSB_0.1.0
 五  22 17:48:58 OA-109048 systemd[1]: fwupd.service: Main process exited, code=exited, status=127/n/a
 五  22 17:48:58 OA-109048 systemd[1]: fwupd.service: Failed with result 'exit-code'.
 五  22 17:48:58 OA-109048 systemd[1]: Failed to start Firmware update daemon.
```

**Steps to Reproduce**
Just execute "fwupdmgr" command

**Expected behavior**
A clear and concise description of what you expected to happen.
Just operate normally.

**fwupd version information**
Please provide the version of the daemon and client.
My fwupd daemon can't not start, so I can't check the version.

```shell
fwupdmgr --version
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

```
sudo apt update
sudo apt upgrade
sudo apt install --reinstall fwupd libgusb2

sudo apt install --reinstall libfwupdplugin5 libgusb2

sudo systemctl daemon-reload
sudo systemctl restart fwupd.service

```

I use 'journalctl -xe' to check error log:
```
五  22 18:17:51 OA-109048 systemd[1]: Starting Firmware update daemon...
-- Subject: A start job for unit fwupd.service has begun execution
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- A start job for unit fwupd.service has begun execution.
-- 
-- The job identifier is 24003.
 五  22 18:17:51 OA-109048 fwupd[114215]: /usr/libexec/fwupd/fwupd: symbol lookup error: /lib/x86_64-linux-gnu/libfwupdplugin.so.5: undefined symbol: g_usb_interface_get_subclass, version LIBGUSB_0.1>
 五  22 18:17:51 OA-109048 systemd[1]: fwupd.service: Main process exited, code=exited, status=127/n/a
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- An ExecStart= process belonging to unit fwupd.service has exited.
-- 
-- The process' exit code is 'exited' and its exit status is 127.
 五  22 18:17:51 OA-109048 systemd[1]: fwupd.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- The unit fwupd.service has entered the 'failed' state with result 'exit-code'.
 五  22 18:17:51 OA-109048 systemd[1]: Failed to start Firmware update daemon.
-- Subject: A start job for unit fwupd.service has failed
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- A start job for unit fwupd.service has finished with a failure.
-- 
-- The job identifier is 24003 and the job result is failed.
 五  22 18:17:51 OA-109048 sudo[114212]: pam_unix(sudo:session): session closed for user root
```

**Additional questions**

- Operating system and version:
`Linux OA-109048 5.15.0-102-generic #112~20.04.1-Ubuntu SMP Thu Mar 14 14:28:24 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux`
- Have you tried rebooting?
Yes
- Is this a regression?
No

superm1, 2024-05-22:
If I was to guess it sounds like a library installed from somewhere that's not the distro.

Where did your libgusb come from?

superm1, 2024-05-22:
```
apt policy libgusb2
apt policy fwupd
```

Should clarify.

sam412081go, 2024-05-23:
@superm1 
The fwupd & libgusb2 information as below:
```
sdrf@OA-109048:~$ apt policy libgusb2
libgusb2:
  Installed: 0.3.4-0.1
  Candidate: 0.3.4-0.1
  Version table:
 *** 0.3.4-0.1 500
        500 http://tw.archive.ubuntu.com/ubuntu focal/main amd64 Packages
        100 /var/lib/dpkg/status
sdrf@OA-109048:~$ apt policy fwupd
fwupd:
  Installed: 1.7.9-1~20.04.3
  Candidate: 1.7.9-1~20.04.3
  Version table:
 *** 1.7.9-1~20.04.3 500
        500 http://tw.archive.ubuntu.com/ubuntu focal-updates/main amd64 Packages
        100 /var/lib/dpkg/status
     1.3.9-4ubuntu0.1 500
        500 http://security.ubuntu.com/ubuntu focal-security/main amd64 Packages
     1.3.9-4 500
        500 http://tw.archive.ubuntu.com/ubuntu focal/main amd64 Packages

```

> Where did your libgusb come from?
Since I took over this PC later, I am not sure how libgusb installed previously.
But I have tried below command to reinstall it.
```
sudo apt update
sudo apt upgrade
sudo apt install --reinstall fwupd libgusb2

sudo apt install --reinstall libfwupdplugin5 libgusb2
```

superm1, 2024-05-23:
Is your /usr/local empty?

sam412081go, 2024-05-23:
> Is your /usr/local empty?

No, there have item inside
```
sdrf@OA-109048:/usr/local$ ls -l
total 40
drwxr-xr-x  3 root root 4096  九  12  2022 bin
drwxr-xr-x  2 root root 4096  八   1  2020 etc
drwxr-xr-x  2 root root 4096  八   1  2020 games
drwxr-xr-x  4 root root 4096  九  12  2022 include
drwxr-xr-x  5 root root 4096  六  17  2021 lib
drwxr-xr-x  3 root root 4096  六  17  2021 libexec
lrwxrwxrwx  1 root root    9  四   7  2021 man -> share/man
lrwxrwxrwx  1 root root   36  九   5  2022 mcuxpressoide -> /usr/local/mcuxpressoide-11.6.0_8187
drwxr-xr-x  6 root root 4096  九   5  2022 mcuxpressoide-11.6.0_8187
drwxr-xr-x  2 root root 4096  八   1  2020 sbin
drwxr-xr-x 15 root root 4096  八   2  2023 share
drwxr-xr-x  2 root root 4096  八   1  2020 src
```

superm1, 2024-05-23:
You should empty it and then rerun ldconfig.

sam412081go, 2024-05-23:
@superm1  It's work now, thank you!

I found my fwupd version is 1.7.9 on my PC, 
```
sdrf@OA-109048:~$ sudo fwupdtool --version
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.4
runtime   org.kernel                    5.15.0-107-generic
compile   org.freedesktop.gusb          0.3.4
compile   com.hughsie.libjcat           0.1.4
runtime   org.freedesktop.fwupd         1.7.9
```
and I try to update to newest version, because I want to use device emulation data related function by fwupdmgr.

I have tried to upgrade my fwupd, but it shows  `fwupd is already the newest version (1.7.9-1~20.04.3).`.
So my Ubuntu 20.04 doesn't support a higher version of fwupd? 
```
sdrf@OA-109048:~$ sudo apt upgrade fwupd
Reading package lists... Done
Building dependency tree       
Reading state information... Done
fwupd is already the newest version (1.7.9-1~20.04.3).
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  gir1.2-appindicator3-0.1 gir1.2-goa-1.0 libfwupdplugin1 libwireshark14 libwiretap11 libwsutil12 libxmlb1 linux-headers-5.15.0-72-generic
  linux-headers-5.15.0-75-generic linux-hwe-5.15-headers-5.15.0-72 linux-hwe-5.15-headers-5.15.0-75 linux-image-5.15.0-72-generic
  linux-image-5.15.0-75-generic linux-modules-5.15.0-72-generic linux-modules-5.15.0-75-generic linux-modules-extra-5.15.0-72-generic
  linux-modules-extra-5.15.0-75-generic shim
Use 'sudo apt autoremove' to remove them.
Get more security updates through Ubuntu Pro with 'esm-apps' enabled:
  libmagickwand-6.q16-6 libsdl2-dev libsdl2-2.0-0 libmagickcore-6.q16-6
  imagemagick-6-common
Learn more about Ubuntu Pro at https://ubuntu.com/pro
The following packages have been kept back:
  wireshark wireshark-common wireshark-qt
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
```

superm1, 2024-05-23:
You can try to build it in fwupd development environment but I suspect you'll have a hard time with dependencies on such old OS.

Maybe you're better off using a docker container or VM.

sam412081go, 2024-05-23:
I have built fwupd source code on my PC.
There have 'fwupdmgr' and 'fwupdtool' in my souce code folder /fwupd/build/src/'.
I use fwupdtool in fwupd  source code folder is show as below:
```
sdrf@OA-109048:~/fwupd_20240522/fwupd/build/src$ sudo ./fwupdtool --version
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.fwupd         2.0.0
runtime   com.hughsie.libxmlb           0.3.x
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    5.15.0-107-generic
**runtime   org.freedesktop.fwupd         2.0.0**
```
But I use fwupdmgr in fwupd  source code folder is show as below:
```
sdrf@OA-109048:~/fwupd_20240522/fwupd/build/src$ sudo ./fwupdmgr --version
**runtime   org.freedesktop.fwupd         1.7.9**
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.4
runtime   org.kernel                    5.15.0-107-generic
compile   com.hughsie.libjcat           0.1.4
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.4

```
Why are the fwupd version different?  
Does it look like fwupdmgr is detecting the version of fwupd installed on my system?
```
sdrf@OA-109048:~$ fwupdtool --version
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.4
runtime   org.kernel                    5.15.0-107-generic
compile   org.freedesktop.gusb          0.3.4
compile   com.hughsie.libjcat           0.1.4
runtime   org.freedesktop.fwupd         1.7.9
```

superm1, 2024-05-23:
You should use fwupd development environment instead. It sets up scripts and wrappers for launching properly.

sam412081go, 2024-05-23:
Sorry, do you have any reference articles or tutorials on setting up the fwupd development environment?

superm1, 2024-05-23:
https://fwupd.github.io/libfwupdplugin/building.html

sam412081go, 2024-05-23:
> https://fwupd.github.io/libfwupdplugin/building.html

Thank you!

sam412081go, 2024-05-23:
> https://fwupd.github.io/libfwupdplugin/building.html

I have followed the step of link to setup environment and build fwupd, and then I use 'fwupdmgr --version', the version of fwupd still 1.7.9,  but use 'fwupdtool --version' is 2.0.0. Is it correct?

```
(fwupd) sdrf@OA-109048:~/fwupd_20240523/fwupd$ fwupdtool --version
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
compile   org.freedesktop.fwupd         2.0.0
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libxmlb           0.3.x
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         2.0.0
runtime   org.kernel                    5.15.0-107-generic

(fwupd) sdrf@OA-109048:~/fwupd_20240523/fwupd$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.4
runtime   org.kernel                    5.15.0-107-generic
compile   com.hughsie.libjcat           0.1.4
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.4
```

Here are the steps I followed :
1.
```
cd ~
git clone https://github.com/fwupd/fwupd.git
cd fwupd
./contrib/setup
```
2.
`source venv/bin/activate`
Then I have the prefix (fwupd) , such as
`(fwupd) sdrf@OA-109048:~/fwupd_20240523/fwupd$ `
3.
`build-fwupd`
It seems all good for build, I can see the build fwupdtool and fwupdmgr in the 'venv/build/src' folder.

sam412081go, 2024-05-23:
I am trying to use docker to setup ubuntu 22.04 environment for fwupd.
Do you have reference of Dockerfile for docker build?