fwupd/fwupd #7280: HSI Security check assistance.

LocoTacoTruck, 2024-05-22:
**Describe the question**
I am trying to pass HSI-4 but im currently getting stuck at HSI-1, if anyone could give me advice it would be greatly appreciated.

**fwupd version information**
```
compile   org.freedesktop.fwupd         1.9.16
compile   com.hughsie.libxmlb           0.3.17
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.8.0-31-generic
runtime   org.freedesktop.fwupd         1.9.16
```

fwupdmgr security --force output
```
Host Security ID: HSI:0! (v1.9.16)

HSI-1
✔ BIOS firmware updates:         Enabled
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI bootservice variables:    Locked
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled
✘ Fused platform:                Unknown
✘ Supported CPU:                 Unknown

HSI-2
✔ IOMMU:                         Enabled
✔ TPM PCR0 reconstruction:       Valid
✘ SPI write protection:          Unknown
✘ Platform debugging:            Unknown

HSI-3
✔ CET Platform:                  Supported
✘ SPI replay protection:         Unknown
✘ Pre-boot DMA protection:       Disabled
✘ Suspend-to-idle:               Disabled
✘ Suspend-to-ram:                Enabled

HSI-4
✔ SMAP:                          Enabled
✘ Processor rollback protection: Unknown
✘ Encrypted RAM:                 Unknown

Runtime Suffix -!
✔ fwupd plugins:                 Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux kernel:                  Untainted
✘ CET OS Support:                Not supported
✘ Linux swap:                    Unencrypted
```

System Info
```
Ubuntu 24.04 LTS
64.0 GiB DDR4
ASUS ROG STRIX B550-F GAMING WIFI II
AMD Ryzen™ 7 5700X × 16
AMD Radeon™ RX 6950 XT
4.5 TB NVME
3607 Firmware version
64-bit
46 Gnone Version
Wayland windowing system
Linux 6.8.0-31-generic Linux kernel
```

superm1, 2024-05-22:
Currently fwupd requires the PSP to advertise some security measurements that it's not advertising for your CPU.

Iow this is expected behavior at the moment.

superm1, 2024-05-27:
Can you please try this series?

https://lore.kernel.org/linux-crypto/20240527152649.3722-1-mario.limonciello@amd.com/T/#t

superm1, 2024-06-03:
I'm going to close this here because there is no work from fwupd side to do.