fwupd/fwupd #7283: Getting "UEFI BIOS settings update pending reboot" after succesful update.

verpejas, 2024-05-22:
**Describe the bug**
Hi! I am getting the following error in fwupdmgr "UEFI BIOS settings update pending reboot" after successfully updating the bios from v1.25 to v1.26 on a ThinkPad T14 G2 AMD (model 20XK). At first,  fwupd did not want to work properly. I updated from a USB stick. Bios update succeeded but the above mentioned message still remained. I run OpenSUSE. I disabled secure boot and secure rollback - downgraded bios to 1.25 from USB - fwupd shows no errors. Updated bios from fwupd to 1.26 - it succeeded without issues, but the above mentioned message still remains in `fwupdmgr get-devices`

**Steps to Reproduce**
1. Check firmware info using `fwupdmgr get-devices`  - no errors shown on bios 1.25
2. On a Lenovo 20XK update bios from 1.25 to 1.26 either through USB, fwupd or from Windows.
3.  Check firmware info usin `fwupdmgr get-devices` - an error is shown: "Update Error:       UEFI BIOS settings update pending reboot"

**Expected behavior**
Just as with version 1.25, it should not show any errors as the bios has been updated succesfully and there does not apper to be any issues. Fwupd won't allow for future updates or downgrades due to this error being shown.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
verpejas@ThinkPad:~$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.20
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libxmlb           0.3.x
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.20
runtime   org.kernel                    6.9.1-1-default

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

Installed from OpenSUSE main repo:

```shell
verpejas@ThinkPad:~$ zypper info fwupd
Loading repository data...
Reading installed packages...

Information for package fwupd:
------------------------------
Repository     : Main Repository (OSS)
Name           : fwupd
Version        : 1.9.20-1.1
Arch           : x86_64
Vendor         : openSUSE
Installed Size : 6.4 MiB
Installed      : Yes (automatically)
Status         : up-to-date
Source package : fwupd-1.9.20-1.1.src
Upstream URL   : https://fwupd.org/
Summary        : Device firmware updater daemon

```

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
LENOVO 20XK0016US
│
├─AMD Ryzen 5 PRO 5650U with Radeon Graphics:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0a50000d
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              79759cdc-94db-5098-be7b-eb02521fbbec ← CPUID\PRO_0&FAM_19&MOD_50
│ │                       20b595b0-5892-5870-8e4c-688133ad6e34 ← CPUID\PRO_0&FAM_19&MOD_50&STP_0
│ │   Device Flags:       • Internal device
│ │ 
│ └─Secure Processor:
[... 923 lines cut ...]
│       ├── xzio.mod
│       └── zstd.mod
├── initrd -> initrd-6.9.1-1-default
├── initrd-6.8.9-1-default
├── initrd-6.9.1-1-default
├── sysctl.conf-6.8.9-1-default -> ../usr/lib/modules/6.8.9-1-default/sysctl.conf
├── sysctl.conf-6.9.1-1-default -> ../usr/lib/modules/6.9.1-1-default/sysctl.conf
├── System.map-6.8.9-1-default -> ../usr/lib/modules/6.8.9-1-default/System.map
├── System.map-6.9.1-1-default -> ../usr/lib/modules/6.9.1-1-default/System.map
├── vmlinuz -> vmlinuz-6.9.1-1-default
├── vmlinuz-6.8.9-1-default -> ../usr/lib/modules/6.8.9-1-default/vmlinuz
└── vmlinuz-6.9.1-1-default -> ../usr/lib/modules/6.9.1-1-default/vmlinuz

57 directories, 499 files

```

**Additional questions**

- Operating system and version: Opensuse Tumbleweed
- Have you tried rebooting?: Yes
- Is this a regression?
- Are you using an NVMe disk?: Yes. WD SN570 2TB
- Is secure boot enabled?: It wouldn't update with SB enabled, disbled it for the duration of update
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?: No

superm1, 2024-05-23:
This sounds like a BIOS bug.  There is a sysfs file that the think-lmi kernel module exports that fwupd reads.
https://github.com/torvalds/linux/blob/master/drivers/platform/x86/think-lmi.c#L1154

If you manually read that file I expect it's "1".

bdaase, 2024-07-04:
I had the same issue while trying to update via the `fwupdmgr` CLI. Interestingly, updating the P14s Gen2 AMD's firmware via GNOME Software worked just fine :shrug: