fwupd/fwupd #7284: Bug/Question: Ubuntu 24.04 mentions firmware has an entry in "Problems" - Cannot be used whilst lid is closed

luke-hill, 2024-05-23:
**Describe the question**
I updated from Ubuntu 20.04 to 24.04 (I now realise I maybe should have waited a few more months), and when running `fwupdate --get-devices` I see lots of entries (10+), with this message

```
Problems:           • Device cannot be used while the lid is closed
```

I have noticed that when it's in standby, I need to lift the lid, to then activate login. But then can close the lid again (I run my laptop with the lid shut).

**fwupd version information**
Please provide the version of the daemon and client if applicable to your current installation of `fwupd`.

```
compile   org.freedesktop.fwupd         1.9.16
compile   com.hughsie.libxmlb           0.3.17
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.8.0-31-generic
runtime   org.freedesktop.fwupd         1.9.16
```

hughsie, 2024-05-23:
What this means is "we can't update the firmware when the laptop lid is closed" -- @superm1 maybe we should tweak the problem text to be something like `Device cannot be updated while the lid is closed`

superm1, 2024-05-23:
> What this means is "we can't update the firmware when the laptop lid is closed" -- @superm1 maybe we should tweak the problem text to be something like `Device cannot be updated while the lid is closed`

Sounds good to me.

luke-hill, 2024-05-23:
@hughsie - Yes I can confirm I haven't tried that one. I did that and lo and behold it all worked and asked me to restart.

Sorry for the wasting of your time.

1million dollar question? Why is this needed? Or is this a "Don't ask, just do" situation?

closed 2024-05-23 by luke-hill

luke-hill, 2024-05-24:
Sorry I meant more like "Why does this not work on Ubuntu 24.04 when it does on 20.04"

hughsie, 2024-05-24:
We added it unconditionally to all Lenovo hardware as it was difficult to work out what models were affected: https://github.com/fwupd/fwupd/blob/main/plugins/uefi-capsule/uefi-capsule.quirk#L13