fwupd/fwupd #7287: SSL issues refreshing metadata

invidian, 2024-05-23:
**Describe the bug**

Seems refreshing metadata does not work via cdn.fwupd.org via cloudflare:
```
0 ✓ (88.4ms) 20:51:16 invidian@dellxps15mateusz ~ $ fwupdmgr refresh -v
(pkttyagent:45989): GLib-GIO-DEBUG: 20:51:32.003: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:45987): GLib-GIO-DEBUG: 20:51:32.011: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:45987): dconf-DEBUG: 20:51:32.011: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:45987): dconf-DEBUG: 20:51:32.011: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:45987): dconf-DEBUG: 20:51:32.011: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:45987): dconf-DEBUG: 20:51:32.011: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:45987): dconf-DEBUG: 20:51:32.011: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:45987): dconf-DEBUG: 20:51:32.011: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
(fwupdmgr:45987): dconf-DEBUG: 20:51:32.011: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
(fwupdmgr:45987): dconf-DEBUG: 20:51:32.011: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
(fwupdmgr:45987): dconf-DEBUG: 20:51:32.011: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
(fwupdmgr:45987): dconf-DEBUG: 20:51:32.011: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
(fwupdmgr:45987): GLib-GIO-DEBUG: 20:51:32.012: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:45987): dconf-DEBUG: 20:51:32.013: watch_established: "/system/proxy/" (establishing: 0)
(fwupdmgr:45987): dconf-DEBUG: 20:51:32.013: watch_established: "/system/proxy/http/" (establishing: 0)
(fwupdmgr:45987): dconf-DEBUG: 20:51:32.013: watch_established: "/system/proxy/https/" (establishing: 0)
(fwupdmgr:45987): dconf-DEBUG: 20:51:32.013: watch_established: "/system/proxy/ftp/" (establishing: 0)
(fwupdmgr:45987): dconf-DEBUG: 20:51:32.013: watch_established: "/system/proxy/socks/" (establishing: 0)
(fwupdmgr:45987): GLib-GIO-DEBUG: 20:51:32.014: _g_io_module_get_default: Found default implementation gvfs (GDaemonVfs) for ‘gio-vfs’
(fwupdmgr:45987): pxbackend-DEBUG: 20:51:32.014: px_config_sysconfig_set_config_file: Could not read file /etc/sysconfig/proxy
(fwupdmgr:45987): pxbackend-DEBUG: 20:51:32.014: Active config plugins:
(fwupdmgr:45987): pxbackend-DEBUG: 20:51:32.014:  - config-env
(fwupdmgr:45987): pxbackend-DEBUG: 20:51:32.014:  - config-xdp
(fwupdmgr:45987): pxbackend-DEBUG: 20:51:32.014:  - config-kde
(fwupdmgr:45987): pxbackend-DEBUG: 20:51:32.014:  - config-gnome
(fwupdmgr:45987): pxbackend-DEBUG: 20:51:32.014:  - config-sysconfig
(fwupdmgr:45987): GLib-GIO-DEBUG: 20:51:32.016: Failed to initialize portal (GNetworkMonitorPortal) for gio-network-monitor: Not using portals
(fwupdmgr:45987): GLib-GIO-DEBUG: 20:51:32.016: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:45987): GLib-GIO-DEBUG: 20:51:32.017: _g_io_module_get_default: Found default implementation networkmanager (GNetworkMonitorNM) for ‘gio-network-monitor’
(fwupdmgr:45987): pxbackend-DEBUG: 20:51:32.018: px_manager_constructed: Up and running
(fwupdmgr:45987): GLib-GIO-DEBUG: 20:51:32.018: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
(fwupdmgr:45987): Fwupd-DEBUG: 20:51:32.019: Emitting ::status-changed() [idle]
Updating lvfs
Fwupd-INFO: 20:51:32.025: downloading https://cdn.fwupd.org/downloads/firmware.xml.zst.jcat
(fwupdmgr:45987): pxbackend-DEBUG: 20:51:32.025: px_manager_get_proxies_sync: url=https://cdn.fwupd.org/downloads/firmware.xml.zst.jcat online=1
(fwupdmgr:45987): pxbackend-DEBUG: 20:51:32.025: px_manager_get_proxies_sync: Proxy[0] = direct://
(fwupdmgr:45987): Fwupd-DEBUG: 20:51:32.025: Emitting ::status-changed() [downloading]
(fwupdmgr:45987): Fwupd-DEBUG: 20:51:32.105: Emitting ::status-changed() [idle]
Downloading…             [   -                                   ]
Failed to download metadata for lvfs: failed to download file: OpenSSL/3.3.0: error:0A000410:SSL routines::ssl/tls alert handshake failure
1 ✗ (159ms) 20:51:32 invidian@dellxps15mateusz ~ $ curl -k https://cdn.fwupd.org/downloads/firmware.xml.zst.jcat
curl: (35) OpenSSL/3.3.0: error:0A000410:SSL routines::ssl/tls alert handshake failure
35 ✗ (137ms) 20:51:36 invidian@dellxps15mateusz ~ $ for i in $(getent ahostsv4 cdn.fwupd.org | grep RAW | awk '{print $1}') cdn.fwupd.org; do echo $i; echo | openssl s_client -connect $i:443; echo $?; done
188.114.96.4
Connecting to 188.114.96.4
CONNECTED(00000003)
Can't use SSL_get_servername
depth=2 C=IE, O=Baltimore, OU=CyberTrust, CN=Baltimore CyberTrust Root
verify return:1
depth=1 C=US, O=Cloudflare, Inc., CN=Cloudflare Inc ECC CA-3
verify return:1
depth=0 C=US, ST=California, L=San Francisco, O=Cloudflare, Inc., CN=sni.cloudflaressl.com
verify return:1
---
Certificate chain
 0 s:C=US, ST=California, L=San Francisco, O=Cloudflare, Inc., CN=sni.cloudflaressl.com
   i:C=US, O=Cloudflare, Inc., CN=Cloudflare Inc ECC CA-3
   a:PKEY: id-ecPublicKey, 256 (bit); sigalg: ecdsa-with-SHA256
   v:NotBefore: Jul  3 00:00:00 2023 GMT; NotAfter: Jul  2 23:59:59 2024 GMT
 1 s:C=US, O=Cloudflare, Inc., CN=Cloudflare Inc ECC CA-3
   i:C=IE, O=Baltimore, OU=CyberTrust, CN=Baltimore CyberTrust Root
   a:PKEY: id-ecPublicKey, 256 (bit); sigalg: RSA-SHA256
   v:NotBefore: Jan 27 12:48:08 2020 GMT; NotAfter: Dec 31 23:59:59 2024 GMT
---
Server certificate
[... 172 lines cut ...]
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.9.1-zen1-2-zen
runtime   org.freedesktop.fwupd         1.9.20

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**Additional questions**

- Operating system and version:
- Have you tried rebooting?
- Is this a regression?

hughsie, 2024-05-23:
Ya, we had a DNS issue; can you try again now please.

invidian, 2024-05-23:
Works fine now, thanks!

hughsie, 2024-05-23:
I guess the best way to find out how many people are really using something is to break it for a few minutes  :) Our apologies.

closed 2024-05-23 by hughsie