fwupd/fwupd #7293: Native arm64 Debian build support

jto6, 2024-05-24:
**Describe the bug**
When trying to build fwupd on Debian 12.5 on an Aarch64 SBC, many packages are missing.

**Steps to Reproduce**
Follow the standard build instructions:

$ clone repo...
$ ./contrib/setup --os=debian
$ source venv/bin/activate
$ build-fwupd

Note that I also have to specify the os or setup fails immediately.

**Expected behavior**
I expect all needed dependencies to be installed.

**Additional questions**

$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm

$ uname -a
Linux localhost 6.1.0-21-arm64 #1 SMP Debian 6.1.90-1 (2024-05-03) aarch64 GNU/Linux

superm1, 2024-05-25:
Yeah dependencies.xml in contrib/ci needs updates for aarch64 deps.

jvoisin, 2024-06-02:
I sent #7305 for fedora. Both building and testing are working.

closed 2024-11-11 by hughsie