fwupd/fwupd #7306: Mangled output when running `fwupdmgr get-plugins`

jvoisin, 2024-06-02:
**Describe the bug**

I'm trying to improve fwupd's support for [Asahi Linux](https://asahilinux.org/). In order to do so, I wanted to check what devices are currently recognized, by running `fwupdmgr get-plugins`. Unfortunately, this results in semi-garbage output. Interestingly, using `--json` works.

**Steps to Reproduce**

```console
$ git clone https://github.com/fwupd/fwupd && cd fwupd
$ ./contrib/setup
$ source venv/bin/activate
(fwupd) $ build-fwupd
(fwupd) $ fwupd &
(fwupd) $ fwupdmgr get-plugins
WARNING: UEFI capsule updates not available or enabled in firmware setup
See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
WARNING: This package has not been validated, it may not work properly.
  Flags:                • Disabled

  Flags:                • UEFI capsule updates not available or enabled in firmware setup

  Flags:                • Disabled
                        • Required hardware was not found
                        • Enabled if hardware matches
                        • Loaded from an external module
                        • Will measure elements of system integrity around an update

  Flags:                • Disabled
                        • Required hardware was not found

  Flags:                • Loaded from an external module

  Flags:                • Disabled
                        • Required hardware was not found

  Flags:                • Disabled
                        • Required hardware was not found
                        • Configuration is only readable by the system administrator

  Flags:                • Disabled
                        • Plugin is only for testing

  Flags:                • Disabled
                        • Plugin is only for testing

  Flags:                • Disabled

  Flags:                • Disabled

  Flags:                • Disabled
                        • Will measure elements of system integrity around an update

  Flags:                • Disabled

  Flags:                • Enabled if hardware matches

(fwupd) $ fwupdmgr get-plugins  --json | head -n64
WARNING: UEFI capsule updates not available or enabled in firmware setup
See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
WARNING: This package has not been validated, it may not work properly.
{
  "Plugins" : [
    {
      "Name" : "acpi_phat",
      "Flags" : [
        "disabled"
      ]
    },
    {
      "Name" : "algoltek_aux",
      "Flags" : [
        "ready"
      ]
    },
    {
      "Name" : "algoltek_usb",
      "Flags" : [
        "ready"
      ]
    },
    {
      "Name" : "amd_gpu",
      "Flags" : [
        "ready"
      ]
    },
    {
      "Name" : "analogix",
      "Flags" : [
        "ready"
      ]
    },
    {
      "Name" : "android_boot",
      "Flags" : [
        "ready"
      ]
    },
    {
      "Name" : "ata",
      "Flags" : [
        "ready"
      ]
    },
    {
      "Name" : "audio_s5gen2",
      "Flags" : [
        "ready"
      ]
    },
    {
      "Name" : "aver_hid",
      "Flags" : [
        "ready"
      ]
    },
    {
      "Name" : "bcm57xx",
      "Flags" : [
        "ready"
      ]
    },
    {
      "Name" : "bios",
(fwupd) $
```

**Expected behavior**

Proper display of the plugins list

**fwupd version information**
Please provide the version of the daemon and client.

```console
(fwupd) $ fwupdmgr --version
runtime   org.freedesktop.fwupd         2.0.0
runtime   com.hughsie.libxmlb           0.3.19
compile   com.hughsie.libxmlb           0.3.19
runtime   org.freedesktop.fwupd-efi     1.6
runtime   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.gusb          0.4.9
compile   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.8.10-400.asahi.fc40.aarch64+16k
compile   org.freedesktop.Passim        0.1.7
runtime   org.freedesktop.gusb          0.4.9
compile   org.freedesktop.fwupd         2.0.0

(fwupd) $
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

Installed from source, via a git clone.

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```console
$ fwupdmgr get-devices --show-all-devices
WARNING: UEFI capsule updates not available or enabled in firmware setup
See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
WARNING: This package has not been validated, it may not work properly.
apple Apple MacBook Air (13-inch, M2, 2022)
│
├─Unknown Device:
│     Device ID:          ec27a242e2363e96b1f306489e9c78c1e41e5817
│     GUID:               534e2aa4-29d0-5089-8a5c-61707de3bd39 ← GPIO\ID_23c100000.pinctrl
│   
├─Unknown Device:
│     Device ID:          babd101560f4248dfef6f93085e825443c7b28e2
│     GUID:               6ddc9ee6-8ac3-5323-bf65-2713658b0065 ← GPIO\ID_23d1f0000.pinctrl
│   
├─Unknown Device:
│     Device ID:          1b6e2ba096c0123c89125564a294bc61b6489cce
│     GUID:               cac130b2-2524-509e-b752-488a591dacef ← GPIO\ID_macsmc-pmu-gpio
│   
├─Unknown Device:
│     Device ID:          613b7e4b186fc5bae228d8f3095503c7219f92f7
│     GUID:               364955e9-a4ba-5d89-a827-774ed2ffc84e ← GPIO\ID_23e820000.pinctrl
│   
├─Unknown Device:
│     Device ID:          ad6ae671d43f317a9b9b7d6033e95eee3dc1e47b
│     GUID:               b5ee50d2-8d4d-5848-aeb9-0847d6eb005c ← GPIO\ID_24a820000.pinctrl
│   
└─nvram:
      Device ID:          bff8acd15001e1e2d38c264f05eff7d76b84aed3
      Summary:            Memory Technology Device
      Vendor:             DMI:apple
      GUIDs:              567bb6ba-53e1-5aba-a22f-0a69e749dcab ← MTD\NAME_nvram
                          c208f809-59f9-52d6-890f-6307d2a0ca25 ← MTD\VENDOR_apple&NAME_nvram
                          bfb9efaa-d0cc-55b7-9fb4-703b7d5feeb9 ← MTD\VENDOR_apple&PRODUCT_Apple-MacBook-Air-(13-inch,-M2,-2022)&NAME_nvram
      Device Flags:       • Internal device
                          • Updatable
                          • Needs a reboot after installation
                          • Cryptographic hash verification is available
    
(fwupd) $
```

</details>

**Additional questions**

- Operating system and version: Asahi Linux 6.8.10-400.asahi.fc40.aarch64+16k #1 SMP PREEMPT_DYNAMIC Mon May 27 16:05:41 UTC 2024 aarch64 GNU/Linux
- Have you tried rebooting? Yup
- Is this a regression? Not that I'm aware of.

superm1, 2024-06-03:
I've posted a PR that should help this.

jvoisin, 2024-06-03:
For the record, the PR is #7307

closed 2024-06-03 by superm1

jvoisin, 2024-06-03:
I confirm that it's indeed fixed here, thanks :)