fwupd/fwupd #7315: Flashrom plugin not always loaded for coreboot

Sean-StarLabs, 2024-06-04:
The flashrom plugin isn't being loaded for all of our models; I can't figure out why:

StarBook Mk VI:
```
$ sudo fwupdtool hwids
Loading…                 [*                                      ]
Computer Information
--------------------
BiosVendor: coreboot
BiosVersion: 24.05
BiosMajorRelease: 24
BiosMinorRelease: 5
FirmwareMajorRelease: 18
FirmwareMinorRelease: 04
Manufacturer: Star Labs
Family: B62-I
ProductName: StarBook
ProductSku: B62-I
EnclosureKind: 9
BaseboardManufacturer: Star Labs
BaseboardProduct: StarBook
Hardware IDs
------------
{88591b2d-0877-57e4-9d7b-81e64f821f8d}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{6aeccad7-f1fb-5727-9268-439f0edd36f7}   <- Manufacturer + Family + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{51cb29ca-a6e9-5cb5-8b7b-4458c7dd3260}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{01bccd1f-2b27-5bd6-a8b8-1a0011aae1d0}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
{f2c28a2b-dd15-516b-9b73-35b9813e05e0}   <- Manufacturer + Family + ProductName + ProductSku
{5df15b8d-1ae0-5dc8-93fb-08734611cde5}   <- Manufacturer + Family + ProductName
{5ad245a8-21c9-5f28-9afd-5d8208163d7c}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
{12122d1c-c383-5583-9cb7-3ba8d220913d}   <- Manufacturer + ProductSku
{debfa822-3db4-54e3-85eb-48988244bc3c}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{f919ce8e-f35e-5e5a-ac02-c4196ed84c17}   <- Manufacturer + ProductName
{5ad245a8-21c9-5f28-9afd-5d8208163d7c}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
{12122d1c-c383-5583-9cb7-3ba8d220913d}   <- Manufacturer + Family
{d17fd437-5ed8-5958-873e-100d620e3f16}   <- Manufacturer + EnclosureKind
{18559a31-6ed9-565c-b8fc-4f7498cde294}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{b31aa89d-cfb8-5c67-8502-ce76a9e7b580}   <- Manufacturer
```
```
09:56:13.371 FuContext            SMBIOS Manufacturer=Star Labs
09:56:13.371 FuContext            SMBIOS EnclosureKind=9
09:56:13.371 FuContext            SMBIOS Family=B62-I
09:56:13.371 FuContext            SMBIOS ProductName=StarBook
09:56:13.371 FuContext            SMBIOS ProductSku=B62-I
09:56:13.371 FuContext            SMBIOS BiosVendor=coreboot
09:56:13.371 FuContext            SMBIOS BiosVersion=24.05
09:56:13.371 FuContext            SMBIOS BiosMajorRelease=18
09:56:13.371 FuContext            SMBIOS BiosMinorRelease=05
09:56:13.371 FuContext            SMBIOS FirmwareMajorRelease=18
09:56:13.371 FuContext            SMBIOS FirmwareMinorRelease=04
09:56:13.371 FuContext            SMBIOS BaseboardManufacturer=Star Labs
09:56:13.371 FuContext            SMBIOS BaseboardProduct=StarBook
...
09:47:49.071 FuEngine             enabling flashrom due to HwId 12122d1c-c383-5583-9cb7-3ba8d220913d
```

Byte Mk II:
```
sean@sean-Byte:~$ sudo fwupdtool hwids
Loading…                 [*                                      ]
Computer Information
--------------------
BiosVendor: coreboot
BiosVersion: 24.04
BiosMajorRelease: 24
BiosMinorRelease: 2
FirmwareMajorRelease: 01
FirmwareMinorRelease: 0a
Manufacturer: Star Labs
Family: Y2
ProductName: Byte
ProductSku: Y2
EnclosureKind: 3
BaseboardManufacturer: Star Labs
BaseboardProduct: Byte
Hardware IDs
------------
{5d118d36-3ce9-5de9-871b-fa6e8f5c27b0}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{6b879d81-9b20-554a-8f15-05c7b94827bf}   <- Manufacturer + Family + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{03294160-7d1a-562d-b24c-e13559b46cef}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{61c18847-8214-5651-af39-75d690dedb41}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
{8099f16f-3273-565a-9b28-b50a551b4159}   <- Manufacturer + Family + ProductName + ProductSku
{8d0a8099-2c7a-55ac-98cf-1a09746b0d5d}   <- Manufacturer + Family + ProductName
{0b4f3029-8692-5898-b1fd-5be33b0ca585}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
{ef9b469e-2429-53e6-b4c0-3646ac0f53ed}   <- Manufacturer + ProductSku
{4da56f91-f856-5f4e-82be-433d221114e8}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{2e5d5438-5e13-59e3-a946-04f2ff48bda6}   <- Manufacturer + ProductName
{0b4f3029-8692-5898-b1fd-5be33b0ca585}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
{ef9b469e-2429-53e6-b4c0-3646ac0f53ed}   <- Manufacturer + Family
{f1088ab7-7369-52e6-8482-8fd5d40eba84}   <- Manufacturer + EnclosureKind
{4394bca5-7695-512c-b0ba-0c43f383209c}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{b31aa89d-cfb8-5c67-8502-ce76a9e7b580}   <- Manufacturer
```
```
09:54:33.559 FuContext            SMBIOS Manufacturer=Star Labs
09:54:33.559 FuContext            SMBIOS EnclosureKind=3
09:54:33.559 FuContext            SMBIOS Family=Y2
09:54:33.559 FuContext            SMBIOS ProductName=Byte
09:54:33.559 FuContext            SMBIOS ProductSku=Y2
09:54:33.559 FuContext            SMBIOS BiosVendor=coreboot
09:54:33.559 FuContext            SMBIOS BiosVersion=24.04
09:54:33.559 FuContext            SMBIOS BiosMajorRelease=18
09:54:33.559 FuContext            SMBIOS BiosMinorRelease=02
09:54:33.559 FuContext            SMBIOS FirmwareMajorRelease=01
09:54:33.559 FuContext            SMBIOS FirmwareMinorRelease=0a
09:54:33.559 FuContext            SMBIOS BaseboardManufacturer=Star Labs
09:54:33.559 FuContext            SMBIOS BaseboardProduct=Byte
...
09:54:33.572 FuEngine             disabling plugin because: failed to startup using flashrom: no HwIDs found

```

hughsie, 2024-06-04:
What fwupd version pls? And can you upload the full daemon -vv log pls.

Sean-StarLabs, 2024-06-04:
This one?
```
12:31:14.373 FuDebug              verbose to debug (on console 1)
12:31:14.373 FuEngine             starting fwupd 1.9.20…
12:31:14.373 FuMain               locked /run/lock/fwupdtool
Loading…                 [ -                                     ]12:31:14.396 FuConfig             loading config /etc/fwupd/fwupd.conf
12:31:14.396 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
12:31:14.396 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
12:31:14.396 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
12:31:14.396 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x35f
12:31:14.396 FuConfig             setting config [msr] MinimumSmeKernelVersion=5.18.0
12:31:14.396 FuConfig             setting config [redfish] IpmiDisableCreateUser=False
12:31:14.396 FuConfig             setting config [redfish] ManagerResetTimeout=1800
12:31:14.396 FuConfig             setting config [thunderbolt] MinimumKernelVersion=4.13.0
12:31:14.396 FuConfig             setting config [thunderbolt] DelayedActivation=false
12:31:14.396 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
12:31:14.396 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
12:31:14.396 FuConfig             ::configuration loaded
12:31:14.397 FuRemoteList         loading /usr/share/fwupd/metainfo
12:31:14.397 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
12:31:14.397 XbSilo               file: 0318f196-344d-080c-482e-bfcfec53bcff, current:0318f196-344d-080c-482e-bfcfec53bcff, cached: (null)
12:31:14.397 XbSilo               loading silo with file contents
12:31:14.397 FuRemoteList         loading remote from /var/lib/fwupd/remotes.d/starlabs-embargo.conf
12:31:14.398 FuRemoteList         auto-fixing remote starlabs-embargo MetadataURI from https://fwupd.org/downloads/firmware-a9c8ca80cc85a7e36f4dbe193cd70b69fd261550.xml.gz to https://fwupd.org/downloads/firmware-a9c8ca80cc85a7e36f4dbe193cd70b69fd261550.xml.zst
12:31:14.398 FuRemoteList         ::remote_list changed
12:31:14.398 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
12:31:14.398 FuRemoteList         ::remote_list changed
12:31:14.398 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
12:31:14.399 FuRemoteList         ::remote_list changed
12:31:14.399 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
12:31:14.399 FuRemoteList         ::remote_list changed
12:31:14.399 FuEngine             priority local and vendor-directory is not download remote, so bumping
12:31:14.399 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
12:31:14.399 FuRemoteList         ::remote_list changed
12:31:14.399 FuEngine             priority local and vendor is not download remote, so bumping
12:31:14.399 FuRemoteList         ordering starlabs-embargo=lvfs+1
12:31:14.399 FuRemoteList         ignoring unfound remote fwupd
12:31:14.399 FuRemoteList         ordering lvfs-testing=lvfs+1
12:31:14.399 FuRemoteList         ignoring unfound remote fwupd
12:31:14.399 FuRemoteList         enabled remotes: vendor-directory[1000], starlabs-embargo[1], lvfs[0]
Loading…                 [                                       ]12:31:14.399 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
12:31:14.399 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
12:31:14.400 Jcat                 loaded 1 certificates
12:31:14.400 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
12:31:14.400 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
12:31:14.400 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
12:31:14.400 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
12:31:14.400 Jcat                 loaded 1 certificates
12:31:14.400 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
12:31:14.400 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
12:31:14.400 FuEngine             client certificate now exists: nothing to do
Loading…                 [                                       ]12:31:14.400 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
12:31:14.401 FuHistory            got schema version of 12
Loading…                 [*                                      ]12:31:14.405 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-1.9.20/libfu_plugin_modem_manager.so)
12:31:14.407 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
12:31:14.407 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
12:31:14.408 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
12:31:14.408 XbSilo               file: 08bcbf90-7030-c709-fb58-691d801596dd, current:08bcbf90-7030-c709-fb58-691d801596dd, cached: (null)
12:31:14.408 XbSilo               loading silo with file contents
Loading…                 [*                                      ]12:31:14.410 FuConfig             loading config /etc/fwupd/fwupd.conf
12:31:14.410 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
12:31:14.410 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
[... cut ...]
  Enabled:              true
  DoneSetup:            true
  CanInvalidate:        false

12:31:14.503 FuEngine             resetting update motd timeout

No detected devices
12:31:14.505 FuMain               No detected devices

12:31:14.507 FuPlugin             finalize(flashrom)
12:31:14.507 FuPluginTest         destroy
12:31:14.507 FuProgress           raw timing data was { 0.003, 0.000, 0.000, 0.000 } -- steps were set as [ 1 1 3 95 ] but should have been [ 92 4 1 3 ] at ../libfwupdplugin/fu-context.c:986

```

hughsie, 2024-06-13:
> 1.9.20

Ohh! We only added the coreboot vendor override in the `main` branch. Can you test main, and if that works I think we need to backport bad36d48fe

Sean-StarLabs, 2024-06-13:
Ha, yeah, that's it - tested one building from source and another from a
tag!

closed 2024-11-11 by hughsie