fwupd/fwupd #7323: Wrong redirect with different hash while updating

hlavki, 2024-06-05:
**Describe the bug**
When running `fwupdmgr update` on HP Thunderbolt Dock G4 I get:
`checksum invalid, expected 9b19fbdf2fcd06b9ece2ffeb15f7708aec95db8f7e904d55c48d66f54829e52f got 46cca91f9f7ea9e39a342621c2f5c1d666e932d6e35310988860acb60c2251b9`

log:
```
Fwupd-INFO: 09:22:54.552: downloading https://fwupd.org/downloads/9b19fbdf2fcd06b9ece2ffeb15f7708aec95db8f7e904d55c48d66f54829e52f-Thunderbolt_Dock_G4_V1.5.22.0.cab
(fwupdmgr:7257): pxbackend-DEBUG: 09:22:54.552: px_manager_get_proxies_sync: url=https://fwupd.org/downloads/9b19fbdf2fcd06b9ece2ffeb15f7708aec95db8f7e904d55c48d66f54829e52f-Thunderbolt_Dock_G4_V1.5.22.0.cab online=1
(fwupdmgr:7257): pxbackend-DEBUG: 09:22:54.552: px_manager_get_proxies_sync: Proxy[0] = direct://
```

But when I put this url to browser I can see there is 302 redirect and Location header is:

```
https://r2.fwupd.org/lvfs-prod/46cca91f9f7ea9e39a342621c2f5c1d666e932d6e35310988860acb60c2251b9-Thunderbolt_Dock_G4_V1.5.22.0.cab
```

**Steps to Reproduce**
Run `fwupdmgr update`

**Expected behavior**
Device is updated

**fwupd version information**

```shell
fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.fwupd         1.9.20
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libxmlb           0.3.x
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.9.3-1-default
runtime   org.freedesktop.fwupd         1.9.20
```

zypper on Opensuse Tumbleweed

hughsie, 2024-06-05:
I'm not sure what's happened here; clearly it's a bit too fragile. I've merged https://gitlab.com/fwupd/lvfs-website/-/commit/3336331e7f00f2c5b178f0ae21aadd6f1495beb7 and will deploy it this afternoon.

hughsie, 2024-06-05:
I've deployed, can you try now please?

hlavki, 2024-06-05:
Thanks, now it works as expected.

closed 2024-06-05 by hlavki