fwupd/fwupd #7332: "local devices supported" number accumulated over multiple remotes, even if the same devices are matching

evgeni, 2024-06-09:
**Describe the bug**
On my Thinkpad X1C Gen 9 with a TBT3G2 Dock, I usually see 6/7 "local devices supported" when I run "fwupdmgr refresh" (it's actually 6, but 7 show up when there is an update, idk why).
When I enable both stable and testing, I get 14 (or 12, I guess) supported.

**Steps to Reproduce**
Enable `lvfs` only:
```
% fwupdmgr refresh --force
Updating lvfs
Downloading…             [  -                                    ]
Successfully downloaded new metadata: 6 local devices supported
```

Enable both `lvfs` and `lvfs-testing`:
```
% fwupdmgr refresh --force
Updating lvfs-testing
Downloading…             [  -                                    ]
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 14 local devices supported
```

**Expected behavior**
Only updateable devices are counted, regardless how many remotes the updates come from

**fwupd version information**

```shell
% fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.20
compile   com.hughsie.libxmlb           0.3.19
compile   org.freedesktop.Passim        0.1.7
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.6
compile   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libxmlb           0.3.19
runtime   org.freedesktop.Passim        0.1.7
runtime   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.8.10-300.fc40.x86_64
runtime   org.freedesktop.fwupd         1.9.20

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

Fedora 40 RPMs

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
% fwupdmgr get-devices --show-all-devices
LENOVO 20XXS2GP02
│
├─ELAN0672:00 04F3:3187:
│     Device ID:          0867c07f040e810fa5ea153981e63e324c29eb1f
│     Summary:            Touchpad
│     Current version:    0x0003
│     Bootloader Version: 0x0002
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              ab16bce2-e219-52f5-89a6-fd6b1c4244c0 ← HIDRAW\VEN_04F3&DEV_3187
│                         e4c6161e-2b1c-59cb-ad07-c8fb7feb155e ← HIDRAW\VEN_04F3&DEV_3187&MOD_0002
│                         5b73c06e-3dfd-5e6c-b06a-24e3489206c2 ← ELANTP\ICTYPE_14&MOD_0002
│                         31a158de-a671-50bf-b787-3bba647a8c49 ← ELANTP\ICTYPE_14&MOD_0002&DRIVER_HID
│     Device Flags:       • Internal device
│                         • Updatable
│   
[... 515 lines cut ...]
│     Update State:       Success
│     GUID:               d2718981-bb13-45b4-b919-16bd42529a70
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
└─UEFI Platform Key:
      Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
      Summary:            Lenovo Ltd. PK CA 2012
      Vendor:             Lenovo Ltd.
      GUID:               71599d14-9b31-5270-b3bd-74c494585820 ← UEFI\CRT_9AEF2123F4DE7C19AFABD909BB2C8CAC4411E07E
    

```

</details>

**Additional questions**

- Operating system and version: Fedora 40
- Have you tried rebooting? No
- Is this a regression? idk

hughsie, 2024-06-09:
What's the difference between `fwupdmgr get-devices --json` when lvfs-testing is enabled/disabled?

evgeni, 2024-06-09:
see yourself:
[devices.lvfs.json](https://github.com/user-attachments/files/15753710/devices.stable.json)
[devices.lvfs_and_testing.json](https://github.com/user-attachments/files/15753711/devices.testing.json)

evgeni, 2024-06-09:
Oh, I did expect the Dock to add a device or two. But there are also "USB Hubs" now with "latest updates". Interesting.

hughsie, 2024-06-10:
So fwupd is acting as designed?

evgeni, 2024-06-10:
Yeah, after comparing the data, it is.

Sorry for the noise!