fwupd/fwupd #7341: [fwupd_daemon.txt](https://github.com/user-attachments/files/15801393/fwupd_daemon.txt)

superm1, 2024-06-13:
[fwupd_daemon.txt](https://github.com/user-attachments/files/15801393/fwupd_daemon.txt)
[fwupdmgr.txt](https://github.com/user-attachments/files/15801397/fwupdmgr.txt)

@hughsie @superm1 
I follow the page [Setup fwupd development environment](https://fwupd.github.io/libfwupdplugin/building.html#setup-fwupd-development-environment) to set up environment to use fwupdmgr emulation device(one terminal tab for fwupd daemon, one terminal tab for fwupdmgr).

I use Device ID to tag my device,and it looks like the emulation-tag can detect my device, but emulation-save is still not working.
It still shows below  when I use emulation-save.
```
WARNING: This package has not been validated, it may not work properly.
failed to save emulation data: no emulation data, perhaps no devices have been added?
```
[image]

I use` script ./fwupd_daemon.txt `and `script ./fwupdmgr.txt `to record log history.
The attached "fwupd_daemon.txt" is log for fwupd daemon.
The attached "fwupdmgr.txt" is log for fwupdmgr  tool.

My fwupd repository is up-to-data as of today(7f8855ac617e9a5cac2ac21c680e2489edf33ad7)

_Originally posted by @sam412081go in https://github.com/fwupd/fwupd/pull/7330#issuecomment-2162498044_

hughsie, 2024-06-13:
@sam412081go I still see three devices:

    8:43:44.282 FuDeviceList         ::added 745fdf5b7ba2d86e9dcd525f1060d91b6a851ee8 [HP HP 65x/45x Series dongle]
    8:43:44.305 FuDeviceList         ::added 71d3042554d6d110f4268c92de0c117b53023512 [HP HP 65x/45x Series dongle]
    8:43:44.388 FuDeviceList         ::added 6e39378578f82f62f53a7d9b528d44ac304c053b [HP HP 65x/45x Series dongle]

Can you share your fix to ignore the other interfaces?

hughsie, 2024-08-16:
@sam412081go still valid?

sam412081go, 2024-08-17:
> @sam412081go still valid?
@hughsie 
I think the problem I encountered is 
`failed to save emulation data: no emulation data, perhaps no devices have been added?`

And as we mentioned earlier in :[7356#](https://github.com/fwupd/fwupd/pull/7356#issuecomment-2175186171)
because PixArt-RF uses hidraw, device emulation is not going to work.

Is there any update?  I can try again.

hughsie, 2024-11-11:
We've fixed this conclusively in the main branch -- can you give it a go again @sam412081go please.

closed 2025-05-03 by hughsie