fwupd/fwupd #7350: After an update failed it wouldn't try again until restarting the service

PureTryOut, 2024-06-15:
**Describe the bug**
Just now I tried to update the UEFI firmware. Due to <reasons> (which I resolved) the upgrade failed but fwupd wouldn't  try again. `fwupdmgr get-updates` would list the upgrade and had it marked as failed but running `fwupdmgr update` didn't retry the update. Only after I restarted the fwupd daemon (`rc-service fwupd restart`, this is on Alpine Linux) would `fwupdmgr update` retry the upgrade again (after which it succeeded luckily).

**Steps to Reproduce**
1. initiate an upgrade
2. somehow make sure it fails. In my case I didn't have the package providing `fwupdx64.efi` installed
3. run `fwupdmgr get-updates` and see the update is marked as failed
4. run `fwupdmgr update` again and see that it doesn't prompt to upgrade again

**Expected behavior**
The update to be retried after running `fwupdmgr update` another time.

**fwupd version information**

```
$ fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.19
compile   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.fwupd         1.9.21
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libxmlb           0.3.19
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.gusb          0.4.9
runtime   org.kernel                    6.6.33-0-lts
runtime   org.freedesktop.fwupd         1.9.21
```

I've installed it from the Alpine Linux repositories using apk.

<details>

<summary>**fwupd device information**</summary>

```shell
$ fwupdmgr get-devices --show-all-devices
WARNING: This package has not been validated, it may not work properly.
Framework Laptop 13 (AMD Ryzen 7040Series)
│
├─AMD Ryzen 7 7840U w/ Radeon 780M Graphics:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              2a1df9de-fe02-59c8-8af1-32c8bb011f94 ← CPUID\PRO_0&FAM_19&MOD_74
│ │                       5a4bca32-6547-584f-bb36-920d9adede25 ← CPUID\PRO_0&FAM_19&MOD_74&STP_1
│ │   Device Flags:       • Internal device
│ │ 
│ ├─Graphics Processing Unit (GPU):
│ │ │   Device ID:        83eba1fa14bdefee1f6b4bcf06069b298e1c338f
│ │ │   Summary:          AMD AMD_PHOENIX_GENERIC
│ │ │   Current version:  022.012.000.027.000001
│ │ │   Vendor:           Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│ │ │   GUID:             6c23af4f-d6cd-5ffc-a502-0b85e472e7cb ← AMD\113-PHXGENERIC-001
│ │ │   Device Flags:     • Internal device
│ │ │ 
│ │ └─NE135FBM-N41:
│ │       Device ID:      aec1a869eb0df71b7cea6b3ac71d39b830faf164
│ │       GUID:           ace3bb34-1bbb-5a5f-9694-73c4cede9692 ← DRM\VEN_BOE&DEV_0BCA
│ │       Device Flags:   • Internal device
│ │     
│ ├─Secure Processor:
│ │     Device ID:        c54ab0237d7a8db8c717b68e0be78e4374a2a079
│ │     Current version:  00.2d.00.7a
│ │     Bootloader Version:00.2d.00.7a
│ │     Vendor:           Advanced Micro Devices, Inc. (PCI:0x1022)
│ │     GUIDs:            9eb6a793-7f97-5fb5-b49c-139d2ce3ee46 ← PCI\VEN_1022&DEV_15C7
│ │                       2e4eb727-a7c8-5eba-80cd-553b8d796d80 ← PCI\VEN_1022&DEV_15C7&SUBSYS_F1110006
│ │     Device Flags:     • Internal device
│ │   
│ └─System Management Unit (SMU):
│       Device ID:        db0330716216c629bb2c07256e5d018f499eb6ce
│       Summary:          Microcontroller used within CPU/APU program 0
│       Current version:  76.82.0
│       Vendor:           Advanced Micro Devices, Inc.
│       GUID:             3ac3159e-6eef-5f6b-bc8d-67686b238747 ← /sys/devices/platform/AMDI0009:00
│       Device Flags:     • Internal device
│     
├─Fingerprint Sensor:
│     Device ID:          23ec719b6aabc2d2dac5176c232f0da7a21881b0
│     Summary:            Match-On-Chip fingerprint sensor
│     Current version:    01000248
│     Vendor:             Goodix (USB:0x27C6)
│     Install Duration:   10 seconds
│     Serial Number:      UIDD7156801_XXXX_MOC_B0
│     GUID:               1e8c8470-a49c-571a-82fd-19c9fa32b8c3 ← USB\VID_27C6&PID_609C
│     Device Flags:       • Updatable
│                         • Device stages updates
│                         • Device can recover flash failures
│                         • Signed Payload
│   
├─GPIO controller:
│     Device ID:          f685512aa07369c9e77742acef941d779d31e766
│     GUID:               37b440a9-2473-5087-a39b-db84f32a8ed8 ← GPIO\ID_AMDI0030:00
│   
├─System Firmware:
│ │   Device ID:          1e4fa9cea0b89e613370cf9657ffa0b6d3f37fec
│ │   Summary:            UEFI System Resource Table device (Updated via capsule-on-disk)
│ │   Current version:    0.0.3.5
│ │   Minimum Version:    0.0.0.1
│ │   Vendor:             Framework (DMI:INSYDE Corp.)
│ │   Update State:       Success
│ │   GUID:               b5f7dcc1-568c-50f8-a4dd-e39d1f93fda1
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │   Device Requests:    • Message
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  371
│       Minimum Version:  371
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│                         115f7cac-f705-5d34-9a47-37177c3e8514 ← UEFI\CRT_B38FAD316F525F27B27A21B486456C3E4279748BF16893827BF16FE659C0F75E&ARCH_X64
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Only version upgrades are allowed
│                         • Signed Payload
│     
├─UEFI Platform Key:
│     Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
│     Summary:            frame.work-LaptopAMDPK
│     GUID:               c701846d-5461-5c19-a7ae-2d89b2f44c14 ← UEFI\CRT_D610D8205624E644D9435383826B2DBF7298859C
│   
└─WD BLACK SN850X 1000GB:
      Device ID:          03281da317dccd2b18de2bd1cc70a782df40ed7e
      Summary:            NVM Express solid state drive
      Current version:    620331WD
      Vendor:             Sandisk Corp (NVME:0x15B7)
      Serial Number:      23283V801294
      GUIDs:              69749ddc-cd7a-542e-981f-620656c59918 ← NVME\VEN_15B7&DEV_5030
                          53d3c0f5-42b2-5cc9-8a52-fa350528dc02 ← NVME\VEN_15B7&DEV_5030&SUBSYS_15B75030
                          0680d54c-3981-5294-90fd-decf472ab2db ← WD_BLACK SN850X 1000GB
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```
</details>

**Additional questions**

- Operating system and version: Alpine Linux edge
- Have you tried rebooting: no, restarting the service ended up being enough
- Is this a regression: no clue

hughsie, 2024-06-15:
> Only after I restarted the fwupd daemon

I think it's expected that you'd need to either reboot or restart the daemon if you install fwupdx64.efi while fwupd is running.

PureTryOut, 2024-06-15:
Is it? :thinking: I'd think it would check for the presence of the file when it needs it, not on load. If it is you can close this issue though.

superm1, 2024-06-15:
Hmm, fwupdx64.efi shouldn't be needed on this system:

> │ │   Summary:            UEFI System Resource Table device (Updated via capsule-on-disk)

What was the error it told you?

PureTryOut, 2024-06-15:
That both `fwupdx64.efi` and `fwupdx64.efi.signed` couldn't be found. Only after installing `fwupd-efi` which provides the first file it worked.

superm1, 2024-06-15:
What version of fwupd did that message come from? Same version you put in the bug report? Or did you upgrade since then?

PureTryOut, 2024-06-15:
Yes same version as I put in the original bug report.

superm1, 2024-06-15:
I think the issue is that we call `fu_uefi_device_prepare()` for all backend types; but it's not needed for CoD.