fwupd/fwupd #7351: No ESP causes segmentation fault in fu_engine_load

istankovic, 2024-06-15:
**Describe the bug**
Running `fwupd` without an ESP results in SIGSEGV:
```
12:07:19.278 FuPluginUefiCapsule  cannot find default ESP: No ESP or BDP found                                                               
12:07:19.494 FuUsbDevice          failed to parse platform BOS descriptor: failed to parse: no supported platform version: did not find magic
12:07:20.367 FuVolume             fu_volume_locker: assertion 'FU_IS_VOLUME(self)' failed                                                    
                                                                                                                                             
Thread 1 "fwupd" received signal SIGSEGV, Segmentation fault.                                                                                
0x00007ffff7ca8ce5 in fu_engine_load () from /usr/lib/fwupd-1.9.21/libfwupdengine.so                                                         
(gdb) bt                                                                                                                                     
#0  0x00007ffff7ca8ce5 in fu_engine_load () from /usr/lib/fwupd-1.9.21/libfwupdengine.so                                                     
#1  0x0000555555561df4 in fu_daemon_setup ()                                                                                                 
#2  0x000055555555ad13 in main ()                                                                                                            
```

**Steps to Reproduce**
1. Make sure not to have a valid ESP available when `fwupd` starts
2. Run `fwupd`

**Expected behavior**
`fwupd` should exit with an error code, not terminate via SIGSEGV.

**fwupd version information**
```
compile   org.freedesktop.fwupd         1.9.21            
compile   com.hughsie.libxmlb           0.3.19            
compile   org.freedesktop.Passim        0.1.8             
compile   com.hughsie.libjcat           0.2.1             
runtime   org.freedesktop.fwupd-efi     1.6               
runtime   com.hughsie.libxmlb           0.3.19            
compile   org.freedesktop.gusb          0.4.9             
runtime   com.hughsie.libjcat           0.2.1             
runtime   org.freedesktop.gusb          0.4.9             
runtime   org.kernel                    6.9.4-arch1-1     
runtime   org.freedesktop.fwupd         1.9.21            
```

I'm on Archlinux and fwupd was installed regularly via `pacman`.

superm1, 2024-06-15:
Do you have something configured in your `fwupd.conf`?  Would you be able to get line numbers of the crash?  I don't see an obvious cause to this; we're generally checking the ESP is non-NULL.

closed 2024-06-15 by superm1

istankovic, 2024-07-07:
Just to add here that this problem is now fixed, tested with 1.9.21. Thanks!