fwupd/fwupd #7374: Lenovo Legion Y740 failing HSI tests (missing Intel MEI)

149segolte, 2024-06-21:
**Describe the bug**
While performing a system update, found out that the system was failing all HSI levels, and fwupd not detecting some UEFI firmware. `fwupdmgr security` shows MEI v0:12.0.71.1681. Intel's csme detection tool states vulnerable. But `fwupdmgr get-devices` does not show MEI.

<details>
  <summary>GNOME device security</summary>

```shell
Device Security Report
======================

Report details
  Date generated:                                  2024-06-21 12:37:50
  fwupd version:                                   1.9.21

System details
  Hardware model:                                  LENOVO 81UH
  Processor:                                       Intel(R) Core(TM) i7-9750H CPU @ 2.60GHz
  OS:                                              Fedora Linux 40 (Workstation Edition)
  Security level:                                  HSI:0! (v1.9.21)

HSI-1 Tests
  UEFI Platform Key:                               Pass (Valid)
  Intel Management Engine Version:               ! Fail (Not Valid)
  UEFI Bootservice Variables:                      Pass (Locked)
  TPM v2.0:                                        Pass (Found)
  Firmware BIOS Region:                            Pass (Locked)
  UEFI Secure Boot:                                Pass (Enabled)
  Firmware Write Protection Lock:                  Pass (Enabled)
  Platform Debugging:                              Pass (Not Enabled)
  Intel Management Engine Manufacturing Mode:      Pass (Locked)
  BIOS Firmware Updates:                           Pass (Enabled)
  Firmware Write Protection:                       Pass (Not Enabled)
  TPM Platform Configuration:                      Pass (Valid)
  Intel Management Engine Override:                Pass (Locked)

HSI-2 Tests
  Platform Debugging:                              Pass (Locked)
  Intel BootGuard ACM Protected:                 ! Fail (Not Valid)
  IOMMU Protection:                                Pass (Enabled)
  Intel BootGuard Fuse:                            Pass (Valid)
  Intel GDS Mitigation:                            Pass (Enabled)
  Intel BootGuard Verified Boot:                 ! Fail (Not Valid)
  TPM Reconstruction:                              Pass (Valid)
  Intel BootGuard:                                 Pass (Enabled)

HSI-3 Tests
  Pre-boot DMA Protection:                       ! Fail (Not Enabled)
  Intel BootGuard Error Policy:                  ! Fail (Not Valid)
  Control-flow Enforcement Technology:           ! Fail (Not Supported)
  Suspend To RAM:                                  Pass (Not Enabled)
  Suspend To Idle:                               ! Fail (Not Enabled)

HSI-4 Tests
  Encrypted RAM:                                 ! Fail (Not Supported)
  Supervisor Mode Access Prevention:               Pass (Enabled)

Runtime Tests
  Firmware Updater Verification:                   Pass (Not Tainted)
  Linux Swap:                                      Pass (Encrypted)
  Linux Kernel Verification:                     ! Fail (Tainted)
  Linux Kernel Lockdown:                           Pass (Enabled)

Host security events
  2024-06-18 19:49:49   Suspend To RAM               Pass (Enabled → Not Enabled)
  2024-05-11 15:22:53   Suspend To RAM             ! Fail (Not Enabled → Enabled)
  2024-05-11 09:14:51   Suspend To RAM               Pass (Enabled → Not Enabled)
  2024-05-07 11:27:23   Suspend To RAM             ! Fail (Not Enabled → Enabled)
  2024-05-07 09:06:38   Linux Kernel Verification  ! Fail (Not Tainted → Tainted)
  2024-05-01 09:58:54   Linux Kernel Verification    Pass (Tainted → Not Tainted)
  2024-05-01 08:22:06   Linux Kernel Verification  ! Fail (Not Tainted → Tainted)
[... 1255 lines cut ...]
│   └── entries  [error opening dir]
├── lost+found  [error opening dir]
├── symvers-6.8.10-300.fc40.x86_64.xz -> /lib/modules/6.8.10-300.fc40.x86_64/symvers.xz
├── symvers-6.8.11-300.fc40.x86_64.xz -> /lib/modules/6.8.11-300.fc40.x86_64/symvers.xz
├── symvers-6.9.4-200.fc40.x86_64.xz
├── System.map-6.8.10-300.fc40.x86_64
├── System.map-6.8.11-300.fc40.x86_64
├── System.map-6.9.4-200.fc40.x86_64
├── vmlinuz-0-rescue-01181f550fab432a8a609e9395d513f9
├── vmlinuz-6.8.10-300.fc40.x86_64
├── vmlinuz-6.8.11-300.fc40.x86_64
└── vmlinuz-6.9.4-200.fc40.x86_64

6 directories, 17 files

```

**Additional questions**

- Operating system and version: Fedora Workstation 40
- Have you tried rebooting? Yes
- Is this a regression? Don't know.
- Are you using an NVMe disk? Yes (as boot disk).
- Is secure boot enabled? Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No, It is a Lenovo device, but no setting named 'Boot Order Lock' in UEFI, still can manipulate the boot order using arrow keys.

149segolte, 2024-11-26:
No. This is a Coffee Lake mobile processor.

hughsie, 2025-05-04:
Does this work better with fwupd 2.0.x?