fwupd/fwupd #7377: IntelMei plugin is creating warning messages

CragW, 2024-06-21:
**Describe the bug**
Red error message is seen from get-devices.
`16:08:58.092 FuPluginIntelMe      failed to get public key using file-id 0x40002300: generic failure [0xb]` 

**Steps to Reproduce**
$ fwupdtool get-devices

**Expected behavior**
No warning message.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
$ fwupdtool --version
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
compile   org.freedesktop.fwupd         2.0.0
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libxmlb           0.3.x
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         2.0.0
runtime   org.kernel                    6.8.0-35-generic
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): source

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdtool get-devices --show-all-devices 2>&1
Loading…                 [****************                       ]16:08:57.767 FuEngine             no GUIDs for device AUX A/DDI A/PHY A [7c2dd454e6f29453896a4824e8234317d237f8c5]
Loading…                 [*********************                  ]16:08:58.092 FuPluginIntelMe      failed to get public key using file-id 0x40002300: generic failure [0xb]
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
Dell Inc. Latitude 7350
│
├─Unknown Device:
│     Device ID:          a68c39791960b3933e30fa803e3ad3bb813e8850
│     GUID:               e4d9b4f9-ae47-5fe3-9ef0-a27b093b8553 ← GPIO\ID_INTC1083:00
│   
├─BC901 NVMe SK hynix 512GB:
│     Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Summary:            NVM Express solid state drive
│     Current version:    51005151
│     Vendor:             SK hynix (NVME:0x1C5C)
│     Serial Number:      4YC5N014810305A0T
│     Problems:           • Device requires AC power to be connected
│     GUIDs:              e843baa5-1ac3-58ef-a1d5-da7c169d7a70 ← STORAGE-DELL-111426
│                         e16555f6-3161-11ec-8d3d-0242ac130003
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Device is usable for the duration of the update
│                         • Updatable
│                         • Signed Payload
│   
├─CoreTM Ultra 7 165U:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x0000001c
│     Vendor:             Intel
│     GUIDs:              e38fa480-0c3d-5240-be48-b590e507eed9 ← CPUID\PRO_0&FAM_06&MOD_AA
│                         189c1794-cc75-56bd-b1f7-8f4af3aa3f82 ← CPUID\PRO_0&FAM_06&MOD_AA&STP_4
[... 112 lines cut ...]
│     GUID:               5f7cc4ea-9f7a-5c5f-b535-dfbcc73f5ba6 ← UEFI\CRT_2493B118EABD632FD3BA49C333A8329B1982E872
│   
└─Unifying Receiver:
      Device ID:          2a00cc81fbe939006a9132911cc58b6f38960326
      Summary:            Miniaturised USB wireless receiver
      Current version:    RQR12.11_B0032
      Bootloader Version: BOT01.04_B0016
      Vendor:             Logitech, Inc. (HIDRAW:0x046D, USB:0x046D)
      Install Duration:   30 seconds
      GUID:               279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
      Device Flags:       • Updatable
                          • Signed Payload

```

</details>

Full log: 
[intel-mei-heci-file-missing.log](https://github.com/user-attachments/files/15930243/intel-mei-heci-file-missing.log)

**Additional questions**

- Operating system and version: Ubuntu 24.04
- Have you tried rebooting? Yes
- Is this a regression? Not sure.

CragW, 2024-06-21:
should we lower the error level to debug?

hughsie, 2024-06-21:
> should we lower the error level to debug?

No, we should fix the bug! :) e.g. see https://blogs.gnome.org/hughsie/2023/05/09/msi-and-insecure-kms/ for instructions

closed 2025-05-08 by superm1