fwupd/fwupd #7397: Texus Instrument tps65982 controller for Librem 5 not recognized in fwupd

CaptainMorgan12, 2024-06-28:
Environment: postmarketos edge, librem 5 evergreen.

Issue: device does not show up even though firmware is upgradable

See: https://www.ti.com/document-viewer/tps65982/datasheet

Please include in supported device list and provide firmware updates.

hughsie, 2024-06-29:
All the tps6598x devices we've seen use the `USB\VID_0451&PID_ACE1` ID -- so if it's visible by the kernel we should already be showing it in `fwupdmgr get-devices`. If that works, I think you need to ask Purism to upload firmware, they're the only ones with redistribution permission. If it's not visible in lsusb then Purism needs to tell us how to enable it -- there's not much we can do otherwise.

CaptainMorgan12, 2024-06-29:
Thanks hughsie I will check if it is visible by the kernel and by get-devices, and file a bug report if it is not with the kernel devs.

closed 2024-11-11 by hughsie