fwupd/fwupd #7402: Several error prints on get-devices

CragW, 2024-07-01:
**Describe the bug**
Run `fwupdtool get-devices`, seeing a bunch of error messages at the top.

Error message that has been tracking in #7376:
> 14:59:24.472 FuEngine             no GUIDs for device AUX A/DDI A/PHY A [7c2dd454e6f29453896a4824e8234317d237f8c5]

Error message that has been tracking in #7377:
> 14:59:24.783 FuPluginIntelMe      failed to get public key using file-id 0x40002300: generic failure [0xb]

New error, likely thunderbolt retimer requires a default unsigned flag:
> 14:59:24.492 FuEngine             thunderbolt [16d35c510618116e7899ae45e62605572872e713] does not declare signed/unsigned payload -- perhaps add fu_device_add_flag(device, FWUPD_DEVICE_FLAG_UNSIGNED_PAYLOAD);

New errors, likely meter lake related:
1
> 14:59:24.710 FuUdevDevice         Meteor Lake-P Serial IO I2C Controller [6cbdb3bced1db1ff556b5aa303f7d88410e1f6c9] forgot to call fu_udev_device_add_flag() with OPEN_READ and/or OPEN_WRITE

2
> 14:59:24.717 FuUdevDevice         Meteor Lake-P Serial IO I2C Controller [b7a9095c55d5e5fde72e6474656c4d1b1ac23bb4] forgot to call fu_udev_device_add_flag() with OPEN_READ and/or OPEN_WRITE

3
> 14:59:24.723 FuUdevDevice         Communication controller [5fed1486be004d67ea79838d2e83aaa11bb72645] forgot to call fu_udev_device_add_flag() with OPEN_READ and/or OPEN_WRITE

4
> 14:59:24.792 FuUdevDevice         Meteor Lake-P Serial IO SPI Controller [5c6cdb64edfe08e8511d7120de20486d25dac1c4] forgot to call fu_udev_device_add_flag() with OPEN_READ and/or OPEN_WRITE

5
> 14:59:24.796 FuUdevDevice         Bridge [5ca35f261be27f4d2786162a4e2dd1494697e678] forgot to call fu_udev_device_add_flag() with OPEN_READ and/or OPEN_WRITE

6
> 14:59:24.799 FuUdevDevice         Meteor Lake-P SPI Controller [321e1276dd9bd3e6641d9ba3878d575b2074ef9e] forgot to call fu_udev_device_add_flag() with OPEN_READ and/or OPEN_WRITE

**Steps to Reproduce**
```
$ build-fwupd
$ fwupdtool get-devices
```

**Expected behavior**
No error messages

**fwupd version information**
Please provide the version of the daemon and client.

```shell
$ fwupdtool --version
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
compile   org.freedesktop.fwupd         2.0.0
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libxmlb           0.3.x
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         2.0.0
runtime   org.kernel                    6.8.0-36-generic

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): source, at git-8963b3de4c9.

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdtool get-devices --show-all-devices
[... 238 lines cut ...]
│     Summary:            Miniaturised USB wireless receiver
│     Current version:    RQR12.11_B0032
│     Bootloader Version: BOT01.04_B0016
│     Vendor:             Logitech, Inc. (HIDRAW:0x046D, USB:0x046D)
│     Install Duration:   30 seconds
│     GUID:               279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
│     Device Flags:       • Updatable
│                         • Signed Payload
│   
└─YJ3JK:
      Device ID:          aec1a869eb0df71b7cea6b3ac71d39b830faf164
      GUID:               340848af-fb81-553f-a1d4-9b81b5539548 ← DRM\VEN_AUO&DEV_CFA4
      Device Flags:       • Internal device
    
```

</details>

**Additional questions**

- Operating system and version: Ubuntu 24.04
- Have you tried rebooting? Yes
- Is this a regression? Maybe
- Full verbose log: 
[errors.log](https://github.com/user-attachments/files/16055019/errors.log)

hughsie, 2024-07-25:
@CragW is main branch better now?

CragW, 2024-07-26:
Yes, much better now. :) 

3 warnings are left with main branch up to this moment:
```
fwupdtool get-devices
[sudo] password for crabo: 
Loading…                 [****************                       ]
01:03:47.164 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:02.0/drm/card1/card1-DP-2/drm_dp_aux2: failed to setup: aux dpcd write failed: failed to write 6 bytes to 19: Input/output error
01:03:47.217 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:02.0/drm/card1/card1-eDP-1/drm_dp_aux0: failed to setup: aux dpcd write failed: failed to write 6 bytes to 19: Input/output error
Loading…                 [*********************                  ]
01:03:47.553 FuPluginIntelMe      failed to get public key using file-id 0x40002300: generic failure [0xb]
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
Dell Inc. Latitude 7350
│
```
[main-warnings.txt](https://github.com/user-attachments/files/16385360/main-warnings.txt)

hughsie, 2024-07-26:
> failed to get public key using file-id 0x40002300:

@CragW you need to follow the MEInfo instructions here https://blogs.gnome.org/hughsie/2023/05/09/msi-and-insecure-kms/ -- we need to know how to get the KM on that hardware. i.e. it should be an actual warning.

closed 2024-08-28 by CragW