fwupd/fwupd #7408: fwupd daemon fails to start after building from source

MarekPieta, 2024-07-02:
**Describe the bug**
fwupd daemon fails to start after building locally from source. The problem appears always.
If I run fwupd using `sudo ./build/src/fwupd`, it works without problems.

**Steps to Reproduce**
<details>
  <summary>Build and run commands</summary>

  rm -rf ./build
  meson build
  ninja -C build
  sudo ninja -C build install
  systemctl daemon-reload
  sudo systemctl restart fwupd
  systemctl status fwupd

</details>

**Expected behavior**
`fwupd` daemon should work without problems after building from source.

**fwupd version information**
The issue was replicated on main (40db88bd5007357bb919fcc32bb4d076bc202c8f).
`fwupd` was built locally from source.

The issue started to appear since 7546027e984811fdbc117dd808ba9ee0a3e67137.
Setting `SystemCallErrorNumber=EPERM` also for `not supported_build` seems to prevent the issue from happening on main.
As suggested by the comment, I ran the `sudo coredumpctl debug` to check what additional syscalls are required. `(gdb) p $rax` returns `315`. 

<details>
  <summary> sudo coredumpctl debug output </summary>
           PID: 34378 (fwupd)
           UID: 0 (root)
           GID: 0 (root)
        Signal: 31 (SYS)
     Timestamp: Tue 2024-07-02 09:02:49 CEST (8s ago)
  Command Line: /usr/local/libexec/fwupd/fwupd
    Executable: /usr/local/libexec/fwupd/fwupd
 Control Group: /system.slice/fwupd.service
          Unit: fwupd.service
         Slice: system.slice
       Boot ID: 27703b51f5a4490fb918052c2c7a9eda
    Machine ID: 67e3d03027a34015b0511fb14bc850a8
       Storage: /var/lib/systemd/coredump/core.fwupd.0.27703b51f5a4490fb918052c2c7a9eda.34378.1719903769000000.zst (present)
     Disk Size: 1.4M
       Message: Process 34378 (fwupd) of user 0 dumped core.
                
                Found module linux-vdso.so.1 with build-id: 58b26c11485dce2f068f8a1a66b19e7d5094bdbd
                Found module libftdi1.so.2 with build-id: d7adfdaa0b24e95f593ff7fca81fd8cdbc783d8d
                Found module libpci.so.3 with build-id: cdb834c0e7a047d5f2d608cf74d5240c1827994b
                Found module libflashrom.so.1 with build-id: e887cb03d567294bc237ae72ecc37b46e4935fc8
                Found module libfu_plugin_flashrom.so with build-id: ee0833efbc68d0cddd4bc4872034f84c892e73f7
                Found module libmbim-glib.so.4 with build-id: 837a5925296d941416ad9ba577a593ea7542051a
                Found module libqmi-glib.so.5 with build-id: a9d793244a1efaead6cd9e9ecd7021c8909c5781
                Found module libmm-glib.so.0 with build-id: cd41b3908f77cf3371f5c1c12c65c8e314c3ac66
                Found module libfu_plugin_modem_manager.so with build-id: 8cfc8ba03fb08c8fab11d15a75bbb7c80d0d8ee9
                Found module libtirpc.so.3 with build-id: 1a361e188043ff5abfdb655af6186b8a0f9b205f
                Found module libnsl.so.2 with build-id: 400d0e5ea8cb66596b2f49bfd0dfe0330ef9f51d
                Found module libnss_nis.so.2 with build-id: 19728f013851322d1aebd3682ca8bd4deef6a3f0
                Found module libnss_compat.so.2 with build-id: 8958aa3ad814d117cefa6b33b1c589a61fd2dfa4
                Found module libgcc_s.so.1 with build-id: e3a44e0da9c6e835d293ed8fd2882b4c4a87130c
                Found module libstdc++.so.6 with build-id: e37fe1a879783838de78cbc8c80621fa685d58a2
                Found module libicudata.so.70 with build-id: b1c2496dd0543023c7a19c961bb7f3abc818f465
                Found module libresolv.so.2 with build-id: 95ebf3f0e57d22d884105b9316288167790a36fb
                Found module libkeyutils.so.1 with build-id: ff27227afa5eeddccab180dd29bd7fcff94aea7c
                Found module libicuuc.so.70 with build-id: bef3ff1d70aadd68aab07d858a759360c8b919ae
                Found module libbrotlicommon.so.1 with build-id: 43a72967cf84155914c8b3e915926733d1e57c11
                Found module libsasl2.so.2 with build-id: b8a924f277180f8743a0c6463a379b1a4ad1aae5
                Found module libkrb5support.so.0 with build-id: 384ac9986064cbae89ac8b2e33eb3604b5ab0395
                Found module libcom_err.so.2 with build-id: ce0901f10854b3c9276066b98d9a72303206e0d5
[... 374 lines cut ...]
    supported_build     : False
    static_analysis     : False
    tests               : True
    umockdev_tests      : True

  daemon features
    bluez               : True
    gusb                : YES 0.4.8
    gudev               : YES 237
    hsi                 : True
    lvfs_metadata_format: xz
    libarchive          : True
    offline             : False
    passim              : YES 0.1.9
    sqlite              : YES 3.37.2
    thinklmi_compat     : False
    GPG support         : True
    PKCS7 support       : True

  plugins
    enabled             : acpi-dmar, acpi-facp, acpi-ivrs, acpi-phat, algoltek-aux, algoltek-usb, amd-gpu, amd-pmc, analogix, android-boot, ata, audio-s5gen2, aver-hid, bcm57xx, bios, ccgx, ccgx-dmc, cfu, ch341a, ch347, colorhug, corsair, cpu, cros-ec, dell, dell-dock, dfu, dfu-csr, ebitdo, elanfp, elantp, emmc, ep963x, fastboot, flashrom, focalfp, fpc, fresco-pd, genesys, genesys-gl32xx, goodix-moc, goodix-tp, gpio, hailuck, intel-gsc, intel-me, intel-usb4, iommu, jabra, jabra-gnp, kinetic-dp, lenovo-thinklmi, linux-display, linux-lockdown, linux-sleep, linux-swap, linux-tainted, logind, logitech-bulkcontroller, logitech-hidpp, logitech-rallysystem, logitech-scribe, logitech-tap, mediatek-scaler, modem-manager, msr, mtd, nitrokey, nordic-hid, nvme, optionrom, parade-lspcon, pci-bcr, pci-mei, pci-psp, pixart-rf, powerd, qsi-dock, realtek-mst, redfish, rts54hid, rts54hub, scsi, steelseries, synaptics-cape, synaptics-cxaudio, synaptics-mst, synaptics-prometheus, synaptics-rmi, synaptics-vmm9, system76-launch, test, thelio-io, thunderbolt, ti-tps6598x, tpm, uefi-capsule, uefi-dbx, uefi-esrt, uefi-pk, uefi-recovery, uf2, upower, usi-dock, vbe, vli, wacom-raw, wacom-usb, wistron-dock
    disabled            : 

Found ninja-1.10.2 at /home/mapi/ninja_new/ninja-linux/ninja
</details>

MarekPieta, 2024-07-02:
Hi @hughsie,
Could you advise on how could I workaround/fix the issue?

superm1, 2024-07-04:
I'd suggest use the fwupd development environment.  You don't need to actually install into filesystem.
https://fwupd.github.io/libfwupdplugin/building.html

MarekPieta, 2024-07-04:
Hi @superm1,
Thanks for your suggestion. As mentioned in the ticket, it works for me even without the development environment (running fwupd with `sudo ./build/src/fwupd` ).

Still, I wonder why the issues replicates if fwupd is run as a daemon. I think it may be good to find out what is the root cause to ensure it would not lead to edge cases.

superm1, 2024-07-04:
Sure I think finding the root cause is the right direction, I just wanted to mention a workaround is the fwupd development environment. It works very similarly to an installer system.

MarekPieta, 2024-12-18:
I still replicate the issue on main: 2de25d27bf59aa42273a4581c99f4b5bf11ef774. The same workarounds still work. Reopening.

hughsie, 2024-12-18:
@MarekPieta we merged your PR to main some time ago; what does your installed fwupd.service look like?

MarekPieta, 2024-12-20:
I build and install fwupd from source, so `fwupd.service` should be automatically aligned, right?

I checked again and noticed that the problem actually stopped replicating at some point (e.g. `2.0.0` is not affected) and then started replicating again (I see the issue on the current main: c2542a0748deac2c773d82824be90d86d36f90fe ). On the main branch `sudo coredumpctl debug` - >`(gdb) p $rax` returns `221`. It seems that the issue is related to another syscall number now.

MarekPieta, 2024-12-20:
I will try to bisect to find the root cause (I guess in the beginning of New Year as I start vacation today and I will have no access to my computer). Then we could continue the investigation. Would it be ok?

hughsie, 2024-12-20:
@MarekPieta syscall 221 is fadvise64 -- can you share the backtrace so we know what's using it on your system? https://github.com/fwupd/fwupd/pull/8249 should fix things.

MarekPieta, 2025-01-02:
Hi @hughsie, I just collected the backtrace on main: a76817416710160850766a86455145b3ce59964c
btw. The problem was replicated on Ubuntu 24.04.1 LTS running as VMware Virtual Machine

<details>

```
#0  __GI___posix_fadvise64_l64 (fd=13, offset=0, len=0, advise=1)
    at ../sysdeps/unix/sysv/linux/posix_fadvise64.c:52
#1  0x0000728dc8ac344e in blkid_probe_set_device () at /lib/x86_64-linux-gnu/libblkid.so.1
#2  0x0000728dc98d4cb5 in fu_block_partition_setup
    (device=0x5e6f57d8f630, error=0x7ffce7c87fd8)
    at ../libfwupdplugin/fu-block-partition.c:293
#3  0x0000728dc9b16278 in fu_android_boot_device_setup
    (device=0x5e6f57d8f630, error=0x7ffce7c87fd8)
    at ../plugins/android-boot/fu-android-boot-device.c:92
#4  0x0000728dc9905f1e in fu_device_setup (self=0x5e6f57d8f630, error=0x7ffce7c87fd8)
    at ../libfwupdplugin/fu-device.c:5893
#5  0x0000728dc9905393 in fu_device_open_internal
    (self=0x5e6f57d8f630, error=0x7ffce7c87fd8) at ../libfwupdplugin/fu-device.c:5518
#6  0x0000728dc990562a in fu_device_open (self=0x5e6f57d8f630, error=0x7ffce7c87fd8)
    at ../libfwupdplugin/fu-device.c:5598
#7  0x0000728dc990d8cd in fu_device_locker_new_full
    (device=0x5e6f57d8f630, open_func=0x728dc990540f <fu_device_open>, close_func=0x728dc9905713 <fu_device_close>, error=0x7ffce7c87fd8) at ../libfwupdplugin/fu-device-locker.c:180
#8  0x0000728dc990d733 in fu_device_locker_new
    (device=0x5e6f57d8f630, error=0x7ffce7c87fd8)
    at ../libfwupdplugin/fu-device-locker.c:128
#9  0x0000728dc995e4e6 in fu_plugin_backend_device_added
    (self=0x5e6f57ca6f10, device=0x5e6f57f8ca50, progress=0x5e6f57d8f830, error=0x7ffce7c87fd8) at ../libfwupdplugin/fu-plugin.c:1750
#10 0x0000728dc995e74d in fu_plugin_runner_backend_device_added
    (self=0x5e6f57ca6f10, device=0x5e6f57f8ca50, progress=0x5e6f57d8f830, error=0x7ffce7c87fd8) at ../libfwupdplugin/fu-plugin.c:1798
#11 0x0000728dc9abe5f3 in fu_engine_backend_device_added_run_plugin
    (self=0x5e6f57c3d9e0, device=0x5e6f57f8ca50, plugin_name=0x5e6f57f7bae0 "android_boot", progress=0x5e6f57d8f830, error=0x7ffce7c87fd8) at ../src/fu-engine.c:7243
#12 0x0000728dc9abe6be in fu_engine_backend_device_added_run_plugins
    (self=0x5e6f57c3d9e0, device=0x5e6f57f8ca50, progress=0x5e6f57d8ffd0)
    at ../src/fu-engine.c:7273
#13 0x0000728dc9abeae8 in fu_engine_backend_device_added
    (self=0x5e6f57c3d9e0, device=0x5e6f57f8ca50, progress=0x5e6f58023300)
    at ../src/fu-engine.c:7343
#14 0x0000728dc9ac009e in fu_engine_backends_coldplug_backend_add_devices
    (self=0x5e6f57c3d9e0, backend=0x5e6f57c43560, progress=0x5e6f580f8300, error=0x7ffce7c88130) at ../src/fu-engine.c:7753
#15 0x0000728dc9ac025d in fu_engine_backends_coldplug_backend
    (self=0x5e6f57c3d9e0, backend=0x5e6f57c43560, progress=0x5e6f580fa430, error=0x7ffce7c88130) at ../src/fu-engine.c:7788
#16 0x0000728dc9ac0402 in fu_engine_backends_coldplug
    (self=0x5e6f57c3d9e0, progress=0x5e6f57c4c940) at ../src/fu-engine.c:7824
#17 0x0000728dc9ac1c3a in fu_engine_load
    (self=0x5e6f57c3d9e0, flags=974, progress=0x5e6f57c49130, error=0x7ffce7c883c0)
    at ../src/fu-engine.c:8194
#18 0x00005e6f370b6bd3 in fu_dbus_daemon_setup
    (daemon=0x5e6f57c36cf0, socket_address=0x0, progress=0x5e6f57c49000, error=0x7ffce7c883c0) at ../src/fu-dbus-daemon.c:2843
#19 0x00005e6f370ad8ad in fu_daemon_setup
    (self=0x5e6f57c36cf0, socket_address=0x0, error=0x7ffce7c883c0)
    at ../src/fu-daemon.c:198
#20 0x00005e6f370acae5 in main (argc=1, argv=0x7ffce7c885d8) at ../src/fu-main.c:179
```
</details>

closed 2025-01-02 by superm1