fwupd/fwupd #7414: Failed to open /dev/hidrawX: Operation not permitted

MarekPieta, 2024-07-03:
**Describe the bug**
fwupd cannot open hidraw devices which makes it unable to perform DFU for [nRF Desktop](https://docs.nordicsemi.com/bundle/ncs-latest/page/nrf/applications/nrf_desktop/README.html) devices (handled by `nordic-hid` plugin).

**Steps to Reproduce**
Build fwupd locally and connect an nRF Desktop device that supports configuration channel DFU.

See that `systemctl status fwupd` displays log similar to:
```
fwupd[19540]: 05:18:46.522 FuEngine failed to add device /sys/devices/pci0000:00/0000:00:15.0/0000:03:00.0/usb3/3-3/3-3.1/3-3.1:1.0/0003:1915:52DE.0003/hidraw/hidraw1: failed to open /dev/hidraw1: Operation not permitted
```

Verify that the connected nRF Desktop device is not reported by `fwupdmgr get-devices` too.

**Expected behavior**
fwupd's `nordic-hid` plugin should properly handle nRF Desktop devices.

**fwupd version information**
fwupd built locally from main:  655d38aaf06aba1366c024071987c764a1047288.
Issue was replicated with Ubuntu 22.04.4 LTS.

**Additional context**
Issue was introduced by 8e74c23d616d2d6bfbaba2406cf47029e430e750.
Removing `ProtectClock=yes` from `data/fwupd.service.in` seems to workaround the problem.

MarekPieta, 2024-07-03:
I think the issue is similar to this one: https://github.com/tailscale/tailscale/issues/1063.
Looking at https://www.freedesktop.org/software/systemd/man/latest/systemd.exec.html#ProtectClock=, it seems that setting `ProtectClock` implies `DeviceAllow=char-rtc r` which seemingly effectively blocks fwupd from accessing `hidraw`.

I checked locally that adding `DeviceAllow=char-hidraw rw` seems to workaround problem with accessing `hidraw`, but I am not sure if it's a proper solution (I am a bit worried that the `DeviceAllow` setting from `ProtectClock` may also affect other plugins). @superm1, @hughsie, do you think we can simply add  `DeviceAllow=char-hidraw rw` to `data/fwupd.service.in`? I can open PR if my proposed solution is accepted.

hughsie, 2024-07-03:
> do you think we can simply add DeviceAllow=char-hidraw rw to data/fwupd.service.in

Yes, I think that's perfect -- good catch!

MarekPieta, 2024-07-03:
Hi @hughsie,
I just prepared the PR: https://github.com/fwupd/fwupd/pull/7420

closed 2024-07-03 by hughsie