fwupd/fwupd #7422: Mei version reporting as unknown

LocoTacoTruck, 2024-07-04:
```
(pkttyagent:7369): GLib-GIO-DEBUG: 20:17:30.897: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
Host Security ID: HSI:0 (v1.9.16)

HSI-1
✔ BIOS firmware updates:         Enabled
✔ me manufacturing mode:         Locked
✔ me override:                   Locked
✔ Platform debugging:            Disabled
✔ SPI write:                     Disabled
✔ SPI lock:                      Enabled
✔ SPI BIOS region:               Locked
✔ Supported CPU:                 Valid
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI bootservice variables:    Locked
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled
✘ MEI version:                   Unknown

HSI-2
✔ Intel BootGuard:               Enabled
✔ Intel BootGuard OTP fuse:      Valid
✔ IOMMU:                         Enabled
✔ Platform debugging:            Locked
✔ TPM PCR0 reconstruction:       Valid
✘ Intel BootGuard ACM protected: Invalid
✘ Intel BootGuard verified boot: Invalid

HSI-3
✘ Intel BootGuard error policy:  Invalid
✘ CET Platform:                  Not supported
✘ Pre-boot DMA protection:       Disabled
✘ Suspend-to-idle:               Disabled
✘ Suspend-to-ram:                Enabled

HSI-4
✔ SMAP:                          Enabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -!
✔ fwupd plugins:                 Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Disabled
✔ Linux kernel:                  Untainted

This system has a low HSI security level.
 » https://fwupd.github.io/hsi.html#low-security-level

Host Security Events
  2024-06-26 01:42:42:  ✔ Linux swap changed: Unencrypted → Disabled
  2024-06-25 19:20:06:  ✔ TPM v2.0 changed: Disabled → Found
```

**Describe the bug**
The mei version is reporting as unknown.

**Steps to Reproduce**
I am unsure what causes this but I do have the Intel management engine. ran this command to check hsi version. fwupdmgr security --verbose --allow-reinstall --allow-older --allow-branch-switch

**Expected behavior**
Mei version check should pass or fail.

**fwupd version information**
```
compile   org.freedesktop.fwupd         1.9.16
compile   com.hughsie.libxmlb           0.3.17
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.8.0-36-generic
runtime   org.freedesktop.fwupd         1.9.16
```
came installed on ubuntu updated via apt
Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```
Panasonic Corporation CF-31-5
│
├─Core™ i7-5600U CPU @ 2.60GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x0000002f
│     Vendor:             Intel
│     GUIDs:              bcad68bd-c33a-56cd-a8a6-b9a78d62ee6e ← CPUID\PRO_0&FAM_06&MOD_3D
│                         a702f59f-10c2-5561-976c-b94c89cb42d8 ← CPUID\PRO_0&FAM_06&MOD_3D&STP_4
│     Device Flags:       • Internal device
│   
├─HD Graphics 5500:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    09
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              64c0ea55-d861-5dd0-aa6b-bcc307f57ed4 ← PCI\VEN_8086&DEV_1616
│                         a99720e5-9a76-544d-91c0-a54cc583dcc9 ← PCI\VEN_8086&DEV_1616&SUBSYS_10F78338
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─System Firmware:
│ │   Device ID:          018053a4fed8e8ddcf5e53028ddbcd6181008794
│ │   Summary:            UEFI System Resource Table device (updated via NVRAM)
│ │   Current version:    17045504
│ │   Minimum Version:    17045504
│ │   Vendor:             Panasonic Corporation (DMI:American Megatrends Inc.)
│ │   Update State:       Failed
│ │   GUID:               853ede8f-b696-51f8-9679-d29385d5e947
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │   Device Requests:    • Message
│ │ 
│ ├─AMT [provisioned]:
│ │     Device ID:        8d5470e73fd9a31eaa460b2b6aea95483fe3f14c
│ │     Summary:          Hardware and firmware technology for remote out-of-band management
│ │     Current version:  10.0.60.3000
│ │     Bootloader Version:10.0.60.3000
│ │     Vendor:           Intel Corporation (MEI:0x8086)
│ │     GUIDs:            12f80028-b4b7-4b2d-aca8-46e0ff65814c
│ │                       4e4029a8-fe0b-5e35-99a7-eca3cbb8c828 ← MEI\VEN_8086&DEV_9CBA
│ │                       e369525b-943b-5a43-be5b-f905913acb19 ← MEI\VEN_8086&DEV_9CBA&SUBSYS_10F78338
│ │     Device Flags:     • Internal device
│ │   
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  371
│       Minimum Version:  371
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUID:             f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Only version upgrades are allowed
│                         • Signed Payload
│     
├─TEAM T2531TB:
│     Device ID:          602b0a6cc821d155208724f0e22f8d111542b74c
│     Summary:            ATA drive
│     Current version:    W0220A0
│     Vendor:             Team Group (ATA:0x0)
│     Serial Number:      TPBF2307040030205822
│     GUIDs:              8077e844-c377-5618-bbdc-f19dd8263347 ← IDE\TEAM_T2531TB____________________________W0220A0
│                         eb8bd3de-ce93-5186-aaf8-5c1393a4ef17 ← IDE\0TEAM_T2531TB____________________________
│                         44cc3ba1-fde6-56c6-b937-5f58abfbd7d5 ← TEAM T2531TB
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─TPM:
│     Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
│     Current version:    5.62.12.13824
│     Vendor:             Infineon (TPM:IFX)
│     GUIDs:              5eebb112-75ad-5536-b173-a11eb3399402 ← TPM\VEN_IFX&DEV_0000
│                         1a10e308-1ceb-5b61-bbd0-07461469ab8f ← TPM\VEN_IFX&MOD_SLB9665
│                         6d81ab63-db2e-50ac-934f-6be9accf5e02 ← TPM\VEN_IFX&DEV_0000&VER_2.0
│                         5cec2f57-3b14-5f7d-9436-882896e3923d ← TPM\VEN_IFX&MOD_SLB9665&VER_2.0
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device can recover flash failures
│                         • Full disk encryption secrets may be invalidated when updating
│                         • Signed Payload
│   
└─UEFI Platform Key:
      Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
      Summary:            Panasonic Corporation
      Vendor:             Panasonic Corporation
      GUID:               e5d9cb21-a0f9-5db0-96e8-ffd05d254323 ← UEFI\CRT_8391BBD570A415D64081146996536895F2757A94
```

</details>

**Additional questions**

- Operating system and version: Ubuntu 24.04 LTS
- Have you tried rebooting? yes
- Is this a regression? I dont know?

LocoTacoTruck, 2024-07-11:
I pulled some more info that seems like it might be helpful.
```
{
  "SecurityAttributes" : [
    {
      "AppstreamId" : "org.fwupd.hsi.Fwupd.Plugins",
      "Created" : 1720713071,
      "HsiLevel" : 0,
      "HsiResult" : "not-tainted",
      "HsiResultSuccess" : "not-tainted",
      "Name" : "fwupd plugins",
      "Summary" : "Firmware Updater Verification",
      "Description" : "Firmware Updater Verification checks that software used for updating has not been tampered with.",
      "Plugin" : "core",
      "Uri" : "https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Fwupd.Plugins",
      "Flags" : [
        "success",
        "runtime-issue"
      ]
    },
    {
      "AppstreamId" : "org.fwupd.hsi.Kernel.Lockdown",
      "Created" : 1720713071,
      "HsiLevel" : 0,
      "HsiResult" : "enabled",
      "HsiResultSuccess" : "enabled",
      "Name" : "Linux kernel lockdown",
      "Summary" : "Linux Kernel Lockdown",
      "Description" : "Linux Kernel Lockdown mode prevents administrator (root) accounts from accessing and changing critical parts of system software.",
      "Plugin" : "linux_lockdown",
      "Uri" : "https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Kernel.Lockdown",
      "Flags" : [
        "success",
        "runtime-issue"
      ]
    },
    {
      "AppstreamId" : "org.fwupd.hsi.Kernel.Swap",
      "Created" : 1720713071,
      "HsiLevel" : 0,
      "HsiResult" : "not-enabled",
      "HsiResultSuccess" : "encrypted",
      "Name" : "Linux swap",
      "Summary" : "Linux Swap",
      "Description" : "Linux Kernel Swap temporarily saves information to disk as you work. If the information is not protected, it could be accessed by someone if they obtained the disk.",
      "Plugin" : "linux_swap",
      "Uri" : "https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Kernel.Swap",
      "Flags" : [
        "success",
        "runtime-issue"
      ]
    },
    {
      "AppstreamId" : "org.fwupd.hsi.Kernel.Tainted",
      "Created" : 1720713071,
      "HsiLevel" : 0,
      "HsiResult" : "not-tainted",
      "HsiResultSuccess" : "not-tainted",
      "Name" : "Linux kernel",
      "Summary" : "Linux Kernel Verification",
      "Description" : "Linux Kernel Verification makes sure that critical system software has not been tampered with. Using device drivers which are not provided with the system can prevent this security feature from working correctly.",
      "Plugin" : "linux_tainted",
      "Uri" : "https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Kernel.Tainted",
      "Flags" : [
        "success",
        "runtime-issue"
      ]
    },
    {
      "AppstreamId" : "org.fwupd.hsi.Bios.CapsuleUpdates",
[... 873 lines cut ...]
      "HsiResultSuccess" : "enabled",
      "Name" : "SMAP",
      "Summary" : "Supervisor Mode Access Prevention",
      "Description" : "Supervisor Mode Access Prevention ensures critical parts of device memory are not accessed by less secure programs.",
      "Plugin" : "cpu",
      "Uri" : "https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.IntelSmap",
      "Flags" : [
        "success"
      ],
    },
    {
      "AppstreamId" : "org.fwupd.hsi.EncryptedRam",
      "Created" : 1720713071,
      "HsiLevel" : 4,
      "HsiResult" : "not-supported",
      "HsiResultSuccess" : "enabled",
      "Name" : "Encrypted RAM",
      "Summary" : "Encrypted RAM",
      "Description" : "Encrypted RAM makes it impossible for information that is stored in device memory to be read if the memory chip is removed and accessed.",
      "Plugin" : "cpu",
      "Uri" : "https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.EncryptedRam",
    }
  ]
}
```

hughsie, 2024-07-25:
@LocoTacoTruck is a really new Meteorlake system by any chance?

LocoTacoTruck, 2024-07-25:
> @LocoTacoTruck is a really new Meteorlake system by any chance?

Actually no it's a few years old it's a cf-31 mark 5 board revision 1 I can give you a list of all the hardware components if that would help?

hughsie, 2024-11-11:
If you run `sudo fwupdtool security --plugins pci_mei -v` do you get a message like `ME family not supported for...` ?