fwupd/fwupd #7477: System journal spammed by "Connection ":1.110" is not allowed to add more match rules" warning

A6GibKm, 2024-07-12:
**Describe the bug**

The exact message is:
```
jul 12 08:33:46 host dbus-daemon[2534]: [system] Connection ":1.110" is not allowed to add more match rules (increase limits in configuration file if required; max_match_rules_per_connection=512)
```

In a span of tow hours I see the same message 1194 times, which suggest that the service tried to add ~1600 match rules.

As per `busctl` the `:1.110` comes from

```
:1.110                                 7069 fwupd           root            :1.110        fwupd.service                 -       -          
```

**Steps to Reproduce**

Open the system journal with `journalctl -p 4 -b`.

**Expected behavior**

See a only a handful of messages or none at all.

**fwupd version information**
```shell
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.8.8-2-pve
runtime   org.freedesktop.fwupd         1.8.12
```

**Additional questions**

- Operating system and version: Proxmox VE 8 (based on debian 12)
- Have you tried rebooting? It happens on every boot
- Is this a regression? No idea.

hughsie, 2024-07-12:
I've never seen that before, I'm afraid you'll have to do a bit more debugging.

A6GibKm, 2024-07-12:
I tried following the bus while restarting the fwupd service and found ~1900 calls with the same `MESSAGE`:
```
Type=method_call  Endian=l  Flags=0  Version=1 Cookie=3949  Timestamp="Fri 2024-07-12 09:35:34.371176 UTC"
  Sender=:1.428  Destination=org.freedesktop.DBus  Path=/org/freedesktop/DBus  Interface=org.freedesktop.DBus  Member=GetNameOwner
  UniqueName=:1.428
  MESSAGE "s" {
          STRING "org.freedesktop.UDisks2";
  };
```
Also ~3200 calls to RemoveMatch for UDisk2:

```
‣ Type=method_call  Endian=l  Flags=0  Version=1 Cookie=3973  Timestamp="Fri 2024-07-12 09:35:34.377576 UTC"
  Sender=:1.428  Destination=org.freedesktop.DBus  Path=/org/freedesktop/DBus  Interface=org.freedesktop.DBus  Member=RemoveMatch
  UniqueName=:1.428
  MESSAGE "s" {
          STRING "type='signal',sender='org.freedesktop.UDisks2',interface='org.freedesktop.UDisks2.Partition',path='/org/freedesktop/UDisks2/block_devices/zd160'";
  };
```

As per `lsblk | wc -l` I have a lot of disks, the command reports 289. So maybe it is a weird interaction with UDisks2 when there is an absurd amount of disks?

hughsie, 2024-11-11:
I don't know what to suggest I'm afraid.

A6GibKm, 2024-11-11:
Me neither, I am not sure if this is even an issue on the app side.

hughsie, 2025-05-04:
I think you'll have to up the max_match_rules_per_connection value somehow if you have that number of disks. Sorry that's not more helpful.

closed 2025-05-04 by hughsie