fwupd/fwupd #7495: Refresh Metadata jcat file (sometimes) has incorrect file name

rishabh-chromium, 2024-07-17:
**Describe the bug**
We mirror fwupd every 4 hours as recommended. Every few of days `refresh metadata` fails and we get this error message: 
```
{
  "Error" : {
    "Domain" : "FwupdError",
    "Code" : 7,
    "Message" : "Failed to download metadata for lvfs: Failed to download, server response was 404: <?xml version='1.0' encoding='UTF-8'?><Error><Code>NoSuchKey</Code><Message>The specified key does not exist.</Message><Details>No such object: chromeos-localmirror/lvfs/firmware-07347-stable.xml.zst</Details></Error>"
  }
}
```

I then manually tried to download the `firmware.xml.xz.jcat` file from our local mirror and unzipped it. I saw that the filename in the jcat file is infact `firmware-07347-stable` however the latest firmware file in our mirror is only `firmware-07346-stable.xml`. Maybe there is a delay somewhere. This causes all the Chromebook devices `refresh metadata` to fail and I am fairly certain we aren't the only ones this is affecting since our mirroring process is exactly as recommended in the docs.

rishabh-chromium, 2024-07-17:
We run `fwupdmgr refresh` every 24 hours and we have only a 60% success rate. 39% failure is because the file is not found

hughsie, 2024-07-17:
Would it be correct to say:

 * There's a `firmware.xml.xz.jcat` in the mirror that talks about `firmware-07347-stable.xml.whatever`
 * There's a file listed in PULP_MANIFEST called `firmware.xml.xz.jcat` and also `firmware-07346-stable.xml.whatever`

So plausibly what's happened is that we regenerated PULP_MANIFEST, then some time later regenerated the metadata (altering `firmware.xml.xz.jcat`) -- perhaps when you're downloaded the data. I think the solution here is to include the `firmware.xml.xz.jcat` of the *generated* data, not of the _file-right-now_. I'm not sure how yet.

Is it possible the Google mirror sync takes a long time (we only resync the public remote every 4 hours), or it's doing the sync from the bottom-up or something crazy? Mine takes less than a couple of minutes and I'm struggling to work out why it would fail so often.

hughsie, 2024-07-17:
Also, what mirror URL are you pulling please? I'm guessing it's not the `cdn.` prefixed one?

rishabh-chromium, 2024-07-17:
It is possible that part of the 39% failure is because user wifi disconnected or some kind of connection issue so I wouldn't think the entire failure rate is due to this problem. However our local tests (stable internet) also have a 20% failure rate. 

Our sync average about 2 minutes, so not a long time... The url we are pulling is infact: `https://cdn.fwupd.org/downloads`

hughsie, 2024-07-17:
> The url we are pulling is infact: https://cdn.fwupd.org/downloads

hughsie, 2024-07-17:
> The url we are pulling is infact: https://cdn.fwupd.org/downloads

Ohh, that's the wrong thing to do! The CDN doesn't geo-invalidate instantly, and there's often hours of delay on each POP. That's the reason the jcat specifies a metadata that always exist, even if not newest. If you're **populating** a mirror for other clients to use you want a coherent set, and for that just remove the `cdn.` prefix.

rishabh-chromium, 2024-08-07:
Thanks that fixed most of the problem

closed 2024-08-07 by rishabh-chromium