fwupd/fwupd #7507: In ubuntu 24.04 LTS + fwupd1.9.16, using "fwupdmgr install" has no error but bios isn't upgrading after restart during POST

OscarHsu777, 2024-07-19:
### **Describe the question**
I am currently using Ubuntu 22.04 LTS (Jammy), and the fwupd version is 1.7.9. 
We can successfully update the BIOS using a cab file, but we cannot generate an offline report. 
We want to upgrade fwupd to version 1.9.11 or later. How can we achieve this?

Additionally, we have installed a new Ubuntu 22.04 LTS (Noble) with fwupd version 1.9.16. 
On this version of Ubuntu, we cannot successfully update the BIOS using a cab file. 
After running the install command, it shows "Successfully installed firmware," but after a system reboot, there is no progress bar for the update. 
The system directly boots into the Ubuntu desktop, and using the get-devices command shows that the system firmware version has not been updated. 

### **fwupd version information**
**#fwupdmgr --version**
compile   org.freedesktop.fwupd         1.9.16
compile   com.hughsie.libxmlb           0.3.17
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.8.0-31-generic
runtime   org.freedesktop.fwupd         1.9.16

**# fwupdmgr get-devices**
LENOVO 12U60004FR
│
├─System Firmware:
│    Device ID:          395919e8bc3ecf9f51dcef70f2c1d3669bb281b6
│    Summary:            UEFI System Resource Table device (Updated via caspule-on-disk)
│    Current version:    0.1.60
│    Minimum Version:    65596
│    Vendor:             LENOVO (DMI:LENOVO)
│    Update State:       Success
│    GUID:               xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxx(This item will not be made public due to privacy issues)
│    Device Flags:       • Internal device
│                        • Updatable
│                        • System requires external power source
│                        • Needs a reboot after installation
│                        • Cryptographic hash verification is available
│                        • Device is usable for the duration of the update
│    Device Requests:    • Message

**# fwupdmgr install XXX.cab** (Privacy Issues File Name Unpublished)
Waiting…                 [***************************************]
Successfully installed firmware
An update requires a reboot to complete. Restart now? [y|N]:  y

hughsie, 2024-07-19:
Is this system set up with Capsule-On-Disk? If you disable the COD support (man fwupd.conf) I assume it works again with the new fwupd? If you want COD to work you need to ask your IBV what exact filename and path fwupd is supposed to write to the ESP.

superm1, 2024-07-21:
Here is how to disable it temporarily:

https://fwupd.github.io/libfwupdplugin/fwupd.conf.html#uefi_capsule-parameters

Restart the daemon after you've disabled it and it should show up in fwupdmgr get-devices output as NVRAM instead of capsule on disk.

If that works you should work with your BIOS team to fix CoD.

closed 2024-11-11 by hughsie