fwupd/fwupd #7508: missing signed bootloader for secure boot: /usr/lib/fwupd/efi/fwupdx64.efi.signed cannot be found

kmille, 2024-07-21:
On Arch Linux, I get

```
kmille@linbox: fwupdmgr update
Devices with no available firmware updates: 
 • ThinkPad Dock
 • ThinkPad Dock
 • private...
Devices with the latest available firmware version:
 • Embedded Controller
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade xxxxxxxxxxxx?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
...
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: 
Waiting…                 [***************************************]
missing signed bootloader for secure boot: /usr/lib/fwupd/efi/fwupdx64.efi.signed cannot be found
```

I use https://github.com/kmille/cryptboot for my SecureBoot setup. Any tips on how to debug?

**fwupd version information**
fwupd 1.9.21-3
fwupd-efi fwupd-efi

```shell
kmille@linbox: fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.19
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.fwupd         1.9.21
runtime   org.freedesktop.Passim        0.1.8
compile   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libxmlb           0.3.19
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.gusb          0.4.9
runtime   org.kernel                    6.9.9-hardened1-1-hardened
runtime   org.freedesktop.fwupd         1.9.21

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
pacman

Sorry I have some privacy concerns answering the rest of the questions.

hughsie, 2024-07-21:
> I use https://github.com/kmille/cryptboot for my SecureBoot setup. 

I've never even heard of that, sorry.

> Any tips on how to debug?

Well, does /usr/lib/fwupd/efi/fwupdx64.efi.signed actually exist? If it doesn't exist in that location, where is it installed?

kmille, 2024-07-21:
It does not exist. Can't find it with 'locate' and 'pacman -F'. What's the purpose of it?
Maybe I can fix it for my self (like disabling SB temporary), but I don't know how fwupd works.

superm1, 2024-07-21:
When you sign your binaries with your key that is where you put the signed fwupd binary.

If it's signed with your own personal key it doesn't exist until YOU sign it.

kmille, 2024-07-23:
Thanks for the support...

```bash
kmille@linbox:~ sudo cp /usr/lib/fwupd/efi/fwupdx64.efi /usr/lib/fwupd/efi/fwupdx64.efi.signed
kmille@linbox:sudo cryptboot-efikeys sign /usr/lib/fwupd/efi/fwupdx64.efi.signed
Signing file '/usr/lib/fwupd/efi/fwupdx64.efi.signed' with UEFI Secure Boot keys...
warning: data remaining[56320 vs 67241]: gaps between PE/COFF sections?
warning: data remaining[56320 vs 67248]: gaps between PE/COFF sections?
Signing Unsigned original image

kmille@linbox: sudo cryptboot-efikeys verify /usr/lib/fwupd/efi/fwupdx64.efi.signed
List of all signatures in '/usr/lib/fwupd/efi/fwupdx64.efi.signed':
warning: data remaining[57856 vs 68784]: gaps between PE/COFF sections?
signature 1
Verifying signature with UEFI Secure Boot keys...
warning: data remaining[57856 vs 68784]: gaps between PE/COFF sections?
Signature verification OK
```

```
Perform operation? [Y|n]: y
Decompressing…           [                                       ] Less than one minute remaining…
Secure boot is enabled, but shim isn't installed to EFI/arch/shimx64.efi
```

What's next? I don't have/use shim... I only have this:
```bash
kmille@linbox:EFI pwd       
/boot/efi/EFI
kmille@linbox:EFI fd                                                         
GRUB/
GRUB/grubx64.efi
arch/
arch/fw/
arch/fw/fwupd-9a7f2771-7ce4-4ff1-892b-c2ed700b77de.cap
```

superm1, 2024-07-23:
See https://fwupd.github.io/libfwupdplugin/fwupd.conf.html#uefi_capsule-parameters

You'll need to mark it as not requiring shim because you signed grub directly.

kmille, 2024-07-24:
Is it possible that the update was installed without noticing?

```
kmille@linbox:~ fwupdmgr update                    
Devices with no available firmware updates: 
 • CT500xxxxxxxxx
Devices with the latest available firmware version:
 • Embedded Controller
kmille@linbox:~ fwupdmgr upgrade
Devices with no available firmware updates: 
 • CT500xxxxxxxxx
Devices with the latest available firmware version:
 • Embedded Controller
```

Can I check if there are new updates somehow? Or see if the update was successfull? I'm also a bit confused that `fwupdmgr` runs without sudo...

superm1, 2024-07-24:
You can look at your history with `fwupdmgr get-history`.

The client doesn't actually install the update, it's installed from the daemon.

You can review https://github.com/fwupd/fwupd/security/policy to better understand how it all works.

kmille, 2024-07-28:
Seems like it worked. Thank you very  much!
```
Devices that have been updated successfully:                                                                           
 • System Firmware (0.1.48 → 0.1.51)   
```

closed 2024-07-28 by kmille