fwupd/fwupd #7509: BT module of Wacom One after flashing reports version 0.0

d4s, 2024-07-21:
**Describe the bug**

`fwupd` daemon reports the version of BT module for Wacom One pen tablet is 0.0.
Reproduced at least with CTC-4110WL and CTC-6110WL tablets.

* the core issue: after reset with the command `d6` the new device appears, but looks like the BT module need additional time, since it is reporting `09 00 00` meaning module exists, but the version is 0.0:
```
Jul 21 21:25:19 amber fwupd[3526677]: FuHidDevice          HID::GetReport [wValue=0x03cb, wIndex=0]:cb ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
Jul 21 21:25:19 amber fwupd[3526677]: FuHidDevice          HID::GetReport [wValue=0x03cb, wIndex=0]:cb 01 46 02 3f 01 01 ff 09 00 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```
**Workaround** with additional 1 seconds delay in `fu_wac_device_setup()` works, but need proper solution.

* minor issue: GetStatus returns code 0x20, which is not covered, so can't be parsed properly:
```
9:53:11.875 FuHidDevice          HID::GetReport [wValue=0x03d5, wIndex=0]:d5 ff ff ff ff 
19:53:11.878 FuHidDevice          HID::GetReport [wValue=0x03d5, wIndex=0]:d5 20 00 00 00 
19:53:11.878 FuPluginWacomUsb     status now: 
```

**Steps to Reproduce**
Try to update, downgrade or reinstall the F/W available at LVFS (1.01 or 1.04 atm), and check the version with `fwupdmgr get-devices`. Bluetooth Module shows it is 0.0

* NB:
Reproduced with ChromeOS 125-15853.0.0(fwupd version: 1.9.15).
Not reproduced on ChromeOS 124-15790.0.0(fwupd version: 1.9.10).
But I reproduce that behavior on my host with different fwupd versions, starting 1.9.6 to current main branch, however the kernel and libusb/libgusb are the same, so probably might be dependent on kernel/libusb combination.

**Expected behavior**
Bluetooth Module should report the proper version.

**fwupd version information**

```shell
$ fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.19
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.1.14
compile   org.freedesktop.fwupd         1.9.22
runtime   org.freedesktop.fwupd-efi     2.0.0
compile   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libxmlb           0.3.19
runtime   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.gusb          0.4.9
runtime   org.kernel                    6.6.31-un-def-alt1
runtime   org.freedesktop.fwupd         1.9.22
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
- source
<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**Additional questions**

- Operating system and version:
- Have you tried rebooting?
- Is this a regression?
  Don't know -- reproduced with older versions.

hughsie, 2024-07-22:
@jigpu what's GetStatus return code 0x20? @d4s I wonder if the 1s delay might be better in reload() perhaps -- does that work?

hughsie, 2024-07-22:
> what about https://github.com/fwupd/fwupd/commit/f7aa5d263504bb2ff08b3e0a706b2b696486a3d8

That's probably a good idea, but I am slightly worried about the case where a module is legitimately version 0.0.

d4s, 2024-07-22:
> I am slightly worried about the case where a module is legitimately version 0.0.

Also worrying about that -- I know not enough about Wacom H/W.
Let's wait for @jigpu.

jigpu, 2024-07-22:
> Describe the bug
>
> fwupd daemon reports the version of BT module for Wacom One pen tablet is 0.0.
> Reproduced at least with CTC-4110WL and CTC-6110WL tablets.

Is this 0.0 version associated with an error in the flashing process? Does the version change from 0.0 to a valid version if you unplug and replug the device? That's what the commit message in your proposed patch says, but I want to confirm because that behavior is different from what I've seen before.

> That's probably a good idea, but I am slightly worried about the case where a module is legitimately version 0.0.

Version 0.0 should only ever be used when the module is in an error state. It should not be used by any legitimate firmware. As far as I understood, it is normally indicative of a semi-permanent condition. For example, you'd expect to see 0.0 if a module flash failed and it requires re-flashing.

If version 0.0 does appear in this case to be more transitory (i.e. waiting longer for the for the device to initialize cures it) then I can understand the desire to retry the `FU_WAC_REPORT_ID_FW_DESCRIPTOR` request a few times.

> @jigpu what's GetStatus return code 0x20?

I don't have an answer for this right now. I'd have to poke the firmware team to see what they say.

d4s, 2024-07-22:
> Is this 0.0 version associated with an error in the flashing process?

nope, flashing went well. But after reset it reports 0.0 for some time (about 1 sec) and later answer correctly. With https://github.com/fwupd/fwupd/commit/f7aa5d263504bb2ff08b3e0a706b2b696486a3d8 is looks like:
```
08:23:07.441 FuHidDevice          HID::GetReport [wValue=0x03cb, wIndex=0]:cb ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
08:23:07.444 FuHidDevice          HID::GetReport [wValue=0x03cb, wIndex=0]:cb 01 46 02 3f 01 04 ff 09 00 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
08:23:07.544 FuHidDevice          HID::GetReport [wValue=0x03cb, wIndex=0]:cb ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
08:23:07.546 FuHidDevice          HID::GetReport [wValue=0x03cb, wIndex=0]:cb 01 46 02 3f 01 04 ff 09 00 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
08:23:07.646 FuHidDevice          HID::GetReport [wValue=0x03cb, wIndex=0]:cb ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
08:23:07.648 FuHidDevice          HID::GetReport [wValue=0x03cb, wIndex=0]:cb 01 46 02 3f 01 04 ff 09 00 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
08:23:07.748 FuHidDevice          HID::GetReport [wValue=0x03cb, wIndex=0]:cb ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
08:23:07.750 FuHidDevice          HID::GetReport [wValue=0x03cb, wIndex=0]:cb 01 46 02 3f 01 04 ff 09 00 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
08:23:07.850 FuHidDevice          HID::GetReport [wValue=0x03cb, wIndex=0]:cb ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
08:23:07.852 FuHidDevice          HID::GetReport [wValue=0x03cb, wIndex=0]:cb 01 46 02 3f 01 04 ff 09 00 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
08:23:07.952 FuHidDevice          HID::GetReport [wValue=0x03cb, wIndex=0]:cb ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
08:23:07.954 FuHidDevice          HID::GetReport [wValue=0x03cb, wIndex=0]:cb 01 46 02 3f 01 04 ff 09 00 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
08:23:08.054 FuHidDevice          HID::GetReport [wValue=0x03cb, wIndex=0]:cb ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
08:23:08.056 FuHidDevice          HID::GetReport [wValue=0x03cb, wIndex=0]:cb 01 46 02 3f 01 04 ff 09 00 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
08:23:08.156 FuHidDevice          HID::GetReport [wValue=0x03cb, wIndex=0]:cb ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
08:23:08.158 FuHidDevice          HID::GetReport [wValue=0x03cb, wIndex=0]:cb 01 46 02 3f 01 04 ff 09 00 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
08:23:08.258 FuHidDevice          HID::GetReport [wValue=0x03cb, wIndex=0]:cb ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
08:23:08.260 FuHidDevice          HID::GetReport [wValue=0x03cb, wIndex=0]:cb 01 46 02 3f 01 04 ff 09 00 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
08:23:08.360 FuHidDevice          HID::GetReport [wValue=0x03cb, wIndex=0]:cb ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
08:23:08.362 FuHidDevice          HID::GetReport [wValue=0x03cb, wIndex=0]:cb 01 46 02 3f 01 04 ff 09 00 21 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
```

closed 2024-07-25 by d4s