fwupd/fwupd #7520: "Encrypted RAM" show Enabled in Linux OS(Ubuntu22.04) with Memory Encryption (TME) is disabled

pma1, 2024-07-23:
**Describe the bug**
On 22.04, after install edge channel of snap.
====================================
It always shows "Not Supported" no matter what kernel is and mem encryption is enabled or not.
fwupd: 1.9.21-7-g91b2faf2f

On 22.04 with default version of fwupd:
================================
Encrypted RAM will always suggests Enabled when running fwupdmgr security --force for both 6.5.0-1026-oem and 6.8.0-38-generic.

On 24.04:
=========
root@P7-SIT-C7:~# fwupdmgr --version
compile org.freedesktop.fwupd 1.9.16
compile com.hughsie.libxmlb 0.3.17
compile com.hughsie.libjcat 0.2.0
runtime org.freedesktop.fwupd-efi 1.4
compile org.freedesktop.gusb 0.4.8
runtime com.hughsie.libjcat 0.2.0
runtime org.freedesktop.gusb 0.4.8
runtime org.kernel 6.8.0-38-generic
runtime org.freedesktop.fwupd 1.9.16

`Encrypted RAM` is aligned with `Memory Encryption`.

**Steps to Reproduce**
Install Ubuntu 22.04 and snap package at edge then run:
fwupdmgr security --force

**Expected behavior**
`Encrypted RAM` should be aligned with `Memory Encryption`.

**Additional questions**

- Operating system and version:
Ubuntu 22.04 failed but Ubuntu 24.04 passed.
- Have you tried rebooting?
Yes.
- Is this a regression?
No.

hughsie, 2024-07-23:
As the P7 is a Intel Xeon system, we map the encrypted ram attribute to TME. We in turn detect TME using the CPU bits `ECX>>13` from CPUID. Does the cpuid output differ on the different OS versions?

Irving2-11-cell, 2024-07-24:
Attached the cpuid log both Ubuntu22.04 and Ubuntu24.04 here.
[cpuid-22-04.log](https://github.com/user-attachments/files/16356380/cpuid-22-04.log)

[cpuid-24-04.log](https://github.com/user-attachments/files/16356381/cpuid-24-04.log)

hughsie, 2024-07-24:
I think we need the raw values, i.e. the output of "cpuid --raw --one-cpu"

Irving2-11-cell, 2024-07-25:
Attach the raw values here.
[raw-value-22-04.log](https://github.com/user-attachments/files/16369254/raw-value-22-04.log)
[raw-value-24-04.log](https://github.com/user-attachments/files/16369255/raw-value-24-04.log)

hughsie, 2024-07-25:
Hmm, unexpected. There are differences, but not in ecx like I expected. I think maybe the MSRs are maybe set in a different way -- i.e. we read IA32_TME_ACTIVATION to check the TME configuration too. Can you grab the output of `sudo fwupdtool security --no-timestamp -vv --plugins cpu --plugins msr` on both versions please. Thanks.

Irving2-11-cell, 2024-07-26:
Attach the MSRs values here.
[msrs-24.04.log](https://github.com/user-attachments/files/16386912/msrs-24.04.log)
[msrs-22-04.log](https://github.com/user-attachments/files/16386913/msrs-22-04.log)

hughsie, 2024-07-26:
Aha, there we go.

> [msrs-22-04.log](https://github.com/user-attachments/files/16386913/msrs-22-04.log)
 > FuEngine             disabling plugin because: failed to startup using msr: missing kernel support

So in 22.04 the kernel support for MSRs is missing and thus the fwupd plugin doesn't get started -- which means it can't read the MSR to check if TME is actually enabled. It seems to work fine in 24.04. Not a fwupd issue (as there's nothing we can do), but I'm glad to find the root cause.

Irving2-11-cell, 2024-07-29:
Attached  the related files that output in Ubuntu 22.04 that kernel update to 6.8.0-38-generic.
[msrs.log](https://github.com/user-attachments/files/16407939/msrs.log)
[raws-value.log](https://github.com/user-attachments/files/16407941/raws-value.log)

pma1, 2024-07-29:
@hughsie thanks for your kindly help.

from log above, msr is supported by 6.8 kernel on ubuntu 22.04.

when bios disabled encrypted RAM, fwmgr still show enabled.

the output of fwngr with 6.5 kernel is different witj 6.8 kernel. but still not correct.

superm1, 2024-11-11:
From what @pma1 said it still sounds like it's not doing it properly because turning off in BIOS fwupd still shows enabled...

o5k, 2025-02-28:
Pitching in here too, I have a system on 24.04.2 with a Xeon W, which supports TME but not MKTME. Curiously, _enabling_ TME in the BIOS (correctly reported as such in dmesg, `[    0.000000] x86/tme: enabled by BIOS`) flipped the state in fwupd from enabled to not-supported...