fwupd/fwupd #7531: plugin ccgx-dmc does not complete fw update: invalid dmc intr req opcode 0x82[reenum]

ChristianAtSIE, 2024-07-25:
**Describe the bug**
The ccgx-dmc plugin should install the firmware for the hx3pd over the dock interface. The infineon software for windows seems to have multiple reconnects to the device, maybe this is one of the reconnects and the update firmware is confused. But I am not sure on that.

Error is thrown in fu-ccgy-dmc-device.c line 621.

Log of a successfull update with the windows tool HX3PD Configuration Utility Version 1.0 Build 23
```shell
Loading dock composite image file
Verifying hash of image#0
Segment #0
Segment #1
Verifying hash of image#1
Segment #0
Segment #1
Verifying hash of image#2
Segment #0
Segment #1
Verifying hash of image#3
Segment #0
Segment #1
Verifying hash of image#4
Segment #0
Verifying hash of image#5
Segment #0
Opening handle to dock
Connecting to the DMC
Starting FW upgrading
Sending start command
Waiting for FW request

Writing image #0 : DMC (Comp-id: 0), Image-0
Segment #0
Sending image write command
Segment #1
Sending image write command

Waiting for FW request
Received re-enumeration notification
Length mismatch in received message
DMC disconnected, waiting for re-connection with DMC
time out = 60000
Starting FW upgrading
Sending start command
Waiting for FW request

Writing image #1 : DMC (Comp-id: 0), Image-1
Segment #0
Sending image write command
Segment #1
Sending image write command

Waiting for FW request
Received re-enumeration notification
Length mismatch in received message
DMC disconnected, waiting for re-connection with DMC
time out = 60000
Starting FW upgrading
Sending start command
Waiting for FW request

Writing image #2 : HX3PD PD Controller (Comp-id: 1), Image-0
Segment #0
Sending image write command
Segment #1
Sending image write command

Waiting for FW request

Writing image #5 : HX3PD Hub Controller (Comp-id: 2), Image-1
Segment #0
Sending image write command

Waiting for FW request

Writing image #4 : HX3PD Hub Controller (Comp-id: 2), Image-0
Segment #0
Sending image write command

Waiting for FW request
Received re-enumeration notification
DMC disconnected, waiting for re-connection with DMC
time out = 60000
Starting FW upgrading
Sending start command
Waiting for FW request

Writing image #3 : HX3PD PD Controller (Comp-id: 1), Image-1
Segment #0
Sending image write command
Segment #1
Sending image write command

Waiting for FW request

Finished FW update process with status COMPLETED (0x3)
DMC disconnected, waiting for re-connection with DMC
time out = 60000
```

**Steps to Reproduce**

1. Installing quirk
`cat /home/phoenix/fwupd/venv/bin/../dist/lib/fwupd/quirks.d/ccgx_dmc.quirk`
```shell
[USB\VID_04B4&PID_521B]
Plugin = ccgx_dmc
RemoveDelay = 73200
```
2. Connect HX3PD device, e.g. CX6611 EVAL Kit
3. Get firmware https://www.infineon.com/cms/en/product/gated-document/ez-usb-hx3pd-cyusb4347-firmware-8ac78c8c7d718a49017d9b0f52484268/ -> For type A silicon Eval Board use version 1.3.0
4. Enter dev environment from within fwupd git repo source venv/bin/activate
5. `fwupdtool --plugins ccgx --plugins ccgx-dmc install-blob CYUSB4347-BZXC_FW_1_3_0.bin 

**Expected behavior**
Successfully update.

**fwupd version information**
Self compiled on master @ https://github.com/fwupd/fwupd/pull/7528/commits/6c7cfc5c9e38dd00339af8dda2d92f44ea4b77fc
Commands run in dev env
```shell
compile   info.libusb                   1.0.26
compile   org.freedesktop.fwupd         2.0.0
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     2.0.0
runtime   com.hughsie.libxmlb           0.3.x
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.5.0-41-generic
runtime   org.freedesktop.fwupd         2.0.0
```

verbose update output. Note: Seleced here nr. 2 wich is the CY6611 eval board.
Log too long -> See file.
[fwupdInstall-blobCcgx-dmc.log](https://github.com/user-attachments/files/16378606/fwupdInstall-blobCcgx-dmc.log)

ChristianAtSIE, 2024-07-25:
Tried CcgxDmcTriggerCode 0x00, 0x01 and 0x02. Did not change the behavior.

hughsie, 2024-07-25:
@IfxBrent can you tell us what we're supposed to do with `invalid dmc intr req opcode 0x82[reenum] with status 0x02` please? I suspect this correlates with the `Received re-enumeration notification` from HX3PD Configuration Utility -- and I'm *guessing* we need to use `FU_CCGX_DMC_RQT_CODE_SOFT_RESET` and then wait for the device to re-emumerate?

@ChristianAtSIE does it look like the update actually completed with success? You can do a manual reboot by power cycling the evb and then comparing the new versions in the devx subdevices.

ChristianAtSIE, 2024-07-25:
Is there a command to display the HX3PD firmware versions of the devices?

I only see them in between the lines of  `fwupdtool --plugins ccgx_dmc get-devices --show-all --verbose --verbose`

The only thing I notice is that the DMC stays in this state:
```shell
  UpdateModel:          none
  EpBulkOut:            0x1
  EpIntrIn:             0x82
  DeviceStatus:         update-phase1-partial
```

<details>
<summary> Before Power Cycle </summary>

```shell
using 526bff27f4a6d533ebbd2cdf15a888a60abdefd2 for usb:01:00:02:02:08
16:52:59.115 FuPluginCcgxDmc      DmcDockStatus:
       │ 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 
───────┼────────────────────────────────────────────────────────────────────────────────────────────────
0x0000 │ 02 03 68 00 00 00 03 01 02 00 01 02 11 10 04 00 c4 06 00 33 6d 64 e5 10 da 07 01 33 6d 64 72 10 
0x0020 │ da 07 01 33 6d 64 72 10 0b 01 01 02 11 00 00 00 ab 06 00 33 64 6d db 10 da 07 01 33 64 6d 72 10 
0x0040 │ da 07 01 33 64 6d 72 10 0a 02 02 01 11 00 01 00 00 00 00 00 00 00 00 00 58 30 00 00 00 00 00 00 
0x0060 │ 58 30 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x0080 │ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x00a0 │ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x00c0 │ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x00e0 │ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x0100 │ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x0120 │ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x0140 │ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x0160 │ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x0180 │ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x01a0 │ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x01c0 │ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x01e0 │ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x0200 │ 00 00 00 00 00 00 00 00 
16:52:59.115 FuStruct             FuStructCcgxDmcDevxStatus:
  device_type: 0x2 [dmc]
  component_id: 0x0
  image_mode: 0x1 [dual-img-sym]
  current_image: 0x2
  img_status: 0x11
  fw_version: 0xC40600336D64E510DA0701336D647210DA0701336D647210
16:52:59.115 FuDevice             using 7164627a931686ac403811cdeabc548e7f91a1e4 for usb:01:00:02:02:08:0x00
16:52:59.115 FuDevice             using 7164627a931686ac403811cdeabc548e7f91a1e4 for usb:01:00:02:02:08:0x00
16:52:59.115 FuDevice             setting parent of DMC [7164627a931686ac403811cdeabc548e7f91a1e4] to be Dock Management Controller [526bff27f4a6d533ebbd2cdf15a888a60abdefd2]
16:52:59.115 FuStruct             FuStructCcgxDmcDevxStatus:
  device_type: 0xb [dmc-pd]
  component_id: 0x1
  image_mode: 0x1 [dual-img-sym]
  current_image: 0x2
  img_status: 0x11
  fw_version: 0xAB060033646DDB10DA070133646D7210DA070133646D7210
16:52:59.116 FuDevice             using ebc069e68140749e370f734170ff8dc0330b68ed for usb:01:00:02:02:08:0x01
16:52:59.116 FuDevice             using ebc069e68140749e370f734170ff8dc0330b68ed for usb:01:00:02:02:08:0x01
16:52:59.116 FuDevice             setting parent of DMC PD [ebc069e68140749e370f734170ff8dc0330b68ed] to be Dock Management Controller [526bff27f4a6d533ebbd2cdf15a888a60abdefd2]
16:52:59.116 FuStruct             FuStructCcgxDmcDevxStatus:
  device_type: 0xa [hx3-pd]
  component_id: 0x2
  image_mode: 0x2 [dual-img-asym]
  current_image: 0x1
  img_status: 0x11
  fw_version: 0x000000000000000058300000000000005830000000000000
16:52:59.116 FuDevice             using bd4f9d6c43ecbf1722f3fa646a054c73982ca2b2 for usb:01:00:02:02:08:0x02
16:52:59.116 FuDevice             using bd4f9d6c43ecbf1722f3fa646a054c73982ca2b2 for usb:01:00:02:02:08:0x02
16:52:59.116 FuDevice             setting parent of HX3 PD [bd4f9d6c43ecbf1722f3fa646a054c73982ca2b2] to be Dock Management Controller [526bff27f4a6d533ebbd2cdf15a888a60abdefd2]
16:52:59.116 FuPlugin             emit added from ccgx_dmc: 526bff27f4a6d533ebbd2cdf15a888a60abdefd2
16:52:59.116 FuDeviceList         ::added 526bff27f4a6d533ebbd2cdf15a888a60abdefd2 [Dock Management Controller]
16:52:59.117 FuMain               ADDED:
[... 347 lines cut ...]
    LogicalId:          0x02
    BackendId:          01:15
    RemoveDelay:        7320000
    AcquiesceDelay:     2500
    Proxy:
      Guid[quirk]:      5e7ae60b-4b49-534c-a225-aa0111efe8c3 ← USB\VID_04B4
      Guid[quirk]:      4bd986bc-4aaf-5baf-8bc2-9f7928182387 ← USB\CLASS_FF
      Guid[quirk]:      78c9681b-f906-52ee-aaa7-9b58bc4a1505 ← USB\CLASS_FF&SUBCLASS_03
      Guid[quirk]:      d27ab148-1c4b-53f6-9414-d9032eb5abfe ← USB\CLASS_FF&SUBCLASS_03&PROT_00
      PhysicalId:       usb:01:00:02:02:08
      BackendId:        01:15
      RemoveDelay:      7320000
      AcquiesceDelay:   2500
      PossiblePlugin:   ccgx_dmc
      ParentPhysicalIds: usb:01:00:02:02,usb:01:00:02
      InternalFlags:    replug-match-guid|is-open|only-wait-for-replug
    DeviceType:         0xa [hx3-pd]
    ImageMode:          0x2 [dual-img-asym]
    CurrentImage:       0x1
    ImgStatus1:         valid
    ImgStatus2:         valid
    FwVersion[boot]:    0x0000000000000000
    FwVersion[img1]:    0x0000000000003058
    FwVersion[img2]:    0x0000000000003058
´´´

ChristianAtSIE, 2024-07-25:
The update behavior with the windows HX3PD Config Utility tool is, that the DMC firmware #1 is updated, then it waits for reconnect, then DMC firmware #2, reconnect and so on. Each of this "update blocks" takes about 15 to 30sec. 
Trying the update with fwupd, it breaks after about 30sec after selecting the device. So it seems / could be that it breaks after the first update block.

I checked with the windows HX3PD Config Utility the status of the eval board:
```shell
Querying dock status
Opening handle to dock
Connecting to the DMC
=================================HX3PD IDENTITY=================================
CDTT version     : 1
VID              : 0x04B4
PID              : 0x521B
Vendor           : Cypress Semiconductor
Product          : EZ-USB HX3PD Gen 2 Hub
Signed FW Update : FALSE
==============================CURRENT HX3PD STATUS==============================
Last FW update status		: Failed (0x2)
Device count			: 3
Composite dock image version	: 1.3.0.0
--------------------------------------------------------------------------------
	Device#0
	Device type                   : DMC
	Component ID                  : 0
	Current image                 : Image-1
	Bootloader Version
		Base Version          : 3.3.0.1732
		Application Version   : dm.1.0.229
	Image-0 Status                : VALID (0x1)
		Base Version          : 3.3.1.2010
		Application Version   : dm.1.0.114
	Image-1 Status                : VALID (0x1)
		Base Version          : 3.3.1.2010
		Application Version   : dm.1.0.114
--------------------------------------------------------------------------------
	Device#1
	Device type                   : HX3PD PD Controller
	Component ID                  : 1
	Current image                 : Image-1
	Bootloader Version
		Base Version          : 3.3.0.1707
		Application Version   : md.1.0.219
	Image-0 Status                : VALID (0x1)
		Base Version          : 3.3.1.2010
		Application Version   : md.1.0.114
	Image-1 Status                : VALID (0x1)
		Base Version          : 3.3.1.2010
		Application Version   : md.1.0.114
--------------------------------------------------------------------------------
	Device#2
	Device type                   : HX3PD Hub Controller
	Component ID                  : 2
	Current image                 : Image-0
	Image-0 Status                : VALID (0x1)
		Version               : 3058
	Image-1 Status                : VALID (0x1)
		Version               : 3058
================================================================================
```
Then I updated it once again with this tool:
<details>

```shell
Starting update C:\Users\KHCH_Win10_local\Downloads\Infineon-HX3PD_FW_Package_for_RevA_and_RevB_Si.zip-Software-v01_00-EN\EZ-USB_HX3PD_RevASi_FW_V1_3_0\CYUSB4347_Composite_FW\CYUSB4347-BZXC_FW_1_3_0.bin
Loading dock composite image file
Verifying hash of image#0
Segment #0
Segment #1
Verifying hash of image#1
Segment #0
Segment #1
[... 341 lines cut ...]
      PhysicalId:       usb:01:00:02:02:08
      BackendId:        01:17
      RemoveDelay:      7320000
      AcquiesceDelay:   2500
      PossiblePlugin:   ccgx_dmc
      ParentPhysicalIds: usb:01:00:02:02,usb:01:00:02
      InternalFlags:    replug-match-guid|is-open|only-wait-for-replug
    DeviceType:         0xa [hx3-pd]
    ImageMode:          0x2 [dual-img-asym]
    CurrentImage:       0x1
    ImgStatus1:         valid
    ImgStatus2:         valid
    FwVersion[boot]:    0x0000000000000000
    FwVersion[img1]:    0x0000000000003058
    FwVersion[img2]:    0x0000000000003058
```
</details>

And I noticed, that the DeviceStatus changed:
```shell
  UpdateModel:          none
  EpBulkOut:            0x1
  EpIntrIn:             0x82
  DeviceStatus:         update-complete-full
```

IfxBrent, 2024-07-26:
> Caution: This e-mail originated outside Infineon Technologies. Please be cautious when sharing information or opening attachments especially from unknown senders. Refer to our intranet guide<https://intranet-content.infineon.com/explore/aboutinfineon/rules/informationsecurity/ug/SocialEngineering/Pages/SocialEngineeringElements_en.aspx> to help you identify Phishing email.

Opcode 0x82 notifies host a re-enumeration might be necessary for the next step. Host tool can wait for around 5 seconds for any coming intr req opcode and process it.
If there is no new intr req within 5 seconds, tool will wait for device reconnect event for 60 seconds. And if there is no reconnect event is received, it results in an update failure.

So, fwupd does not need to send a FU_CCGX_DMC_RQT_CODE_SOFT_RESE. After 0x82, it can wait 5 seconds for new intr first, and wait another 60 seconds for device reset.

ChristianAtSIE, 2024-07-26:
Okay, I tried to add a workaround changing the function fu_ccgx_dmc_write_firmare from
```shell
if (rqt_opcode != FU_CCGX_DMC_INT_OPCODE_FW_UPGRADE_RQT)
			break;
```
to
```shell
if (rqt_opcode != FU_CCGX_DMC_INT_OPCODE_FW_UPGRADE_RQT){
			if (rqt_opcode == FU_CCGX_DMC_INT_OPCODE_REENUM){//KHCH
				fu_device_add_flag(device, FWUPD_DEVICE_FLAG_WAIT_FOR_REPLUG);
				fu_device_sleep(FU_DEVICE(self), 5000);//ms
				//fu_ccgx_dmc_device_attach(FU_DEVICE(self),progress, error);
				//fu_ccgx_dmc_device_probe(FU_DEVICE(self), error);
				continue;
			}
			break;
		}
```
But this does not work. Watching in a seperate terminal window the usb devices with `watch -n 0.1 lsusb` I see that the Dock Management Controller is disconnected and then reconnected again, but on a different device number:
```shell
Bus 001 Device 036: ID 04b4:521b Cypress Semiconductor Corp. Dock Management Con
troller
Bus 001 Device 022: ID 04b4:2347 Cypress Semiconductor Corp. Cypress HX3PD Hub
Bus 001 Device 031: ID 8564:1000 Transcend Information, Inc. JetFlash
Bus 001 Device 003: ID 2109:2813 VIA Labs, Inc. VL813 Hub
Bus 001 Device 004: ID 04b4:521b Cypress Semiconductor Corp. Dock Management Con
troller
```

ChristianAtSIE, 2024-07-26:
[RenumWaitReturnTrue_FirmwareWriteTest.log](https://github.com/user-attachments/files/16391828/RenumWaitReturnTrue_FirmwareWriteTest.log)
Changing the code to (Got that idea from fu_engine.c line 3489)
```shell
		rqt_opcode = fu_struct_ccgx_dmc_int_rqt_get_opcode(dmc_int_rqt);
		if (rqt_opcode != FU_CCGX_DMC_INT_OPCODE_FW_UPGRADE_RQT){
			if (rqt_opcode == FU_CCGX_DMC_INT_OPCODE_REENUM){//KHCH
				fu_device_add_flag(device, FWUPD_DEVICE_FLAG_ANOTHER_WRITE_REQUIRED);
				fu_device_sleep(FU_DEVICE(self), 5000);//ms
				return TRUE;
			}
			break;
		}
```
I get a 
```shell
12:28:34.031 FuPluginCcgxDmc      DmcIntRqt-opcode=0x81[img-write-status]:00 00 00 
Writing…                 [*******************************        ]12:28:34.191 FuPluginCcgxDmc      DmcIntRqt-opcode=0x82[reenum]:02 00 00 
12:28:39.191 FuProgress           child is at 2/3 step_max and parent done [../../libfwupdplugin/fu-device.c:4756]
4) ../../src/fu-tool.c:935 (2/3)
3) ../../src/fu-engine.c:3452 (1/3)
2) ../../plugins/ccgx-dmc/fu-ccgx-dmc-device.c:820 (1/4)
1) ../../libfwupdplugin/fu-device.c:4756 (1/2)
0) ../../plugins/ccgx-dmc/fu-ccgx-dmc-device.c:534 (2/3)

12:28:39.192 FuUsbDevice          re-binding kernel driver as not waiting for replug
12:28:39.192 FuUsbDevice          failed to release interface 0x00: USB error: No such device (it may have been disconnected) [-4]
12:28:39.192 FuDeviceList         no replug or re-enumerate required
Restarting device…       [*******************************        ]12:28:39.192 FuEngine             install phase now attach
12:28:39.192 FuDeviceList         no replug or re-enumerate required
12:28:39.192 FuEngine             attach -> FuCcgxDmcDevice:
  DeviceId:             526bff27f4a6d533ebbd2cdf15a888a60abdefd2
  Name:                 Dock Management Controller
  Status:               device-restart
  Guid:                 fae414e5-0a7d-57c2-9e5a-74cfd6cdf1f2 ← USB\VID_04B4&PID_521B
  Serial:               11AD1D82C8592F0D1F0B0B00
  Plugin:               ccgx_dmc
  Protocol:             com.cypress.ccgx.dmc
  Protocol:             com.infineon.ccgx.dmc
  Flags:                require-ac|registered|another-write-required|dual-image|self-recovery|updatable-hidden|unsigned-payload
  Problems:             update-in-progress
  Vendor:               Cypress Semiconductor Corp.
  VendorId:             USB:0x04B4
  Version:              1.3.0.0
  VersionFormat:        quad
  VersionRaw:           0x01030000
  Created:              2024-07-26
  Modified:             2024-07-26
  UpdateError:          An update is in progress
  Guid[quirk]:          5e7ae60b-4b49-534c-a225-aa0111efe8c3 ← USB\VID_04B4
  Guid[quirk]:          4bd986bc-4aaf-5baf-8bc2-9f7928182387 ← USB\CLASS_FF
  Guid[quirk]:          78c9681b-f906-52ee-aaa7-9b58bc4a1505 ← USB\CLASS_FF&SUBCLASS_03
  Guid[quirk]:          d27ab148-1c4b-53f6-9414-d9032eb5abfe ← USB\CLASS_FF&SUBCLASS_03&PROT_00
  PhysicalId:           usb:01:00:02:02:08
  BackendId:            01:1b
  RemoveDelay:          7320000
  AcquiesceDelay:       2500
  Order:                0
  PossiblePlugin:       ccgx_dmc
  ParentPhysicalIds:    usb:01:00:02:02,usb:01:00:02
  InternalFlags:        replug-match-guid|only-wait-for-replug
  Inhibit:              [update-in-progress] An update is in progress
  Events:
  InterfaceNumber#00:   released
  Class:                interface-desc
  Interfaces:
    FuUsbInterface:
        InterfaceClass: 0xff
        Length:         0x9
        DescriptorType: 0x4
        InterfaceSubClass: 0x3
[... 122 lines cut ...]
      BackendId:        01:1b
      RemoveDelay:      7320000
      AcquiesceDelay:   2500
      Order:            0
      PossiblePlugin:   ccgx_dmc
      ParentPhysicalIds: usb:01:00:02:02,usb:01:00:02
      InternalFlags:    replug-match-guid|only-wait-for-replug
      Inhibit:          [update-in-progress] An update is in progress
      Events:
    DeviceType:         0xa [hx3-pd]
    ImageMode:          0x2 [dual-img-asym]
    CurrentImage:       0x1
    ImgStatus1:         valid
    ImgStatus2:         valid
    FwVersion[boot]:    0x0000000000000000
    FwVersion[img1]:    0x0000000000003058
    FwVersion[img2]:    0x0000000000003058

12:28:39.193 FuPlugin             attach(ccgx_dmc)
12:28:39.193 FuDeviceLocker       ignoring close error on aborted open: device 04b4:521b has not been opened

failed to attach: failed to attach using ccgx_dmc: failed to subclass open: failed to open device: USB error: No such device (it may have been disconnected) [-4]
12:28:39.193 FuMain               failed to attach: failed to attach using ccgx_dmc: failed to subclass open: failed to open device: USB error: No such device (it may have been disconnected) [-4]
```
It does no longer try to read the IRQ, but just continues to the attach phase.

ChristianAtSIE, 2024-07-26:
[RenumNoWaitSetAnotherInstallSetReplug_FirmwareWriteTest.log](https://github.com/user-attachments/files/16392070/RenumNoWaitSetAnotherInstallSetReplug_FirmwareWriteTest.log)

Changing to this
```shell
		if (rqt_opcode != FU_CCGX_DMC_INT_OPCODE_FW_UPGRADE_RQT){
			if (rqt_opcode == FU_CCGX_DMC_INT_OPCODE_REENUM){//KHCH
				fu_device_add_flag(device, FWUPD_DEVICE_FLAG_ANOTHER_WRITE_REQUIRED);
				fu_device_add_flag(device, FWUPD_DEVICE_FLAG_WAIT_FOR_REPLUG);
				return TRUE;
			}
			break;
		}
```
and the quirk `RemoveDelay = 10000`

I get this 
```shell
12:53:51.093 FuPluginCcgxDmc      DmcIntRqt-opcode=0x81[img-write-status]:00 00 00 
Writing…                 [*******************************        ]12:53:51.253 FuPluginCcgxDmc      DmcIntRqt-opcode=0x82[reenum]:02 00 00 
12:53:51.253 FuProgress           child is at 2/3 step_max and parent done [../../libfwupdplugin/fu-device.c:4756]
4) ../../src/fu-tool.c:935 (2/3)
3) ../../src/fu-engine.c:3452 (1/3)
2) ../../plugins/ccgx-dmc/fu-ccgx-dmc-device.c:818 (1/4)
1) ../../libfwupdplugin/fu-device.c:4756 (1/2)
0) ../../plugins/ccgx-dmc/fu-ccgx-dmc-device.c:532 (2/3)

12:53:51.253 FuDeviceList         waiting 10000ms for replug
12:53:51.254 FuEngine             writing motd target /home/phoenix/fwupd/venv/bin/../dist/cache/fwupd/motd.d/85-fwupd
12:53:51.936 FuEngine             usb removed 01:2e
12:53:51.936 FuEngine             auto-removing backend device Dock Management Controller [526bff27f4a6d533ebbd2cdf15a888a60abdefd2]
12:53:51.936 FuDeviceList         waiting 10000ms for Dock Management Controller device removal
12:53:51.936 FuEngine             resetting update motd timeout
12:53:51.937 FuEngine             auto-removing backend device DMC [7164627a931686ac403811cdeabc548e7f91a1e4]
12:53:51.937 FuDeviceList         waiting 10000ms for DMC device removal
12:53:51.937 FuEngine             resetting update motd timeout
12:53:51.937 FuEngine             auto-removing backend device DMC PD [ebc069e68140749e370f734170ff8dc0330b68ed]
12:53:51.938 FuDeviceList         waiting 10000ms for DMC PD device removal
12:53:51.938 FuEngine             resetting update motd timeout
12:53:51.940 FuEngine             auto-removing backend device HX3 PD [bd4f9d6c43ecbf1722f3fa646a054c73982ca2b2]
12:53:51.940 FuDeviceList         waiting 10000ms for HX3 PD device removal
12:53:51.940 FuEngine             resetting update motd timeout
12:53:52.841 FuEngine             01:2f added FuUsbDevice:
  Flags:                none
  BackendId:            01:2f
  AcquiesceDelay:       2500
  Class:                interface-desc

12:53:52.843 FuUsbDevice          ignoring: not available as bcdUSB 0x0200 <= 0x0200
12:53:52.843 FuEngine             01:2f added FuUsbDevice:
  Guid:                 fae414e5-0a7d-57c2-9e5a-74cfd6cdf1f2 ← USB\VID_04B4&PID_521B ⚠
  Flags:                none
  Vendor:               Cypress Semiconductor Corp.
  VendorId:             USB:0x04B4
  Guid[quirk]:          5e7ae60b-4b49-534c-a225-aa0111efe8c3 ← USB\VID_04B4
  Guid[quirk]:          4bd986bc-4aaf-5baf-8bc2-9f7928182387 ← USB\CLASS_FF
  Guid[quirk]:          78c9681b-f906-52ee-aaa7-9b58bc4a1505 ← USB\CLASS_FF&SUBCLASS_03
  Guid[quirk]:          d27ab148-1c4b-53f6-9414-d9032eb5abfe ← USB\CLASS_FF&SUBCLASS_03&PROT_00
  PhysicalId:           usb:01:00:02:02:08
  BackendId:            01:2f
  RemoveDelay:          10000
  AcquiesceDelay:       2500
  PossiblePlugin:       ccgx_dmc
  ParentPhysicalIds:    usb:01:00:02:02,usb:01:00:02
  BusNum:               0x1
  DevNum:               0x2f
  Class:                interface-desc
  Interfaces:
    FuUsbInterface:
        InterfaceClass: 0xff
        Length:         0x9
[... 444 lines cut ...]
new: aebed407064b8b8b8ba5a3df7a274247ef0bf7af [0x5cc40f9296c0] 
3 [0x5cc40f9319e0] 
new: 3b880d6eb74481ba77aafc27b609adc1103c42a1 [0x5cc40f927f40] 
4 [0x5cc40f955610] 
new: 526bff27f4a6d533ebbd2cdf15a888a60abdefd2 [0x5cc40f9a41e0] WAIT_FOR_REPLUG
old: 526bff27f4a6d533ebbd2cdf15a888a60abdefd2 [0x5cc40f94f050] 
5 [0x5cc40f9552a0] 
new: 7164627a931686ac403811cdeabc548e7f91a1e4 [0x5cc40f97a150] 
old: 7164627a931686ac403811cdeabc548e7f91a1e4 [0x5cc40f952720] 
6 [0x5cc40f955240] 
new: ebc069e68140749e370f734170ff8dc0330b68ed [0x5cc40f85aaf0] 
old: ebc069e68140749e370f734170ff8dc0330b68ed [0x5cc40f953850] 
7 [0x5cc40f94f220] 
new: bd4f9d6c43ecbf1722f3fa646a054c73982ca2b2 [0x5cc40f9a9f30] 
old: bd4f9d6c43ecbf1722f3fa646a054c73982ca2b2 [0x5cc40f954140] 

failed to attach: failed to wait for attach replug: device 526bff27f4a6d533ebbd2cdf15a888a60abdefd2 did not come back
12:54:02.883 FuMain               failed to attach: failed to wait for attach replug: device 526bff27f4a6d533ebbd2cdf15a888a60abdefd2 did not come back

12:54:02.883 FuPluginTest         destroy
12:54:02.885 FuProgress           raw timing data was { 0.001, 0.000, 0.001, 0.000 } -- steps were set as [ 1 1 3 95 ] but should have been [ 66 4 27 2 ] at ../../libfwupdplugin/fu-context.c:1010
```
So it seems that the device is now seen to be renumerating. But also not, as it later tries to go on to the attach phase.

I also tried to unplug the device in the 10 sec replug window: Then the update is repeated, but starting every time with segment 0/5.

ChristianAtSIE, 2024-07-26:
@IfxBrent @hughsie Is there any information, on what hardware this plugin was tested on? Was it ever tested with the HX3PD controller CYUSB4347?

hughsie, 2024-07-26:
@ChristianAtSIE I'm about to disappear for 10 days on holiday, but I'm happy for you to keep going here with @IfxBrent -- the hardware we're testing with are all actual dock devices, e.g. the HP G5 and things like that. When I get back maybe I can help more, apologies.

hughsie, 2024-11-11:
I don't think I have much more to add here, I think @IfxBrent is the only one that can help.

closed 2024-11-11 by hughsie