fwupd/fwupd #7533: Pre-boot DMA protection HSI check is not accurate

chrisccoulson, 2024-07-25:
On Intel devices, fwupd checks the `DMA_CTRL_PLATFORM_OPT_IN_FLAG` bit in the `DMAR` ACPI table to determine if pre-boot DMA protection is enabled as part of the HSI checks. But this flag seems to only be a hint to the OS as to whether it should enable DMA protection, rather than as an indication of how the pre-boot environment was configured.

In the kernel, this bit is used to force enable the IOMMU even if `iommu=off` is passed on the commandline, although Ubuntu's kernel enables the IOMMU regardless of this bit in the `DMAR` ACPI table because it is built with ` CONFIG_INTEL_IOMMU_DEFAULT_ON=y`. With `CONFIG_INTEL_IOMMU_DEFAULT_ON=n` and without the `DMA_CTRL_PLATFORM_OPT_IN_FLAG` bit set, the kernel doesn't configure the IOMMU.

On my Dell XPS15, there are 2 firmware options related to DMA protection - I can toggle on or off pre-boot DMA protection and OS DMA protection independently.

If I turn off just the pre-boot DMA protection option, this doesn’t change the value of `DMA_CTRL_PLATFORM_OPT_IN_FLAG` and the HSI check still passes when it should fail. Turning off the OS DMA protection option instead (whilst keeping the pre-boot DMA protection option enabled) does disable the `DMA_CTRL_PLATFORM_OPT_IN_FLAG` bit in the `DMAR` ACPI table. In this case, the pre-boot DMA protection HSI check fails unexpectedly, because it’s actually enabled as per the firmware settings. The IOMMU HSI test also continues to pass in this case because as I mentioned previously, the Ubuntu kernel ignores the firmware hint and still enables it regardless of the value of the `DMA_CTRL_PLATFORM_OPT_IN_FLAG` bit.

Whenever DMA protection is degraded via the firmware settings on my Dell XPS15, either during the pre-boot environment or by turning off the hint that the OS uses via the `DMAR` ACPI table, I get an additional `EV_EFI_ACTION` event with the data “DMA Protection Disabled” measured to PCR7 as part of the OS-present environment. This doesn’t seem to be documented in any TCG spec, but is referenced in some Tianocore documentation (https://tianocore-docs.github.io/edk2-TrustedBootChain/release-1.00/3_TCG_Trusted_Boot_Chain_in_EDKII.html ) which references it as a Windows requirement, although the only Windows requirement I can find is https://learn.microsoft.com/en-us/windows-hardware/design/device-experiences/oem-kernel-dma-protection.

I’m not sure what the best heuristic is for detecting whether pre-boot DMA protection is enabled as part of the HSI checks. If the IOMMU HSI check passes but PCR7 contains this additional `EV_EFI_ACTION` “DMA Protection Disabled” event, then it's probably reasonable to assume that pre-boot DMA protection is disabled, but I'm not sure if this is a reliable test. I don't know if the additional PCR measurement is universally implemented by other OEMs, and I can't find an implementation of this in EDK2 or any of the public Microsoft Project Mu repositories.

This was tested with fwupd 1.9.16

hughsie, 2024-07-25:
If you look at it from the scope of "if the firmware doing the right thing" -- I think the fwupd checks make sense; but isn't preboot in this context for when the linux kernel hasn't started yet?

superm1, 2024-07-25:
From a 10k foot view I agree with @hughsie , but I also think you found a real problem here that could be a "bigger deal than it seems".

It's a bit weird to hear that this is split in two separate knobs. The whole point of it is that the firmware asserts that it was safe for the boot process and the OS keys off this bit you referenced from the ACPI table to know that it can do DMA to the device.

Having two knobs totally breaks that contract.
* For Windows I suppose by extending PCR7 this means you no longer automatically decrypt bitlocker from this change.
* But for Linux this means that the existing policy that boltd and thunderbolt.ko use is no longer valid. I actually think the kernel should catch this and the sysfs file `iommu_dma_protection` shouldn't be 1 in the case you've outlined.

Can you give us some snapshots of the tpm event log with that event asserted?

I don't see why we can't flag that event too from fwupd, and having your system event log for the test suite makes good sense.

chrisccoulson, 2024-08-14:
My Dell has 2 firmware settings:
- Enable Pre-Boot DMA Support
- Enable OS Kernel DMA Support

With them both enabled, PCR7 looks as one would expect:
```
PCR  DIGEST                                                            TYPE                           DETAILS
7    ccfc4bb32888a345bc8aeadaba552b627d99348c767681ab3141f5b01e40a40e  EV_EFI_VARIABLE_DRIVER_CONFIG  SecureBoot: 1
7    2abfe9865a654102acb12f0fefe52dc4d01bce40901410eb3dadaf212700a2b7  EV_EFI_VARIABLE_DRIVER_CONFIG  PK: entries(x509)=1
7    63a525134bfbc242058c0e6b42794f8b1d142d13029a9aa38a3272c5ca2390c5  EV_EFI_VARIABLE_DRIVER_CONFIG  KEK: entries(x509)=2
7    58a0c07f6f04c9d61afe2b70e99bc3caf569856119bd543cebae7dcd9ac2c7d1  EV_EFI_VARIABLE_DRIVER_CONFIG  db: entries(x509)=4
7    89fe668802a699be6aa74ec18cbd80cfc8d4f1da222014844175c028cb28d61b  EV_EFI_VARIABLE_DRIVER_CONFIG  dbx: entries(x509)=1 entries(sha256)=421
7    df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119  EV_SEPARATOR                   
7    4d4a8e2c74133bbdc01a16eaf2dbb5d575afeb36f5d8dfcf609ae043909e2ee9  EV_EFI_VARIABLE_AUTHORITY      subject: "CN=Microsoft Corporation UEFI CA 2011,O=Microsoft Corporation,L=Redmond,ST=Washington,C=US", owner: 77fa9abd-0359-4d32-bd60-28f4e78f784b, source: db
7    e8e9578f5951ef16b1c1aa18ef02944b8375ec45ed4b5d8cdb30428db4a31016  EV_EFI_VARIABLE_AUTHORITY      SbatLevel: sbat,1,2023012900 ...
7    68bdff38e48c399326ca7356eb992693d13301f3925caf10e7b39dc9240789cd  EV_EFI_VARIABLE_AUTHORITY      subject: "CN=Canonical Ltd. Master Certificate Authority,O=Canonical Ltd.,L=Douglas,ST=Isle of Man,C=GB", owner: 605dab50-e046-4300-abb6-3dd810dd8b23, source: MokListRT
```

And the decompiled DMAR table looks as expected, with the `DMA_CTRL_PLATFORM_OPT_IN_FLAG` set - here's the relevant part of it:
```
[000h 0000 004h]                   Signature : "DMAR"    [DMA Remapping Table]
[004h 0004 004h]                Table Length : 00000088
[008h 0008 001h]                    Revision : 02  
[009h 0009 001h]                    Checksum : 51
[00Ah 0010 006h]                      Oem ID : "INTEL "
[010h 0016 008h]                Oem Table ID : "Dell Inc"
[018h 0024 004h]                Oem Revision : 00000002
[01Ch 0028 004h]             Asl Compiler ID : "    "
[020h 0032 004h]       Asl Compiler Revision : 01000013
                                                                    
[024h 0036 001h]          Host Address Width : 26
[025h 0037 001h]                       Flags : 05
```

If I turn off the "Enable Pre-Boot DMA Support" option in the firmware settings, nothing in the DMAR table changes. The  `DMA_CTRL_PLATFORM_OPT_IN_FLAG` is still set and the `Pre-boot DMA protection` HSI check is still marked as enabled. However, PCR7 looks like this:
```
PCR  DIGEST                                                            TYPE                           DETAILS
7    ccfc4bb32888a345bc8aeadaba552b627d99348c767681ab3141f5b01e40a40e  EV_EFI_VARIABLE_DRIVER_CONFIG  SecureBoot: 1
7    2abfe9865a654102acb12f0fefe52dc4d01bce40901410eb3dadaf212700a2b7  EV_EFI_VARIABLE_DRIVER_CONFIG  PK: entries(x509)=1
7    63a525134bfbc242058c0e6b42794f8b1d142d13029a9aa38a3272c5ca2390c5  EV_EFI_VARIABLE_DRIVER_CONFIG  KEK: entries(x509)=2
7    58a0c07f6f04c9d61afe2b70e99bc3caf569856119bd543cebae7dcd9ac2c7d1  EV_EFI_VARIABLE_DRIVER_CONFIG  db: entries(x509)=4
7    89fe668802a699be6aa74ec18cbd80cfc8d4f1da222014844175c028cb28d61b  EV_EFI_VARIABLE_DRIVER_CONFIG  dbx: entries(x509)=1 entries(sha256)=421
7    df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119  EV_SEPARATOR                   
7    5b2b1f0c5470397d4efa2fe23110c8b6f61e299b9fa2c098f834ff06416196c3  EV_EFI_ACTION                  
7    4d4a8e2c74133bbdc01a16eaf2dbb5d575afeb36f5d8dfcf609ae043909e2ee9  EV_EFI_VARIABLE_AUTHORITY      subject: "CN=Microsoft Corporation UEFI CA 2011,O=Microsoft Corporation,L=Redmond,ST=Washington,C=US", owner: 77fa9abd-0359-4d32-bd60-28f4e78f784b, source: db
7    e8e9578f5951ef16b1c1aa18ef02944b8375ec45ed4b5d8cdb30428db4a31016  EV_EFI_VARIABLE_AUTHORITY      SbatLevel: sbat,1,2023012900 ...
7    68bdff38e48c399326ca7356eb992693d13301f3925caf10e7b39dc9240789cd  EV_EFI_VARIABLE_AUTHORITY      subject: "CN=Canonical Ltd. Master Certificate Authority,O=Canonical Ltd.,L=Douglas,ST=Isle of Man,C=GB", owner: 605dab50-e046-4300-abb6-3dd810dd8b23, source: MokListRT
```

Notice the extra `EV_EFI_ACTION` event - I'm not sure why my tool isn't decoding it, I think it's because `EV_EFI_ACTION` event data is not meant to be NULL terminated but this contains a NULL terminated string. Looking at the hexdump for this specific event:
```
PCR: 7                                                                                                                                  
TYPE: EV_EFI_ACTION
DIGEST(TPM_ALG_SHA256): 5b2b1f0c5470397d4efa2fe23110c8b6f61e299b9fa2c098f834ff06416196c3
DETAILS:
EVENT DATA:                                                                                                     
        00000000  44 4d 41 20 50 72 6f 74  65 63 74 69 6f 6e 20 44  |DMA Protection D|
        00000010  69 73 61 62 6c 65 64 00                           |isabled.|
```

If I turn off the "Enable OS Kernel DMA Support" option in the firmware settings instead, the `DMA_CTRL_PLATFORM_OPT_IN_FLAG` goes away from the DMAR ACPI table and the `Pre-boot DMA protection` HSI check then subsequently fails. Here's the relevant part of the decompiled DMAR table:
```
[... cut ...]
[010h 0016 008h]                Oem Table ID : "Dell Inc"
[018h 0024 004h]                Oem Revision : 00000002
[01Ch 0028 004h]             Asl Compiler ID : "    "
[020h 0032 004h]       Asl Compiler Revision : 01000013
                                                                    
[024h 0036 001h]          Host Address Width : 26
[025h 0037 001h]                       Flags : 01
```

PCR7 looks the same in both cases, whether I turn off "pre-boot" or "OS kernel" support options in the firmware settings.

chrisccoulson, 2024-08-14:
[fwupd-7533.tar.gz](https://github.com/user-attachments/files/16609395/fwupd-7533.tar.gz)

I've attached a tarball with a set of binary TCG logs and binary DMAR tables from all 3 test cases that you can use for testing in fwupd.

chrisccoulson, 2024-08-14:
> If you look at it from the scope of "if the firmware doing the right thing" -- I think the fwupd checks make sense; but isn't preboot in this context for when the linux kernel hasn't started yet?

Yeah, "preboot" is a bit of an overloaded term and I wish it would be more specific as to what it means. However, in this context, I suspect "preboot" is referring to the "pre-OS" part of the boot in TCG language (ie, everything up to when the ready to boot signal happens and we get `EV_SEPARATOR` event measured to PCRs 0-6 and the GPT for the initial boot device measured to PCR5 - at which point BDS has already processed `DriverXXXX` and `SysPrepXXXX` entries but hasn't loaded the initial OS loader yet. That's just a guess though, but I can't imagine it meaning anything else.