fwupd/fwupd #7544: Support for 8bitdo Retro Keyboard

vanfanel, 2024-07-27:
**Is your feature request related to a problem? Please describe.**
The fantastic 8bitdo Retro Keyboard doesn't appear at all on the devices listed when doing:
`fwupdmgr get-devices`

**Describe the solution you'd like**
Ideally, it should appear there and it should be possible to update it's firmware with `fwupdmgr get-updates`

**Describe alternatives you've considered**
I have no alternatives. Using Windows is not an alternative for me.

**Additional context**
lsusb says this about the keyboard (only when it's set to OFF):
`Bus 001 Device 008: ID 2dc8:5200 8BitDo 8BitDo Retro Keyboard`

vanfanel, 2024-11-11:
@hughsie I tried to contact them, but they ignored my emails too. Not exactly a company whose products I would buy again, tbh.
Do you happen to know if there is a standard method to achieve the update? As in dd-ing fw blobs to /dev nodes or similar...

vanfanel, 2024-11-11:
@hughsie Just for fun, this is the last e-mail I sent them abut the matter:

```
Hello,

I have an 8bitdo Retro Keyboard whose firmware I would like to update on GNU/Linux.
Are you planning to support FWUPD on Linux? Is there any other procedure to update manually on GNU/Linux?
PLEASE don't tell me "all we support is Windows": My future acquisitions of 8BitDo hardware (and the acquisitions of my friends and relatives) totally depend on being able to update on Linux.

Thanks

```

They never answered but I will be bombing it for some days to try my luck, if they don't give us a solution, then 8BitDo is banned forever for me and friends.

hughsie, 2024-11-11:
@vanfanel many thanks! Rock on!

vanfanel, 2024-11-12:
Look at their response:
```

Hi,

Sorry, not compatible.
```

Time to move on from those idiots I guess.

closed 2025-05-03 by hughsie