fwupd/fwupd #7546: Can't update firmware - Secure boot is enabled, but shim isn't installed to EFI/ubuntu/shimx64.efi

thomasbuilds, 2024-07-28:
Hello you can find all the details on this Ask ubuntu post https://askubuntu.com/questions/1521529/cant-update-firmware-secure-boot-is-enabled-but-shim-isnt-installed-to-efi

thanks ;)

mikbini, 2024-08-03:
I had the same symptoms and for me the reason is that I have 2 EFI partitions on my laptop and one of the two (labelled `PQSERVICE`, I think it's used for system recovery) is indeed missing shimx64.efi (and other EFI files).

As far as I can tell the `PQSERVICE` partition has been there since forever but fwupd recently started picking it over the "good" one (the "good" one being the one that is mounted on `/boot/efi`) for some unknown reason - maybe some change in udisks but I really don't know.

This was reported previously as [fwupd issue 7035](https://github.com/fwupd/fwupd/issues/7035), there are steps there to confirm that you indeed have the same problem.

Anwyay I solved it by adding

```
[fwupd]
EspLocation=/boot/efi
```

to `/etc/fwupd/fwupd.conf` as per one of the suggestions in [this comment](https://github.com/fwupd/fwupd/issues/7035#issuecomment-2038772735)

even though a more proper solution would be arguably to exclude the recovery partition to be used by fwupd.

I hope it helps.

superm1, 2024-08-03:
> I had the same symptoms and for me the reason is that I have 2 EFI partitions on my laptop and one of the two (labelled `PQSERVICE`, I think it's used for system recovery) is indeed missing shimx64.efi (and other EFI files).
> 
> As far as I can tell the `PQSERVICE` partition has been there since forever but fwupd recently started picking it over the "good" one (the "good" one being the one that is mounted on `/boot/efi`) for some unknown reason - maybe some change in udisks but I really don't know.
> 
> This was reported previously as [fwupd issue 7035](https://github.com/fwupd/fwupd/issues/7035), there are steps there to confirm that you indeed have the same problem.
> 
> Anwyay I solved it by adding
> 
> ```
> [fwupd]
> EspLocation=/boot/efi
> ```
> 
> to `/etc/fwupd/fwupd.conf` as per one of the suggestions in [this comment](https://github.com/fwupd/fwupd/issues/7035#issuecomment-2038772735)
> 
> even though a more proper solution would be arguably to exclude the recovery partition from being used by fwupd.
> 
> I hope it helps.
> 

What version of fwupd? We added code to exclude it a few versions ago.

mikbini, 2024-08-03:
fwupd 1.9.16, which comes with ubuntu

```
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.16
compile   com.hughsie.libxmlb           0.3.17
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.8.0-39-generic
runtime   org.freedesktop.fwupd         1.9.16
```

superm1, 2024-08-08:
Yeah; that's a bit old from an upstream perspective.  I thought we backported this stuff in the last few releases.  If you want to check out the 1_9_X branch (which has the 1.9.23 release) and build it you can see if it's fixed.  The whole thing can run in a venv after you install build deps, so you shouldn't break your regular install trying it.

mikbini, 2024-08-08:
I did compile 1.9.24 (almost: I complied the head of the 1_9_X branch) even though I updated the firmware in the meanwhile and, as currently there's no update for this system, I cannot trigger the error anymore, not even with 1.9.16.

Anyway if I understand correctly fwupd 1.9.24 still doesn't filter out the `PQSERVICE` partition:

```
[...]
17:08:22.811 FuPluginUefiCapsule  found /boot/efi/EFI/BOOT/grubx64.efi which indicates a Linux ESP, using /boot/efi
17:08:22.811 FuVolume             mounting /org/freedesktop/UDisks2/block_devices/nvme0n1p2
17:08:22.931 FuPluginUefiCapsule  found /media/root/PQSERVICE/EFI/boot/grubx64.efi which indicates a Linux ESP, using /media/root/PQSERVICE
17:08:22.931 FuVolume             unmounting /org/freedesktop/UDisks2/block_devices/nvme0n1p2
17:08:22.963 FuPluginUefiCapsule  more than one ESP possible:
 - 0x31733:     /org/freedesktop/UDisks2/block_devices/nvme0n1p2
 - 0x30258:     /org/freedesktop/UDisks2/block_devices/nvme0n1p1
[...]
```
And then when it lists the ESP that it picks it always selects `nvme0n1p2`
```
  EspId:                /org/freedesktop/UDisks2/block_devices/nvme0n1p2
```

Also, in 1.9.16 `PQSERVICE` was already listed as a partition to filter out by UEFI capsule plugin https://github.com/fwupd/fwupd/blob/1.9.16/plugins/uefi-capsule/fu-uefi-capsule-plugin.c#L611 so I'm inclined to think that the check is not working correctly.

mikbini, 2024-08-08:
Full output of `fwupdtool get-devices -vv` in case it's needed [get-devices.txt](https://github.com/user-attachments/files/16551680/get-devices.txt)

superm1, 2024-08-08:
Hmm can you double check main? I wonder if we're missing some backports still .

mikbini, 2024-08-08:
Same with main

```
$ fwupdtool --version
Supported debug values: gc-friendly fatal-warnings fatal-criticals all help
Supported debug values: resident-modules bind-now-modules all help
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.0
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
runtime   com.hughsie.libxmlb           0.3.x
runtime   com.hughsie.libjcat           0.2.0
runtime   org.kernel                    6.8.0-39-generic
runtime   org.freedesktop.fwupd         2.0.0
```

Full output of `fwupdtool get-devices -vv` here [get-devices-main.txt](https://github.com/user-attachments/files/16551779/get-devices-main.txt)

superm1, 2024-08-08:
Yeah it certainly looks like it's not properly excluding it still even on main.  I'll see if I can figure out why.

```
18:01:59.269 FuVolume             Looking for volumes of type c12a7328-f81f-11d2-ba4b-00a0c93ec93b
18:01:59.274 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
18:01:59.295 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p3, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: ext4
18:01:59.305 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
18:01:59.329 FuVolume             Looking for volumes of type ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
18:01:59.333 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
18:01:59.351 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p3, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: ext4
18:01:59.361 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
18:01:59.364 FuContext            no volumes of type ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
18:01:59.364 FuContext            found /boot/efi/EFI/BOOT/grubx64.efi which indicates a Linux ESP, using /boot/efi
18:01:59.364 FuVolume             mounting /org/freedesktop/UDisks2/block_devices/nvme0n1p2
18:01:59.479 FuContext            found /media/root/PQSERVICE/EFI/boot/grubx64.efi which indicates a Linux ESP, using /media/root/PQSERVICE
18:01:59.479 FuVolume             unmounting /org/freedesktop/UDisks2/block_devices/nvme0n1p2
18:01:59.507 FuContext            more than one ESP possible:
 - 0x31733:	/org/freedesktop/UDisks2/block_devices/nvme0n1p2
 - 0x30258:	/org/freedesktop/UDisks2/block_devices/nvme0n1p1
```

superm1, 2024-08-08:
Can I please ask for `udisksctl dump` output as well?   I expect the `Name` key on the partition is `PQSERVICE` but maybe it's not.

mikbini, 2024-08-08:
It is indeed `PQSERVICE`:

```
[...]
/org/freedesktop/UDisks2/block_devices/nvme0n1p2:
  org.freedesktop.UDisks2.Block:
    Configuration:              []
    CryptoBackingDevice:        '/'
    Device:                     /dev/nvme0n1p2
[...]
    IdLabel:                    PQSERVICE
[...]
```

Full output of `udisksctl dump` [udiskctl-dump.txt](https://github.com/user-attachments/files/16553236/udiskctl-dump.txt)

superm1, 2024-08-08:
Ah... I see it's not populated in the `Name` key on the partition for some reason.  It's only populated on the `IdLabel` key for the block device.  Not sure why though.  Does it change while it's mounted?

mikbini, 2024-08-08:
'Name' is empty even after mounting the partition.

```
/org/freedesktop/UDisks2/block_devices/nvme0n1p2:
  org.freedesktop.UDisks2.Block:
    Configuration:              []
    CryptoBackingDevice:        '/'
    Device:                     /dev/nvme0n1p2
    DeviceNumber:               66306
    Drive:                      '/org/freedesktop/UDisks2/drives/WD_PC_SN740_SDDQMQE_2T00_1201_24025K402765'
    HintAuto:                   false
    HintIconName:               
    HintIgnore:                 true
    HintName:                   
    HintPartitionable:          true
    HintSymbolicIconName:       
    HintSystem:                 true
    Id:                         by-id-nvme-WD_PC_SN740_SDDQMQE-2T00-1201_24025K402765-part2
    IdLabel:                    PQSERVICE
    IdType:                     vfat
    IdUUID:                     6259-A6FF
    IdUsage:                    filesystem
    IdVersion:                  FAT32
    MDRaid:                     '/'
    MDRaidMember:               '/'
    PreferredDevice:            /dev/nvme0n1p2
    ReadOnly:                   false
    Size:                       6227701760
    Symlinks:                   /dev/disk/by-diskseq/9-part2
                                /dev/disk/by-id/nvme-WD_PC_SN740_SDDQMQE-2T00-1201_24025K402765-part2
                                /dev/disk/by-id/nvme-WD_PC_SN740_SDDQMQE-2T00-1201_24025K402765_1-part2
                                /dev/disk/by-id/nvme-eui.e8238fa6bf530001001b444a41b167dc-part2
                                /dev/disk/by-label/PQSERVICE
                                /dev/disk/by-partuuid/d68e2598-3e81-459c-a9da-c9c3e2021590
                                /dev/disk/by-path/pci-0000:02:00.0-nvme-1-part2
                                /dev/disk/by-uuid/6259-A6FF
    UserspaceMountOptions:      uhelper=udisks2
  org.freedesktop.UDisks2.Filesystem:
    MountPoints:        /media/michele/PQSERVICE
    Size:               0
  org.freedesktop.UDisks2.Partition:
    Flags:              0
    IsContained:        false
    IsContainer:        false
    Name:               
    Number:             2
    Offset:             630194176
    Size:               6227701760
    Table:              '/org/freedesktop/UDisks2/block_devices/nvme0n1'
    Type:               c12a7328-f81f-11d2-ba4b-00a0c93ec93b
    UUID:               d68e2598-3e81-459c-a9da-c9c3e2021590
```

Actually the only partition with a non-empty `Name` key is the "right" EFI partition `nvme0n1p1`. Even e.g. my root partition has `IdLabel: Ubuntu` but an empty `Name`.

closed 2024-08-09 by superm1

mikbini, 2024-08-11:
Hi @superm1, I see that https://github.com/fwupd/fwupd/pull/7586 is tagged "backport" but I'm not familiar with this project: what are the plans to backport it to the 1_9_X branch?

superm1, 2024-08-11:
> Hi @superm1, I see that https://github.com/fwupd/fwupd/pull/7586 is tagged "backport" but I'm not familiar with this project: what are the plans to backport it to the 1_9_X branch?

I tried but it's a relatively difficult backport because of contextual changes. @hughsie can you try?