fwupd/fwupd #7547: TPM v2.0 not found after last updates

iyanmv, 2024-07-29:
**Describe the bug**
I don't know if it happened after updating to 1.9.21 or 1.9.22, but `fwupdmgr security` now reports TPM v2.0 as not found.

```
Host Security ID: HSI:0 (v1.9.22)

HSI-1
✔ BIOS firmware updates:         Enabled
✔ csme manufacturing mode:       Locked
✔ csme override:                 Locked
✔ csme v0:18.0.5.2098:           Valid
✔ Platform debugging:            Disabled
✔ SPI write:                     Disabled
✔ SPI lock:                      Enabled
✔ SPI BIOS region:               Locked
✔ Supported CPU:                 Valid
✔ UEFI bootservice variables:    Locked
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled
✘ TPM v2.0:                      Not found

HSI-2
✔ BIOS rollback protection:      Enabled
✔ Intel BootGuard:               Enabled
✔ Intel BootGuard OTP fuse:      Valid
✔ IOMMU:                         Enabled
✘ Intel BootGuard ACM protected: Invalid
✘ Intel BootGuard verified boot: Invalid
✘ Platform debugging:            Unlocked

HSI-3
✔ CET Platform:                  Supported
✔ Pre-boot DMA protection:       Enabled
✔ Suspend-to-idle:               Enabled
✔ Suspend-to-ram:                Disabled
✘ Intel BootGuard error policy:  Invalid

HSI-4
✔ SMAP:                          Enabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -!
✔ fwupd plugins:                 Untainted
✔ CET OS Support:                Supported
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Encrypted
✔ Linux kernel:                  Untainted

This system has a low HSI security level.
 » https://fwupd.github.io/hsi.html#low-security-level

Host Security Events
  2024-05-20 15:40:26:  ✘ TPM v2.0 changed: Found → Not found
  2024-05-20 15:40:26:  ✘ Platform debugging changed: Locked → Unlocked
  2024-04-30 12:59:17:  ✔ CET OS Support changed: Not supported → Supported
  2024-04-29 19:28:42:  ✔ Secure Boot enabled
  2024-04-29 12:28:05:  ✘ Secure Boot disabled
  2024-04-25 22:07:12:  ✔ Linux swap changed: Disabled → Encrypted
  2024-04-24 22:59:56:  ✔ Kernel lockdown enabled
  2024-04-24 22:53:13:  ✔ BIOS rollback protection changed: Disabled → Enabled
  2024-04-24 20:55:32:  ✔ Secure Boot enabled
```

This is wrong because TPM is enabled and in used. `bootclt`, for example, correctly reports that

```
System:
      Firmware: UEFI 2.70 (Lenovo 0.4752)
 Firmware Arch: x64
   Secure Boot: enabled (user)
[... 344 lines cut ...]
│   
├─UEFI Platform Key:
│     Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
│     Summary:            Platform Key
│     GUID:               0d495878-553a-5803-ad0b-307df77318d6 ← UEFI\CRT_FEB00E5A22F47D40463AF378E6A5E774C0F0ED9E
│   
└─USB4 Retimer:
      Device ID:          16d35c510618116e7899ae45e62605572872e713
      Summary:            A physical layer protocol-aware, software-transparent extension device that forms two separate electrical link segments
      Current version:    23.81
      Vendor:             Intel (THUNDERBOLT:0x8087)
      GUIDs:              74fb1b87-53e6-5115-bd02-a9fefd06ed9a ← THUNDERBOLT\VEN_8087&DEV_0D9C
                          dac6f95d-7c5b-52b5-8c79-a0436a5ef784 ← TBT-80870d9c-retimer0-0:1.1
      Device Flags:       • Internal device
                          • Updatable
                          • Device stages updates
```

</details>

**Additional questions**

- Operating system and version: Arch Linux
- Have you tried rebooting? Yes
- Is this a regression? Yes

iyanmv, 2024-07-29:
Also curious about the new Platform debugging item in HSI-2. Is this due to [Intel DCI](https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.PlatformDebugEnabled)? If so, how can this be locked? I can't find any information online.

michaellass, 2024-07-29:
I think this is the same issue as #7545. Could you please try the following?

```shell
grep -v DeviceAllow /usr/lib/systemd/system/fwupd.service > /etc/systemd/system/fwupd.service
systemctl daemon-reload
systemctl stop fwupd.service
```

and then try again to run `fwupdmgr security`?

iyanmv, 2024-07-29:
> I think this is the same issue as #7545. Could you please try the following?

Yes, it's the same.

iyanmv, 2024-07-29:
I think the problem is that the default `DevicePolicy=auto` only applies if no explicit `DeviceAllow=` is present. Now that there is one to fix #7414, all devices should be added explicitly.

iyanmv, 2024-07-29:
Yes, that seems to be the case. Adding the following two lines makes TPM appear again:

```
DeviceAllow=char-tpm rw
DeviceAllow=char-cpu/* r
```

However, more `DeviceAllow` are needed to fix #7545. I wonder if there is an alternative solution to #7414 that modifies the implicit `DeviceAllow=char-hidraw r` to `DeviceAllow=char-hidraw rw` without having to manually add all the devices to the allow list.

space-blade, 2024-07-30:
> Also curious about the new Platform debugging item in HSI-2. Is this due to [Intel DCI](https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.PlatformDebugEnabled)? If so, how can this be locked? I can't find any information online.

I'm curious about how can I get this to be locked as well. Have you found any solution?

closed 2024-07-30 by superm1

MrRinkana, 2024-08-07:
This might not be relevant for fwupdmgr but might help people searching this issue:

For me this issue was fixed by installing `tpm2.0-tools`. Reboot and tpm2.0 is detected. I am running Tumbleweed.

I knew I had working tpm2.0 as I'm using it for FDE and it was previously found (2024-04-01 20:25:03:  ✘ TPM v2.0 changed: Found → Not found) which I don't know why but maybe tpm2.0-tools got uninstalled if dropped as some dependency?