fwupd/fwupd #7555: about redfish plugin-specific quirks usage

vincent-tang1, 2024-07-30:
**Describe the question**
about redfish quirks, '**RedfishResetPreDelay**' parameter can be secified in redfish.quirk file
https://github.com/fwupd/fwupd/tree/main/plugins/redfish
can we also specify this '**RedfishResetPreDelay**' parameter to firmware **metadata** for usage ?

**fwupd version information**
Please provide the version of the daemon and client if applicable to your current installation of `fwupd`.

```shell
# fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.17
compile   com.hughsie.libjcat           0.2.0
compile   org.freedesktop.fwupd         1.9.16
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.8.0-31-generic
runtime   org.freedesktop.fwupd         1.9.16

```

vincent-tang1, 2024-07-31:
Accodring your metadata guide
https://lvfs.readthedocs.io/en/latest/metainfo.html
I can not found any plugin quirk configuration usage in metadata.
so seems like that firmware metadata is uable to support plugin quirk configuration, right ?
 
For example in redfish
```
https://github.com/fwupd/fwupd/blob/main/plugins/redfish/redfish.quirk
https://github.com/fwupd/fwupd/blob/main/plugins/redfish/README.md

about below quirk parameters:

RedfishResetPreDelay
Delay in ms to use before querying the manager after a cleanup reset, default 0ms.

Since: 1.8.0

RedfishResetPostDelay
Delay in ms to use before querying /redfish/v1/UpdateService after a cleanup reset, default 0ms.

Since: 1.8.0
```

hughsie, 2024-11-11:
Sorry for the delay. No, we can't set that parameter in the metadata. This all has to work 100% offline too, and so the best place for any device quirks would be in the `redfish.quirk` file.

closed 2024-11-11 by hughsie