fwupd/fwupd #7557: Local build - running fwupd binary fail

vincent-tang1, 2024-07-30:
**Describe the bug**
use trunk source today, after finish the build, unable to execute binary fwupd 

**Steps to Reproduce**
follow below official steps below and finnally fail to execute fwupd
https://github.com/fwupd/fwupd/blob/main/docs/building.md
```
(fwupd) root@vincent:/home/vincent/fwupd# fwupd
14:05:14.649 FuMain               Daemon ready for requests (locale en_US.UTF-8)
14:05:14.649 FuMain               another service has claimed the dbus name org.freedesktop.fwupd
```

**Expected behavior**
fwupd binary can running in the background

- Operating system and version:  ubuntu 24.04
- Other : remove distrubution fwupd using "sudo apt purge fwupd" and make sure snap fwup to be removed with "snap remove fwupd"

vincent-tang1, 2024-07-30:
detail log
[0730.log](https://github.com/user-attachments/files/16429353/0730.log)

superm1, 2024-07-30:
I think the action of purging the installed package daemon also removed the dbus rule that lets dbus daemon recognize fwupd.

For a workaround for now you can reinstall package but run your locally built one.

We'll need some changes to our setup script for this case I think💬.

vincent-tang1, 2024-07-31:
got it, following your suggestion, it can work, thanks

closed 2024-08-08 by superm1