fwupd/fwupd #7560: v1.9.21 to v1.9.22 - Intel GDS mitigation & Platform debugging both changed to ✘

genkieagle, 2024-07-31:
I noted from v1.9.21 to v1.9.22 something changed for "Intel GDS mitigation" and Platform debugging on my system. 

On version 1.9.21 i had such details
```
Host Security ID: HSI:2 (v1.9.21)

HSI-1
✔ BIOS firmware updates:         Enabled
✔ MEI key manifest:              Valid
✔ csme manufacturing mode:       Locked
✔ csme override:                 Locked
✔ csme v0:14.1.72.2287:          Valid
✔ Platform debugging:            Disabled
✔ SPI write:                     Disabled
✔ SPI lock:                      Enabled
✔ SPI BIOS region:               Locked
✔ Supported CPU:                 Valid
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI bootservice variables:    Locked
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled

HSI-2
✔ BIOS rollback protection:      Enabled
✔ Intel BootGuard ACM protected: Valid
✔ Intel BootGuard:               Enabled
✔ Intel BootGuard OTP fuse:      Valid
✔ Intel BootGuard verified boot: Valid
✔ Intel GDS mitigation:          Enabled
✔ IOMMU:                         Enabled
✔ Platform debugging:            Locked
✔ TPM PCR0 reconstruction:       Valid

HSI-3
✔ Intel BootGuard error policy:  Valid
✔ Pre-boot DMA protection:       Enabled
✔ Suspend-to-ram:                Disabled
✘ CET Platform:                  Not supported
✘ Suspend-to-idle:               Disabled

HSI-4
✔ SMAP:                          Enabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -!
✔ fwupd plugins:                 Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Encrypted
✔ Linux kernel:                  Untainted

Host Security Events
 2024-03-16 09:26:15:  ✔ TPM v2.0 changed: Not found → Found
 2024-03-16 08:58:20:  ✔ Platform debugging changed: Not supported → Disabled
 2024-03-16 08:58:20:  ✔ IOMMU device protection enabled
 2024-03-16 08:58:20:  ✔ Platform debugging changed: Not supported → Locked
```

While on the current version 1.9.22 shown as 
```
Host Security ID: HSI:1 (v1.9.22)

HSI-1
✔ BIOS firmware updates:         Enabled
✔ csme manufacturing mode:       Locked
✔ csme override:                 Locked
✔ csme v0:14.1.72.2287:          Valid
✔ Platform debugging:            Disabled
✔ SPI write:                     Disabled
✔ SPI lock:                      Enabled
[... 142 lines cut ...]
│     Vendor:             DMI:Dell Inc.
│     Update State:       Success
│     GUID:               4dd8e6a7-2e7f-884f-9879-73328600e2d3
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
```

Distro details
```
Operating System: openSUSE Tumbleweed 20240730
KDE Plasma Version: 6.1.3
KDE Frameworks Version: 6.4.0
Qt Version: 6.7.2
Kernel Version: 6.10.2-1-default (64-bit)
Graphics Platform: Wayland
Processors: 8 × Intel® Core™ i7-10700 CPU @ 2.90GHz
Memory: 30.9 GiB of RAM
Graphics Processor: Mesa Intel® UHD Graphics 630
```

Please let me know if other info needed to be shown.

superm1, 2024-07-31:
See if https://github.com/fwupd/fwupd/commit/de292181ae418312a7dd823df24ecd30917454e8 helps

And if it does can you please share your /proc/devices output?

genkieagle, 2024-07-31:
Are you saying to do the same action as [michaellass](https://github.com/fwupd/fwupd/issues/7547#issuecomment-2255566316)?

superm1, 2024-07-31:
> Are you saying to do the same action as [michaellass](https://github.com/fwupd/fwupd/issues/7547#issuecomment-2255566316)?

Well I want to confirm if it's the same regression or not. If reverting that helps then I want to get the right fix into main.

superm1, 2024-07-31:
Thanks we're missing another one for mei then. I'll add it to main.

closed 2024-07-31 by superm1

hughsie, 2024-08-05:
@superm1 want me to roll a 1.9.23 today?

superm1, 2024-08-05:
Yes

hughsie, 2024-08-05:
Done -- thanks for holding the fort for me -- I really appreciate it.