fwupd/fwupd #7562: "0 local devices supported" is incorrect - it should be "0 updates are available".

RokeJulianLockhart, 2024-07-31:
#### **Is your feature request related to a problem? Please describe.**

Despite `fwupdmgr get-devices` returning multiple devices:

<details>

<blockQuote>
	
```log
PS /home/RokeJulianLockhart> fwupdmgr get-devices                                                                            
ASRock X670E Taichi
│
├─600 Series Chipset SATA Controller:
│     Device ID:          75f132be50e56140af0b7154b0192c35bd0bfb7b
│     Current version:    01
│     Vendor:             Advanced Micro Devices, Inc. [AMD] (PCI:0x1022)
│     GUIDs:              c12a74b6-78b0-5cfb-95a3-cd65a12f9e94 ← PCI\VEN_1022&DEV_43F6
│                         38b47a77-8710-5b30-b79b-91df5c9a6d96 ← PCI\VEN_1022&DEV_43F6&SUBSYS_1B211062
│                         67d3b17f-d200-5c56-979e-8a3a4ab46c2b ← PCI\VEN_1022&DEV_43F5
│                         6b2e126d-81a8-530b-a8d3-a93e65bd30b2 ← PCI\VEN_1022&DEV_43F5&SUBSYS_1B213328
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─600 Series Chipset SATA Controller:
│     Device ID:          1e5b4ef16e1bf9cc6601c6c3f7c5736719fa9e19
│     Current version:    01
│     Vendor:             Advanced Micro Devices, Inc. [AMD] (PCI:0x1022)
│     GUIDs:              c12a74b6-78b0-5cfb-95a3-cd65a12f9e94 ← PCI\VEN_1022&DEV_43F6
│                         38b47a77-8710-5b30-b79b-91df5c9a6d96 ← PCI\VEN_1022&DEV_43F6&SUBSYS_1B211062
│                         67d3b17f-d200-5c56-979e-8a3a4ab46c2b ← PCI\VEN_1022&DEV_43F5
│                         6b2e126d-81a8-530b-a8d3-a93e65bd30b2 ← PCI\VEN_1022&DEV_43F5&SUBSYS_1B213328
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Navi 10 [Radeon RX 5600 OEM/5600 XT / 5700/5700 XT]:
│     Device ID:          594baa45f6742120d71dd82166e762f1529c2a8a
│     Current version:    113-1701_XL_NAVI10_8GBD6_MS_190612_W8
│     Vendor:             Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│     GUIDs:              dd0fc8c4-f037-5a78-bdea-c85eb0fe68fa ← PCI\VEN_1002&DEV_731F
│                         07cab971-5486-5e92-b663-7ff6d9cd5400 ← PCI\VEN_1002&DEV_731F&SUBSYS_16825705
│                         800200db-2c4f-5f72-b833-04fc8a49959f ← PCI\VEN_1002&DEV_1479
│                         a2e4f111-f931-56a4-aed7-cd100613e661 ← PCI\VEN_1002&DEV_1479&SUBSYS_10021479
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
└─Unifying Receiver:
      Device ID:          348854c472ac6a43cb8e97ef3a72984e2f154b74
      Summary:            Miniaturised USB wireless receiver
      Current version:    RQR24.11_B0036
      Bootloader Version: BOT03.03_B0009
      Vendor:             Logitech, Inc. (HIDRAW:0x046D, USB:0x046D)
      Install Duration:   30 seconds
      GUIDs:              cc4cbfa9-bf9d-540b-b92b-172ce31013c1
                          279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
      Device Flags:       • Updatable
                          • Signed Payload
    
```

</blockQuote>

</details>

...`fwupdmgr refresh` returned:

> Successfully downloaded new metadata: 0 local devices supported

As a new user, this was confusing, because my devices *are supported*, as the previous command demonstrates.

Consequently, I asked about this discrepancy. In response, https://github.com/fwupd/fwupd/discussions/7548#discussioncomment-10184001 stated:

> Supported means the manufacturer has published updates for your device to LVFS. If nothing is marked supported, none of the manufacturers for devices in your system published updates.

#### **Describe the solution you'd like**

As https://github.com/fwupd/fwupd/discussions/7548#discussioncomment-10184020 explains:

> Surely that phrasal is incorrect, then? The devices are supported by `fwupd`, but no updates are available.

Consequently, I suggest that the relevant string explain that "0 updates are available", rather than "0 local devices supported".

#### **Describe alternatives you've considered**

Phrasal doesn't usually have alternatives if something specific needs to be communicated.

#### **Additional context**

1.	https://github.com/fwupd/fwupd/discussions/7548#discussioncomment-10184076 stated:

	> It's a matter of perspective from the fwupd software or from LVFS service. If you have some better proposal feel free to open a pull request.

1.	I wanted to create a [new PR](https://github.com/fwupd/fwupd/compare), but I've never dealt with projects translated to multiple languages, so I don't know what to modify, nor whether I should instead go through something like Weblate.

superm1, 2024-08-02:
The problem with your proposed change is that is that it masks the case of there is an update published to the LVFS but you are already up to date.  In this case you have a device "supported" by the remote server but that there just aren't updates available for.

IMO this case is important to message to a user because it shows that the manufacturer has published _some_ updates.

How about this; https://github.com/fwupd/fwupd/pull/7567

RokeJulianLockhart, 2024-08-02:
> The problem with your proposed change is that is that it masks the case of there is an update published to the LVFS but you are already up to date. In this case you have a device "supported" by the remote server but that there just aren't updates available for.

@superm1, isn't that an expected state? As an example, if I check for an update using `dnf` or F-Droid, and see no results, I can't see how that might indicate that there are no versions of the respective package on the server. If it were (for a reason unbenknownst to me) important for the user to be informed at the CLI that currently installed updates are held on the server, an additional "`%u versions on server`" line would communicate this clearly, allowing us to inform the user that the current version matches the latest version on the server.

As an example:

1.	```sh
	fwupdmgr refresh --force
	```
	
1.	```log
	Updating lvfs
	Downloading…             [***************************************]
	Successfully downloaded new metadata:
	    Installed version matches latest on server
        52 versions on server
	```

superm1, 2024-08-02:
The point of the message is mostly to let you know whether LVFS (or any configured remote) potentially can update any of your devices by telling you whether any have updates ever published by the manufacturer.

Similarly if there was an older version on LVFS than you have on your system it should be marked supported.

It's not to tell you how "many" updates are ever published.

Maybe when zero devices are supported on the server we should just show an "error" instead like:

"Fwupd can update X devices on this system but no firmware updates have been published to any remote servers".

superm1, 2024-08-04:
I've updated the pull request.

RokeJulianLockhart, 2024-08-04:
> https://github.com/fwupd/fwupd/issues/7562#issuecomment-2267637548

@superm1, https://github.com/fwupd/fwupd/commit/41385a1ed3a83f8517630bfad51030547bc576b6#diff-4c9148ee5d2903edb97fb6b4c7347653ebb8077dd0ea9df77b65fcb08a1f5be5R2151-R2163 does at least demonstrate to the user what that terminology means in this case, so I'll close this issue (since I can't see any reason your PR might be rejected). Many thanks.

closed 2024-08-04 by RokeJulianLockhart